mobizt/ESP-Mail-Client #250: Bad time from NTP servers causing email send fail?

Kingmiwok, 2023-04-02:
**Build tool used:**
Using Arduino IDE

- [x] Arudino
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

ESP8266  HiLetgo

- [x] ESP8266-based
- [ ] ESP32-based

**Other Libraries That are used:**

Sorry, this is a mess.  I hope you understand it.  
I'm running your sample code Send_HTML.  I'm trying both the 3.1.1 and the 3.1.3 version. The only changes are for my email account and wifi info.   

**Description of problem:**

The problem is very inconsistent, so it's hard to pin down.  Sometimes it sends an email, and sometimes it fails.  When it fails with 3.1.3, it does not give details why.  It just stops in the process, but I get the impression it's failing to contact the time server.

With 3.1.1, when it fails, the email server references this info.

554 Transaction failed, reject due to policy restrictions
Problem:
The email was rejected by the IONOS SMTP server because the standards defined in RFC 5321 and RFC 5322 were not adhered to.
Solution:
The email client must meet the following criteria:
The header lines of the email must contain exactly one valid "Date" line according to RFC 2822.
The following header lines may only exist once: From, Sender, To, CC, Subject
The header lines mentioned must be syntactically correct
Note: The To and CC lines can contain multiple recipients.
In accordance with the recommendations from RFC7103, incorrect messages are not automatically corrected by IONOS, otherwise DKIM signatures become invalid. For more information about DKIM signatures, see the IONOS article Configuring DMARC Records for a Domain.

When you look at the serial output (see below), you see that the date is not complete.  The last character of the year is missing.  When it works, it either has the default date of 1970, or you see a full 2023 date.  (The problem is not consistent.  It's not very repeatable.)  The only other thing I noticed is that it worked yesterday before midnight, but once it turned April 1 (no joke) it started failing sometimes.  However, I only started testing the example late in the evening.
....
21:19:47.047 -> #### Finishing the message sending...
21:19:47.047 -> > C: finishing the message sending
21:19:47.079 -> < S: 554-Transaction failed
21:19:47.079 -> < S: 554-Reject due to policy restrictions.
21:19:47.079 -> < S: 554 For explanation visit https://www.ionos.com/help/index.php?id=2425&ip=68.105.212.108&c=hd
21:19:47.111 -> #### Error, send body failed
21:19:47.111 -> ! E: send body failed
21:19:47.111 -> 
21:19:47.111 -> #### -------------------------
21:19:47.111 -> #### Message sent success: 0
21:19:47.111 -> #### Message sent failed: 1
21:19:47.111 -> #### -------------------------
21:19:47.111 -> #### Status: failed
21:19:47.111 -> #### Date/Time: Sun Apr  2 03:19:46 202
21:19:47.111 -> #### Recipient: glen@glen-miller.com
21:19:47.111 -> #### Subject: Test sending html Email
21:19:47.111 -> Error sending Email, send body failed
21:19:47.157 -> Free Heap: 35944

**Share code snippet to reproduce the issue:**
```yaml
PASTE .cpp / .ino code here

```

**Additional information and things you've tried:**

This one worked.  It has a proper date.

21:17:13.428 -> Connecting to Wi-Fi...........
21:17:17.485 -> Connected with IP: 192.168.1.47
21:17:17.485 -> 
21:17:17.485 -> #### Connecting to SMTP server...
21:17:17.485 -> > C: ESP Mail Client v3.1.3
21:17:17.485 -> > C: wait for NTP server time synching
21:17:17.641 -> > C: connecting to SMTP server
21:17:17.641 -> > C: Host > smtp.1and1.com
21:17:17.641 -> > C: Port > 587
21:17:17.732 -> 
21:17:17.732 -> #### SMTP server connected
21:17:17.732 -> > C: SMTP server connected, wait for greeting...
21:17:17.810 -> < S: 220 perfora.net (mreueus003) Nemesis ESMTP Service ready
21:17:17.810 -> 
21:17:17.810 -> #### Sending greeting response...
21:17:17.810 -> > C: send SMTP command, HELO
21:17:17.903 -> < S: 250-perfora.net Hello mydomain.net [68.105.212.108]
21:17:17.903 -> < S: 250-8BITMIME
21:17:17.903 -> < S: 250-SIZE 140000000
21:17:17.903 -> < S: 250 STARTTLS
21:17:17.903 -> 
21:17:17.903 -> #### Sending STARTTLS command...
21:17:17.903 -> > C: send command, STARTTLS
21:17:18.043 -> < S: 220 OK
21:17:18.043 -> > C: perform SSL/TLS handshake
21:17:18.043 -> 
21:17:18.043 -> > C: SSL/TLS negotiation
21:17:19.115 -> 
21:17:19.115 -> #### Sending greeting response...
21:17:19.115 -> > C: send SMTP command, HELO
21:17:19.240 -> < S: 250-perfora.net Hello mydomain.net [68.105.212.108]
21:17:19.240 -> < S: 250-8BITMIME
21:17:19.240 -> < S: 250-AUTH LOGIN PLAIN
21:17:19.240 -> < S: 250 SIZE 140000000
21:17:19.240 -> 
21:17:19.240 -> #### Logging in...
21:17:19.240 -> > C: send SMTP command, AUTH PLAIN
21:17:19.287 -> > C: arduino@glen-miller.com
21:17:19.287 -> > C: *************
21:17:19.396 -> < S: 235 Authentication succeeded
21:17:19.441 -> 
21:17:19.441 -> Successfully logged in.
21:17:19.441 -> 
21:17:19.441 -> #### Sending Email...
21:17:19.441 -> > C: send Email
21:17:19.441 -> 
21:17:19.441 -> #### Sending message header...
21:17:19.441 -> > C: send message header
21:17:19.598 -> < S: 250 Requested mail action okay, completed
21:17:19.738 -> < S: 250 OK
21:17:19.738 -> 
21:17:19.738 -> #### Sending message body...
21:17:19.738 -> > C: send message body
21:17:19.862 -> < S: 354 Start mail input; end with <CRLF>.<CRLF>
21:17:20.051 -> 
21:17:20.051 -> #### Finishing the message sending...
21:17:20.051 -> > C: finishing the message sending
21:17:20.082 -> < S: 250 Requested mail action okay, completed: id=0MEovc-1pciKT284o-00G2KY
21:17:20.082 -> 
21:17:20.082 -> #### Closing the session...
21:17:20.082 -> > C: terminate the SMTP session
21:17:20.082 -> 
21:17:20.082 -> #### Message sent successfully
21:17:20.082 -> > C: message sent successfully
21:17:20.082 -> 
21:17:20.082 -> ----------------
21:17:20.127 -> Message sent success: 1
21:17:20.127 -> Message sent failed: 0
21:17:20.127 -> ----------------
21:17:20.127 -> 
21:17:20.127 -> Message No: 1
21:17:20.127 -> Status: success
21:17:20.127 -> Date/Time: Sun Apr  2 03:17:19 2023
21:17:20.127 -> 
21:17:20.127 -> Recipient: glen@glen-miller.com
21:17:20.127 -> Subject: Test sending html Email
21:17:20.127 -> ----------------
21:17:20.127 -> 
21:17:20.127 -> Free Heap: 36584

Kingmiwok, 2023-04-02:
I think there are two things happening:   
1)  Failing to connect to the time server: 3.1.1 uses Jan 1 1970 and the email sends.  3.1.3 just stops (by design?)
2)  Sometimes the returned time is getting truncated, and the date is not complete, and the email fails to send because the smtp server rejects it.

mobizt, 2023-04-02:
The library internal time for ESP32, ESP8266 and Pico obtained from system time or time(nullptr. if the system time does not set, it returns the seconds since device started.

I already known and aware of the RFC requirements, that's why NTP synching was involved in this library.

Library tries to connect to NTP server when you call `smtp.connect` and `imap.connect` or in case server certificate was set to the config.

In library version prior to 3.x.x, if the time server synching was timed out, the error will be reported in the debug port and the processes will be continued.

The library versions v3.0.0 to v3.1.5, the connection was rejected when NTP synching timed out, showing error on debug port and terminate connection.

As some reports from users, then in version 3.1.5 and later, library was just showing error message on debug port and the server connection processes can be continued.

You should take care of setting your device time first (in case ESP32. ESP8266 and Pico). In other Arduino devices, you should set the library time manually.

To set your device time and/or library time manually, please see
https://github.com/mobizt/ESP-Mail-Client/blob/master/examples/SMTP/Set_Time/Set_Time.ino

closed 2023-04-02 by mobizt

mobizt, 2023-04-02:
There is nothing different between v 3.1.1 and v3.1.3 for date/time related functions. The time synching timed out error will always report.

The valid date header for SMTP was set since[ v1.5.10](https://github.com/mobizt/ESP-Mail-Client/releases/tag/v1.5.10) which the NTP synching process will be done first when using SMTP.

Prior to v1.5.10, the time synching is optional when using ESP8266 and server certificate was assigned in config object.

mobizt, 2023-04-02:
The time offset (second starts with zero) is Jan 1, 1970 midnight. Then when the timestamp was not set in library due to NTP synching failure, the time getting from library internal time will be the seconds since boot and the time string will be somethings like I wrote in the examples comments.
https://github.com/mobizt/ESP-Mail-Client/blob/f41509840a34edcba3671be2d1428484cab922ce/examples/SMTP/Send_Text/Send_Text.ino#L346-L355

mobizt, 2023-04-02:
You may don't notice the meaning of TAG that library used in debug message which everyone knows it. 

```
< S: 
> C: 
! E:
! I:
! W:
```
The response code e.g., 220, 250, 235 returns directly from server, and it tells everything, and it's not related to library version but the server specific.

mobizt, 2023-04-02:
It is better to remove useless and irrelevant additional information of time string from Arduino debug when you want to post the debug message of the library.
`21:17:20.127 ->`

Kingmiwok, 2023-04-02:
Thank you for the replies.  
The thing that caught my attention the most was the incomplete Date/Time when it failed to send the email.  
21:19:47.111 -> #### Date/Time: Sun Apr 2 03:19:46 202
Somehow the 3 (from 2023) was truncated.  I suspected that was why the smtp server gave the 554 error.  

Other times, the Date/Time was complete, and the email was sent. 
21:17:20.127 -> Date/Time: Sun Apr 2 03:17:19 2023

mobizt, 2023-04-02:
Which device you are using? ESP8266, ESP32 or Pico.

I guess it can be related to compiler and string manipulation.

mobizt, 2023-04-02:
This never happens for my intensive test in ESP8266, ESP32 and Pico with latest SDK.

mobizt, 2023-04-02:
Ok, that is only error in debug information printing time string buffer that convert from timestamp (seconds since Jan 1, 1970 midnight) using `asctime` function which the time string has additional CRLF appending then I have to replace them with null before printing.

This depends on GCC version of your current SDK version that the time string created by `asctime` function will contain CRLF or not.

But it is not related to the message to send as the date header treated differently which `asctime` was not used to make time string.

Anyway, because of time was not set or synch failure, you should connect your device to stable and reliable network or manually set the time before calling connect.

mobizt, 2023-04-03:
Ok now I understand your point. I can confirm that the library complied with the related RFC standards for SMTP and IMAP.

In general, SMTP error 554 indicates that the mail server did not accept the email. There are several reasons that cause this error which you also post above from server documentation and [these typical reasons and solutions](https://www.courier.com/error-solutions/smtp-error-554/).

The invalid recipient email and invalid headers are just minor causes and can be fixed as easy but there are other possible reasons that you have to solve on your side.

You should make sure that your account, your server IP or your router public IP are not blacklisted from SMTP server.

If your router IP and server IP are blacklisted, please consider the Gmail SMTP relay.

For example, if you install mail server on your own server and the email was rejected with status 554 when trying to send email on your SMTP sever, the most possible cause is your server (router) public IP from internet ISP is blacklisted.

Kingmiwok, 2023-04-03:
Thank you for the replies.   I'm using a HiLetgo ESP8266

Per your suggestion, I've been using the Set_Time example, with ESP Mail Client v3.1.6.  So far it's worked every time.  Whatever the problem was before, it does not happen with Set_Time.  

Concerning the 554 error... The program would randomly fail or succeed, within minutes, with no changes to the code.  The only difference I saw was the truncated year character.  I don't know if that was the cause, but the character was consistently missing if it failed.  It's interesting anyhow.   

So far my email server host seems fine with my emails. No blacklist.  I'm trying not to send too many emails this way.  I often email myself during the day as a reminder to do things, so it's not too unusual.

mobizt, 2023-04-03:
The incorrect date occurred only in debug message. The library has built in time library as the time provider, and it works normally.

You should update to v1.3.7 which fixed this incorrect debug info.

The status code 554 means your message was rejected by server. If you are not blacklisted by server, the server may limit the rate of messages sent per minute and consider your message as spam when you send email too often than its limit.