mobizt/ESP-Mail-Client #260: ttttttttt

Daryavahus, 2023-05-01:
When I compile sketch a get error like this:
In file included from d:\Moje Dokumenty\Arduino\libraries\ESP_Mail_Client\src\ESP_Mail_Client.cpp:42:
d:\Moje Dokumenty\Arduino\libraries\ESP_Mail_Client\src\ESP_Mail_SMTP.h: In member function 'void ESP_Mail_Client::getResponseStatus(const char*, esp_mail_smtp_status_code, int, esp_mail_smtp_response_status_t&)':
d:\Moje Dokumenty\Arduino\libraries\ESP_Mail_Client\src\ESP_Mail_SMTP.h:2962:25: warning: comparison of integer expressions of different signedness: 'size_t' {aka 'unsigned int'} and 'int' [-Wsign-compare]
2962 | if (strlen(buf) > codeLength && (buf[codeLength] == ' ' || buf[codeLength] == '-') && textLength > 0)
| ~~~~~~~~~~~~^~~~~~~~~~~~
d:\Moje Dokumenty\Arduino\libraries\ESP_Mail_Client\src\ESP_Mail_SMTP.h:2983:26: warning: comparison of integer expressions of different signedness: 'int' and 'size_t' {aka 'unsigned int'} [-Wsign-compare]
2983 | while (i < strlen(buf) && buf[i] != ' ')
| ~~^~~~~~~~~~~~~
d:\Moje Dokumenty\Arduino\libraries\ESP_Mail_Client\src\ESP_Mail_SMTP.h:2987:24: warning: comparison of integer expressions of different signedness: 'int' and 'size_t' {aka 'unsigned int'} [-Wsign-compare]
2987 | i = (i < strlen(buf) - 1) ? i + 1 : codeLength + 1;
| ~~^~~~~~~~~~~~~~~~

Is it posiible to generate pm_send_nullfunc error(SWDT - Random Fatal exception:4) by those errors?

mobizt, 2023-05-01:
That is just compiler warning.

mobizt, 2023-05-01:
The error in compilation error you asked is from your code.

closed 2023-05-01 by mobizt

Daryavahus, 2023-05-02:
I use it. Thank You. Always when I see warnings I think: If  is it important?

mobizt, 2023-05-02:
The warning is just about comparison between signed integer (int) and unsigned integer (size_t) which is not important and not affected to library operation as the value stores in the signed integer variable is actually positive.

Daryavahus, 2023-05-02:
OK. Thanks for explanation.