mobizt/ESP-Mail-Client #261: SD_MMC not attaching file to email

wiehen, 2023-05-16:
**Build tool used:**

- [X] Arudino
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based
- [X] ESP32-based

**Other Libraries That are used:**

#include "esp_camera.h"
#include "Arduino.h"
#include "FS.h"                // SD Card ESP32
#include "SD_MMC.h"            // SD Card ESP32
#include "soc/soc.h"           
#include "soc/rtc_cntl_reg.h"  
#include <EEPROM.h>            // read and write from flash memory
#include <WiFi.h>

**Description of problem:**
I am using SD_MMC to write a picture that is taken to the sd card. This is done successfully and I can see the picture on the card if I look at it. I cannot get the mail client to attach the picture to the email using SD_MMC. The email is sent successfully and received just with no attachment.

It does not seem like these definitions in my code
define ESP_MAIL_DEFAULT_SD_FS SD_MMC //For ESP32 SDMMC
#define ESP_MAIL_CARD_TYPE_SD_MMC 1
do anything when you include SDHelper.h it never reaches the if statement that is defined by the above. It always goes to the section defined like this:

#elif defined(ESP32) // if ESP32 and no SdFat library installed
    Serial.print("\nMounting SD Card... No SDFat... ");

You can see that I changed the serial.print statement in my code and you can see the matching output in my serial output below.

**Share code snippet to reproduce the issue:**

PASTE .cpp / .ino code here

```
#include <ESP_Mail_Client.h>

#include "esp_camera.h"
#include "Arduino.h"
#include "FS.h"                // SD Card ESP32
#include "SD_MMC.h"            // SD Card ESP32
#include "soc/soc.h"           
#include "soc/rtc_cntl_reg.h"  
#include <EEPROM.h>            // read and write from flash memory
#include <WiFi.h>

#include <extras/SDHelper.h>

// define the number of bytes you want to access
#define EEPROM_SIZE 1

// Pin definition for CAMERA_MODEL_AI_THINKER
#define PWDN_GPIO_NUM     32
#define RESET_GPIO_NUM    -1
#define XCLK_GPIO_NUM      0
#define SIOD_GPIO_NUM     26
#define SIOC_GPIO_NUM     27

#define Y9_GPIO_NUM       35
#define Y8_GPIO_NUM       34
#define Y7_GPIO_NUM       39
#define Y6_GPIO_NUM       36
#define Y5_GPIO_NUM       21
#define Y4_GPIO_NUM       19
[... 226 lines cut ...]
[ 11206][E][sd_diskio.cpp:802] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[ 11451][I][ESP32_SSL_Client.cpp:244] connect_ssl(): WARNING: Skipping SSL Verification. INSECURE!
[ 13802][W][sd_diskio.cpp:180] sdCommand(): crc error
[ 13902][W][sd_diskio.cpp:516] ff_sd_initialize(): GO_IDLE_STATE failed
[ 13903][E][sd_diskio.cpp:802] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[ 13907][W][sd_diskio.cpp:186] sdCommand(): token error [0] 0x41
[ 14855][D][WiFiClient.cpp:546] connected(): Disconnected: RES: 0, ERR: 128
--------------------

I have changed SDHelper.h top match my SPI settings:
#if defined(ESP32)

//#define SPI_CS_PIN 13
//#define SPI_SCK_PIN 14
//#define SPI_MISO_PIN 2
//#define SPI_MOSI_PIN 15
//#define SPI_CLOCK_IN_MHz 16

 #define SPI_CS_PIN 5
 #define SPI_SCK_PIN 18
 #define SPI_MISO_PIN 19
 #define SPI_MOSI_PIN 23
 #define SPI_CLOCK_IN_MHz 4 // may work on lower clock rate

I try calling SD_Card_Mounting() in my code to see if that will work you can see from the output that it also fails to mount the card.

mobizt, 2023-05-16:
Please read the comments in the examples carefully.

Why you define this in your sketch? Where is the document you've seen that show you to do this.

```cpp
#define ESP_MAIL_DEFAULT_SD_FS SD_MMC //For ESP32 SDMMC
#define ESP_MAIL_CARD_TYPE_SD_MMC 1
```
That causes library can't see those maros you defined in your sketch and it will cause macro re-definition conflicts when compiling your sketch.

Only the place where you can define those maros is [ESP_Mail_FS.h](https://github.com/mobizt/ESP-Mail-Client/blob/master/src/ESP_Mail_FS.h#L117-L118)
https://github.com/mobizt/ESP-Mail-Client/blob/21962ab69ab583abe54bc286d9b1b3fe56536ceb/src/ESP_Mail_FS.h#L117-L118

The [ESP_Mail_FS.h](https://github.com/mobizt/ESP-Mail-Client/blob/master/src/ESP_Mail_FS.h) contains the default library configurations which your can create your own configuration file (Custom_ESP_Mail_FS.h) where it does not overwrite when you update the library which already stated in that file.

https://github.com/mobizt/ESP-Mail-Client/blob/21962ab69ab583abe54bc286d9b1b3fe56536ceb/src/ESP_Mail_FS.h#L159-L190
 
Also, you should remove your following card initialization code from your sketch
` if(!SD_MMC.begin()){`
And use library provided card initialization function instead.
` SD_Card_Mounting();`

If you intend to save your image to storage card before sending email, and you don't use multimedia card (mmc) unless use only SD or micro-SD card, please use SD library or high-performance library [SdFat](https://github.com/greiman/SdFat) instead.

If you only want to send raw image, please follow [this example](https://github.com/mobizt/ESP-Mail-Client/blob/master/examples/SMTP/Send_Camera_Image/ESP_Camera/ESP_Camera.ino).

closed 2023-05-16 by mobizt

wiehen, 2023-05-17:
I added the custom Custom_ESP_Mail_FS.h file with the SD_MMC override like is documented. When I compile my program after removing these statements from my ino file:
//#define ESP_MAIL_DEFAULT_SD_FS SD_MMC //For ESP32 SDMMC
//#define ESP_MAIL_CARD_TYPE_SD_MMC 1

I now get this error message:

In file included from /Users/nathanwiehe/Documents/Arduino/libraries/ESP-Mail-Client/src/extras/RFC2047.h:13,
                 from /Users/nathanwiehe/Documents/Arduino/libraries/ESP-Mail-Client/src/ESP_Mail_Client.h:39,
                 from /Users/nathanwiehe/Documents/Arduino/camera_mail_attachment/camera_mail_attachment.ino:1:
/Users/nathanwiehe/Documents/Arduino/libraries/ESP-Mail-Client/src/extras/MB_FS.h: In member function 'bool MB_FS::sdSPIBegin(int, SPIClass*, uint32_t)':
/Users/nathanwiehe/Documents/Arduino/libraries/ESP-Mail-Client/src/extras/MB_FS.h:256:74: error: no matching function for call to 'fs::SDMMCFS::begin(int&, SPIClass&, uint32_t&)'
             sd_rdy = MBFS_SD_FS.begin(ss, *sd_config.spiConfig, frequency);
                                                                          ^
In file included from /Users/nathanwiehe/Documents/Arduino/libraries/ESP-Mail-Client/src/Custom_ESP_Mail_FS.h:17,
                 from /Users/nathanwiehe/Documents/Arduino/libraries/ESP-Mail-Client/src/ESP_Mail_FS.h:194,
                 from /Users/nathanwiehe/Documents/Arduino/libraries/ESP-Mail-Client/src/extras/RFC2047.h:12,
                 from /Users/nathanwiehe/Documents/Arduino/libraries/ESP-Mail-Client/src/ESP_Mail_Client.h:39,
                 from /Users/nathanwiehe/Documents/Arduino/camera_mail_attachment/camera_mail_attachment.ino:1:
/Users/nathanwiehe/Library/Arduino15/packages/esp32/hardware/esp32/2.0.9/libraries/SD_MMC/src/SD_MMC.h:53:10: note: candidate: 'bool fs::SDMMCFS::begin(const char*, bool, bool, int, uint8_t)'
     bool begin(const char * mountpoint="/sdcard", bool mode1bit=false, bool format_if_mount_failed=false, int sdmmc_frequency=BOARD_MAX_SDMMC_FREQ, uint8_t maxOpenFiles = 5);
          ^~~~~
/Users/nathanwiehe/Library/Arduino15/packages/esp32/hardware/esp32/2.0.9/libraries/SD_MMC/src/SD_MMC.h:53:10: note:   no known conversion for argument 2 from 'SPIClass' to 'bool'
Multiple libraries were found for "SD.h"
  Used: /Users/nathanwiehe/Library/Arduino15/packages/esp32/hardware/esp32/2.0.9/libraries/SD
  Not used: /Users/nathanwiehe/Library/Arduino15/libraries/SD
exit status 1

Compilation error: exit status 1

mobizt, 2023-05-17:
Fist, remove your `Custom_ESP_Mail_FS.h` and modify the `ESP_Mail_FS.h` with SD_MMC to make sure you do the right thing.

If everything is ok with modified ESP_Mail_FS.h then implement the `Custom_ESP_Mail_FS.h`.

Consult the C/C++ programming documentation for programming.

mobizt, 2023-05-17:
### As I mentioned already, you should use SD.h library and SdFat library instead of SD_MMC.