mobizt/ESP-Mail-Client #263: SMTP plain text sends results in spinlock error

PureTek-Innovations, 2023-05-20:
**Build tool used:**

- [ ] Arudino
- [x ] Platformio (Visual Studio Code plugin) PlatformIO Core, version 6.1.7 Arduino V0.6.0
- [ ] Platformio CLI 
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based
- [x ] ESP32-based

**Other Libraries That are used:**

When I try and send a plain text email via SMTP using the example 
https://github.com/mobizt/ESP-Mail-Client/blob/master/examples/SMTP/Send_Text/Send_Text.ino

I get this error
WiFi connected
IP address:
192.168.1.238
#### Connecting to SMTP server...
> C: ESP Mail Client v3.1.10
> C: connecting to SMTP server
> C: Host > smtpout.secureserver.net
> C: Port > 465

> C: SSL/TLS negotiation
> C: seeding the random number generator
> C: setting up the SSL/TLS structure
! W: Skipping SSL Verification. INSECURE!
> C: setting hostname for TLS session
> C: perform the SSL/TLS handshake
> C: verifying peer X.509 certificate

#### SMTP server connected
> C: SMTP server connected, wait for greeting...
< S: 220 p3plsmtpa09-06.prod.phx3.secureserver.net :SMTPAUTH: 0KuBqRBaEvGwu : ESMTP server p3plsmtpa09-06.prod.phx3.secureserver.net ready

#### Sending greeting response...
> C: send SMTP command, EHLO
< S: 501 EHLO requires valid address
> C: cleaning SSL connection
> C: No ESMTP supported, send SMTP command, HELO
#### Error, connection closed
! E: connection closed

assert failed: spinlock_acquire spinlock.h:122 (result == core_id || result == SPINLOCK_FREE)

Backtrace:0x400836b9:0x3ffb22800x4008df49:0x3ffb22a0 0x400931b9:0x3ffb22c0 0x40090ed7:0x3ffb23f0 0x4008e459:0x3ffb2430 0x400d4db2:0x3ffb2450 0x400e4927:0x3ffb2470 0x400d8caf:0x3ffb2490 0x400dbfc9:0x3ffb24b0 0x400dc592:0x3ffb2550 0x400dfda6:0x3ffb25b0 0x400d384d:0x3ffb25d0 0x400e741e:0x3ffb2820

  #0  0x400836b9:0x3ffb2280 in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:402
  #1  0x4008df49:0x3ffb22a0 in esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c:128
  #2  0x400931b9:0x3ffb22c0 in __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c:85
  #3  0x40090ed7:0x3ffb23f0 in spinlock_acquire at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include/soc/spinlock.h:122
      (inlined by) xPortEnterCriticalTimeout at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/port.c:288
  #4  0x4008e459:0x3ffb2430 in vPortEnterCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/include/freertos/portmacro.h:578
      (inlined by) xEventGroupClearBits at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/event_groups.c:535
  #5  0x400d4db2:0x3ffb2450 in WiFiSTAClass::status() at C:/Users/Jem/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiSTA.cpp:149
  #6  0x400e4927:0x3ffb2470 in ESP32_TCP_Client::networkReady() at .pio/libdeps/ttgo-lora32-v1/ESP Mail Client/src/wcs/esp32/ESP32_TCP_Client.cpp:176
  #7  0x400d8caf:0x3ffb2490 in ESP_Mail_Client::reconnect(SMTPSession*, unsigned long) at .pio/libdeps/ttgo-lora32-v1/ESP Mail Client/src/ESP_Mail_SMTP.h:3006
      (inlined by) ESP_Mail_Client::reconnect(SMTPSession*, unsigned long) at .pio/libdeps/ttgo-lora32-v1/ESP Mail Client/src/ESP_Mail_SMTP.h:2999
  #8  0x400dbfc9:0x3ffb24b0 in ESP_Mail_Client::handleSMTPResponse(SMTPSession*, esp_mail_smtp_command, esp_mail_smtp_status_code, int) at .pio/libdeps/ttgo-lora32-v1/ESP Mail Client/src/ESP_Mail_SMTP.h:2743
  #9  0x400dc592:0x3ffb2550 in ESP_Mail_Client::smtpAuth(SMTPSession*, bool&) at .pio/libdeps/ttgo-lora32-v1/ESP Mail Client/src/ESP_Mail_SMTP.h:123
  #10 0x400dfda6:0x3ffb25b0 in SMTPSession::connect(esp_mail_session_config_t*, bool) at .pio/libdeps/ttgo-lora32-v1/ESP Mail Client/src/ESP_Mail_SMTP.h:3344
  #11 0x400d384d:0x3ffb25d0 in setup() at src/main.cpp:214
  #12 0x400e741e:0x3ffb2820 in loopTask(void*) at C:/Users/Jem/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42   

ELF file SHA256: 0000000000000000

[... 274 lines cut ...]
      // In case, ESP32, ESP8266 and SAMD device, the timestamp get from result.timestamp should be valid if
      // your device time was synched with NTP server.
      // Other devices may show invalid timestamp as the device time was not set i.e. it will show Jan 1, 1970.
      // You can call smtp.setSystemTime(xxx) to set device time manually. Where xxx is timestamp (seconds since Jan 1, 1970)
      
      ESP_MAIL_PRINTF("Message No: %d\n", i + 1);
      ESP_MAIL_PRINTF("Status: %s\n", result.completed ? "success" : "failed");
      ESP_MAIL_PRINTF("Date/Time: %s\n", MailClient.Time.getDateTimeString(result.timestamp, "%B %d, %Y %H:%M:%S").c_str());
      ESP_MAIL_PRINTF("Recipient: %s\n", result.recipients.c_str());
      ESP_MAIL_PRINTF("Subject: %s\n", result.subject.c_str());
    }
    Serial.println("----------------\n");

    // You need to clear sending result as the memory usage will grow up.
    smtp.sendingResult.clear();
  }
}

void loop(){}
```

**Additional information and things you've tried:**
I've tried different ports.
I've checked that I can ping the SMTP server
I've tried   /** Or connect without log in and log in later */

mobizt, 2023-05-20:
This is not library issue, but your information you provided was rejected by server.
`< S: 501 EHLO requires valid address.`

This is not valid host and your mail server was rejected the connection.
`myprofessionalmail.com`

closed 2023-05-20 by mobizt

PureTek-Innovations, 2023-05-20:
Thank you very much for your help.
It is strange that a failed login causes the ESP32 to crash though.

mobizt, 2023-05-20:
Yes, I agree for strange assert error, and it is in ESP32 core library `WiFi.h` when the library is trying to call WiFi.status(). 
`#5 0x400d4db2:0x3ffb2450 in WiFiSTAClass::status() at C:/Users/Jem/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiSTA.cpp:149`

And that error is outside of the library scope. You should set the debug option for WiFi in Arduino IDE.

PureTek-Innovations, 2023-05-20:
The loopback address "127.0.0.1" works, thank you very much.

I've added the ESP32Ping library so that the code only tries to connect and send an email if a ping to the SMTP server is successful.

When my system is in the field, it has a 4G internet connection and these are not 100% reliable.

Hopefully, this will stop it from rebooting if there are any connection issues.
It would be good if the spinlock error could be caught in the library though.

mobizt, 2023-05-20:
It's not possible to detect the error that does not rise from library but in the core.