mobizt/ESP-Mail-Client #267: E: [AUTHENTICATIONFAILED] Invalid credentials (Failure)

heyzzk, 2023-06-17:
Hi there,

**I just used the following code, but found the error "E: [AUTHENTICATIONFAILED] Invalid credentials (Failure)".**
https://github.com/mobizt/ESP-Mail-Client/blob/master/examples/IMAP/Mailbox_Changes_Notification/Mailbox_Changes_Notification.ino

**I am sure my gmail account and password are correct. But don't know why. The whole log is as below. Could anyone give me some tips, thanks.**
18:58:54.952 -> > C: connecting to IMAP server
18:58:54.952 -> > C: Host > imap.gmail.com
18:58:54.952 -> > C: Port > 993
18:58:54.983 -> 
18:58:54.983 -> > C: SSL/TLS negotiation
18:58:54.983 -> > C: seeding the random number generator
18:58:54.984 -> > C: setting up the SSL/TLS structure
18:58:54.984 -> ! W: Skipping SSL Verification. INSECURE!
18:58:54.984 -> > C: setting hostname for TLS session
18:58:54.984 -> > C: perform the SSL/TLS handshake
18:58:56.635 -> > C: verifying peer X.509 certificate
18:58:56.635 -> 
18:58:56.635 -> #### IMAP server connected
18:58:56.635 -> > C: IMAP server connected
18:58:56.635 -> 
18:58:56.635 -> #### Checking the capability...
18:58:56.635 -> > C: check the capability
18:58:56.667 -> 
18:58:56.667 -> #### Logging in...
18:58:56.667 -> > C: send IMAP command, AUTHENTICATE PLAIN
18:58:56.971 -> #### Error, [AUTHENTICATIONFAILED] Invalid credentials (Failure)

mobizt, 2023-06-17:
Please post in discussions topic. 
It's not library issue but your log in credentials is not valid instead.

closed 2023-06-17 by mobizt

mobizt, 2023-06-17:
https://github.com/mobizt/ESP-Mail-Client/blob/21962ab69ab583abe54bc286d9b1b3fe56536ceb/examples/IMAP/Mailbox_Changes_Notification/Mailbox_Changes_Notification.ino#L57-L69