mobizt/ESP-Mail-Client #269: Problem with WiFi

Patriboom, 2023-07-03:
**Build tool used:**

- [* ] Arudino
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based
- [ *] ESP32-based  (ESP32-cam - AI-Thinker)

**Other Libraries That are used:**
- include "EloquentSurveillance.h"
- include <ESP_Mail_Client.h>
- include <FS.h>
- include <SD_MMC.h>
- include "soc/soc.h"             // disable brownout problems
- include "soc/rtc_cntl_reg.h"    // disable brownout problems

**Description of problem:**

**Share code snippet to reproduce the issue:**
I've touched nothing on Wifi or any other library of that familiy.

```yaml
/**
 * Save motion capture to SD filesysten
 * @author  EloquentArduino <support@eloquentarduino.com>
  https://eloquentarduino.com/esp32-cam-motion-detection/   --- projet 4 ou 6
 */
#include "EloquentSurveillance.h"
#include <ESP_Mail_Client.h>
#include <FS.h>
#include <SD_MMC.h>
#include "soc/soc.h"             // disable brownout problems
#include "soc/rtc_cntl_reg.h"    // disable brownout problems
//#include "camera_pins.h"

//Instantiation de l'objet « détection de mouvement »
SMTPSession smtp;
EloquentSurveillance::Motion mvt;
String fichierPhoto = "capture_1.jpg";
String fichierPhoto_PATH = "/capture_1.jpg";
int flashMAX = 255;
int flashMIN = 10;
int flashETA = 0;
bool avertit = true;
// Déclaration de la classe nécessaire aux messages
SMTP_Message courriel;

bool        Affichons = true;               //Question d'économie d'énergie, nous coupons l'affichage et les calculs lors d'inactivité
const int   intervalAffi  =  3500;          // interval d'affichage
const int   intervalLect  =  15000;          // interval de lecteur des capteurs
const int   intervalCarte = 59876;          // interval de transmission des données vers meteo.cartefoi.net
const int   intervalOLED  =300000;          // interval de lecteur des capteurs ( 5 * 60 * 10 ) - 5 minutes
const int   intervalLumEx =180000;          // interval d'allumage de l'ampoule extérieure  ( 3 * 60 * 1000 ) - 3 minutes
volatile unsigned long millisActu = millis();        // Mise à jour des données sur cartefoi.net
volatile unsigned long precMillis = 0;               // Précédente lecture des valeurs météorologiques
volatile unsigned long precCartef = millisActu;      // Mise à jour des données sur cartefoi.net
volatile unsigned long precLUME = millisActu;        // Milliseconde au moment d'allumer la lumière extérieure
volatile unsigned long precOLED = millisActu;        // Redémarrage de l'écran OLED qui pourrait être un peu mêlé après une visite www
volatile unsigned long precAffich = 0;               // Précédent affichage sur écran OLED

//void streamJpg(RequeteWeb *laRequete) {
//  Serial.println("Nous sommes ici dans une ligne inutile");
//}

void setup() {
  Serial.begin(115200);
  delay(3000);

  //Définitions utiles à la caractérisation de la caméra et de sa sensibilité aux changements
  WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0); //disable brownout detector
  camera.aithinker();
  camera.vga();
  camera.highQuality();
  //25 juin 2023: Minchange = 0,25, MinPixelDiff = 0,10,  minSizeDiff = 0,10    ----> j'espère que ça fonctionne mieux le soir, période pendant laquelle les différences sont plutôt des nuances
  mvt.setMinChanges(0.25);          // taux (%) des pixels ayant changé de valeur valeur de 0 à 1
  mvt.setMinPixelDiff(10);          // ratio de changement des pixels changeant   valeur de 0 à 255
  mvt.setMinSizeDiff(0.10);         // variation de la taille du fichier          valeur de 0 à 1
  mvt.setFlashON(flashMAX);         //Intensité du flash au moment de saisir une image
  mvt.setFlashOFF(flashMAX);        //Intensité du DEL entre les images (au repos).
  mvt.debounce(10000L);

  //Initialisation de la caméra
  while (!camera.begin())
      debug("ERROR", camera.getErrorMessage());
  Serial.println("Dëmarrage de la caméra réussi");

  if (!SD_MMC.begin("/sdcard", true)) {
    Serial.println("Erreur au montage de lecteur de cartes microSD");
    ESP.restart();
  } else {
    delay(500);
    Serial.println("Lecteur de cartes microSD monté correctement.");
  }
  uint8_t cardType = SD_MMC.cardType();
  if (cardType == CARD_NONE){
    Serial.println("Veuillez insérer une carte microSD");
    return;
  }

  //Connexion Wifi
  WiFi.begin("SSID", "PassWord");
  Serial.print("Connexion WiFi en cours ...");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.print("\nVotre adresse IP: http://");
  Serial.println(WiFi.localIP());

  Serial.println("Nous sommes prêts");
}

```

**Additional information and things you've tried:**

I'm used to use this library.  I installed the latest update this morning (one day after it has been issued). Everything was good, but now: 

```
/home/patrick/.arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/WiFi/src/WiFiClient.h:88:9: note: candidate: int WiFiClient::setSocketOption(int, char*, size_t)
     int setSocketOption(int option, char* value, size_t len);
         ^

/home/patrick/.arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/WiFi/src/WiFiClient.h:88:9: note:   candidate expects 3 arguments, 4 provided

```
The WifiClient.h file is the "2021-03-26" version.  So nothing wrong with it from the update, beside libraries/ESP_Mail_Client/src/ where all file a from 2023-07-02

The most important part, I think is: 
**candidate expects 3 arguments, 4 provided**

Thanks.

Patriboom, 2023-07-03:
Up's version:  3.1.11
Back to 3.1.10, changing nothing else; my script compiles good.

mobizt, 2023-07-03:
If possible, please update your ESP32 core.

I know and understand this if you want to stick with outdated ESP32 core.

mobizt, 2023-07-04:
Now [v3.1.13](https://github.com/mobizt/ESP-Mail-Client/releases/tag/v3.1.13) is available for error fixes, please update.

closed 2023-07-04 by mobizt

Patriboom, 2023-07-04:
Manually installed 3.1.13 gives no error back.
Thanks.