mobizt/ESP-Mail-Client #270: imap.connect() encounters error (mbedtls) and consumes heap at (re)connect

dkalliv, 2023-07-06:
**Build tool used:**

- [x] Arudino
- [x] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based
- [x] ESP32-based

**Other Libraries That are used:**

**Description of problem:**
I noticed, that sometimes my heap is significantly lower either right after starting a program or when the imap server is reconnected (for example after a WiFi disconnect). After some research I noticed, that sometimes the imap.connect() reports the error "_esp32_ssl_handle_error(): [send_ssl_data():545]: (-27136) SSL - A buffer is too small to receive or write a message" (I guess this is from mbedtls) and the retries. Every time this happens I loose roughly 34k of heap memory. In a weak WiFi network the heap is quickly depleted. 
The code below does just open a imap connection and reconnects, after a network interruption. After a few reconnects the issue appears, sometimes already with the initial connection. Do you have any idea how to prevent this or how to release the memory occupied?
**Share code snippet to reproduce the issue:**
```yaml
#include <Arduino.h>
#include <WiFi.h>
#include <ESP_Mail_Client.h>
#include <extras/SDHelper.h>

#define WIFI_SSID "*********"
#define WIFI_PASSWORD "********"

#define IMAP_HOST "imapserver"
#define IMAP_PORT 993

#define AUTHOR_EMAIL "****@*********"
#define AUTHOR_PASSWORD "*********"

IMAPSession imap;

unsigned long readMillis = 0;
int totalMessage = 0;
int msgNum = 0;
int sign = -1;

Session_Config config;

IMAP_Data imap_data;

#if defined(ARDUINO_RASPBERRY_PI_PICO_W)
WiFiMulti multi;
#endif

// For Free Heap checking
#include "HeapStat.h"
HeapStat heapInfo;

void setup()
{
    Serial.begin(115200);
    Serial.println();
    WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
    Serial.print("Connecting to Wi-Fi");
    while (WiFi.status() != WL_CONNECTED)
    {
        Serial.print(".");
        delay(300);

    }
    Serial.println();
    Serial.print("Connected with IP: ");
    Serial.println(WiFi.localIP());
    Serial.println();
    MailClient.networkReconnect(true);
    imap.debug(1);
[... 197 lines cut ...]
> C: verifying peer X.509 certificate
[ 50222][V][ESP32_SSL_Client.cpp:456] connect_ssl(): Verifying peer X.509 certificate...
[ 50222][V][ESP32_SSL_Client.cpp:472] connect_ssl(): Certificate verified.
[ 50229][V][ESP32_SSL_Client.cpp:491] connect_ssl(): Free internal heap after TLS 172160

> C: IMAP server connected

> C: check the capability
[ 50248][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 16 bytes...
[ 50252][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 2 bytes...

> C: send IMAP command, AUTHENTICATE PLAIN
[ 50760][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 73 bytes...
[ 50762][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 2 bytes...

> C: send IMAP command, ID
[ 51349][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 54 bytes...
[ 51351][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 2 bytes...

> C: selecting the INBOX folder...
[ 51975][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 21 bytes...
[ 51978][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 2 bytes...
#### Heap Info
#### Current: 171492, Min: 171492, Max: 205736, Diff_1: -34104, Diff_5: -34244
```

mobizt, 2023-07-07:
This is not the issue about heap consumption which the SSL engine library i.e. mbedTLS requires approx. 50k for working with SSL during session opening. 

All resource was released after session was closed and no leaks and heap fragmentation guaranteed.

The internal SSL client used in this library is the modified and based on the `WiFiClientSecure` library which works in ESP32 Core that states [here](https://github.com/mobizt/ESP-Mail-Client#esp32-and-esp8266-sdks).

The only case that causes the error `"(-27136) SSL - A buffer is too small to receive or write a message"` is the incompatibility of mbedTLS in your outdated ESP32 Core and the SSL client implementation.

If you confirm that you are using the latest ESP32 core and still get the issue, the device malfunction was suspected because of unstable power and noise in device bus due to poor pcb quality.

After intensively testing for years, I never seen this issue in the recent ESP32 core unless in nonsupport version.

Please update your ESP32 core, provided better power supply (using short power cable or USB cable) or change your device if possible.

mobizt, 2023-07-07:
Ok I see some strange issue after network disconnected and trying to reconnect.

mobizt, 2023-07-07:
Now this issue was fixed in [v3.1.14](https://github.com/mobizt/ESP-Mail-Client/releases/tag/v3.1.14).

Thanks for reporting this.

closed 2023-07-07 by mobizt

mobizt, 2023-07-08:
You said true, after intensively test by moving router away from device.

This happens only in case MailClient.readMail or MailClient.sendMail was not called to handle network reconnection and free resources.

The resources are not released properly when calling `imap.connect  ` repeatedly.

Now the issue was fixed in [v3.1.15](https://github.com/mobizt/ESP-Mail-Client/releases/tag/v3.1.15), please update.

Unlike BearSSL used in ESP8266, mbedTLS data and SSL buffer cannot change without modifying the sources and recompiling the core library.

The SSL buffer should not change as IMAP server requires 16k buffer to work because it not support fragmentation.

That's why we need to increase the in buffer to 16k for working with IMAP.

dkalliv, 2023-07-08:
Thank you, this now works perfectly! Thank you as well for your explanation, I now understand, what the issue was and that the buffer in libmbedtls is already set to the maximum of 16k...
Keep up the great work!