mobizt/ESP-Mail-Client #271: Can not read email in a gmail account using example code Read_Single_Email.ino

koolwood, 2023-07-10:
**Build tool used:**

- [x ] Arudino
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ x] ESP8266-based
- [ ] ESP32-based

**Other Libraries That are used:**

Using -arduino ide 2.1
esp mail client 3.1.15
Can not read email in a gmail account using example code Read_Single_Email.ino
I tried other examples that read emails and they did not work either.

Not sure what I am doing wrong.  The only thing I changed was adding the login information.

Any thoughts?

Thanks
Ted

**Share code snippet to reproduce the issue:**
```yaml
PASTE .cpp / .ino code here
/**
 * This example shows how to read Email and store the message in SD card.
 *
 * Email: suwatchai@outlook.com
 *
 * Github: https://github.com/mobizt/ESP-Mail-Client
 *
 * Copyright (c) 2023 mobizt
 *
 */

/** ////////////////////////////////////////////////
 *  Struct data names changed from v2.x.x to v3.x.x
 *  ////////////////////////////////////////////////
 *
 * "ESP_Mail_Session" changes to "Session_Config"
 * "IMAP_Config" changes to "IMAP_Data"
 *
 * Changes in the examples
 *
 * ESP_Mail_Session session;
 * to
 * Session_Config config;
 *
 * IMAP_Config config;
 * to
 * IMAP_Data imap_data;
 *
 */

/** Assign SD card type and FS used in src/ESP_Mail_FS.h and
 * change the config for that card interfaces in src/extras/SDHelper.h
 */

#include <Arduino.h>
#if defined(ESP32) || defined(ARDUINO_RASPBERRY_PI_PICO_W)
#include <WiFi.h>
#elif defined(ESP8266)
#include <ESP8266WiFi.h>
#else

[... 526 lines cut ...]
#### Get UID...
> C: get UID

#### UID is 472
> C: UID is 472
! E: SD Storage is not ready.

#### Open the mailbox folder...
> C: selecting the INBOX folder...

#### Fetch message 1, UID: 472
> C: send IMAP command, FETCH
> C: fetch message header

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (3):
epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x402728f5 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffff7b0 end: 3fffffd0 offset: 0150
3ffff900:  00000000 00000000 3ffe8c4c 4010079a
```

mobizt, 2023-07-10:
It's because your device memory is running out. 

Please read and follow [this](https://github.com/mobizt/ESP-Mail-Client#memory-options-for-esp8266) first.

closed 2023-07-10 by mobizt

koolwood, 2023-07-10:
That fixed it.

Thank You Very Much
Ted