mobizt/ESP-Mail-Client #272: RAM not released ?

dneault333, 2023-07-14:
**Build tool used:**

- [x ] Arudino
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based
- [x ] ESP32-based

**Other Libraries That are used:**

**Description of problem:**

Resources not freeing, using Send_Text.ino example
Any way to release RAM used, or is this normal behavior ?
242748 down to 197160, and never coming back.
Note it stays at 197160, on repeated send's, so memory leak appears fixed to me :)

Heap : 242748

#### Connecting to SMTP server...
> C: ESP Mail Client v3.1.15
#### Error, library or device time was not set, see examples/SMTP/Set_Time.ino for manually time setting
! E: library or device time was not set, see examples/SMTP/Set_Time.ino for manually time setting
> C: connecting to SMTP server
> C: Host > smtp.gmail.com
> C: Port > 587

#### SMTP server connected
> C: SMTP server connected, wait for greeting...
< S: 220 smtp.gmail.com ESMTP d1-20020a0cdb01000000b0063659410b04sm3809768qvk.107 - gsmtp

#### Sending greeting response...
> C: send SMTP command, EHLO
< S: 250-smtp.gmail.com at your service, [99.226.111.9]
< S: 250-SIZE 35882577
< S: 250-8BITMIME
< S: 250-STARTTLS
< S: 250-ENHANCEDSTATUSCODES
< S: 250-PIPELINING
< S: 250-CHUNKING
< S: 250 SMTPUTF8

#### Sending STARTTLS command...
> C: send command, STARTTLS
< S: 220 2.0.0 Ready to start TLS
> C: perform SSL/TLS handshake

> C: SSL/TLS negotiation
> C: seeding the random number generator
> C: setting up the SSL/TLS structure
! W: Skipping SSL Verification. INSECURE!
> C: setting hostname for TLS session
> C: perform the SSL/TLS handshake
> C: verifying peer X.509 certificate

#### Sending greeting response...
> C: send SMTP command, EHLO
< S: 250-smtp.gmail.com at your service, [99.226.111.9]
< S: 250-SIZE 35882577
< S: 250-8BITMIME
< S: 250-AUTH LOGIN PLAIN XOAUTH2 PLAIN-CLIENTTOKEN OAUTHBEARER XOAUTH
< S: 250-ENHANCEDSTATUSCODES
< S: 250-PIPELINING
< S: 250-CHUNKING
< S: 250 SMTPUTF8

#### Logging in...
> C: send SMTP command, AUTH PLAIN
> C: xxxx@gmail.com
> C: ****************
< S: 235 2.7.0 Accepted

Successfully logged in.

#### Sending Email...
> C: send Email

#### Sending message header...
> C: send message header
< S: 250 2.1.0 OK d1-20020a0cdb01000000b0063659410b04sm3809768qvk.107 - gsmtp
< S: 250 2.1.5 OK d1-20020a0cdb01000000b0063659410b04sm3809768qvk.107 - gsmtp

#### Sending message body...
> C: send message body
< S: 354  Go ahead d1-20020a0cdb01000000b0063659410b04sm3809768qvk.107 - gsmtp

#### Finishing the message sending...
> C: finishing the message sending
< S: 250 2.0.0 OK  1689332975 d1-20020a0cdb01000000b0063659410b04sm3809768qvk.107 - gsmtp

#### Closing the session...
> C: terminate the SMTP session
< S: 221 2.0.

#### Message sent successfully
> C: message sent successfully

----------------
Message sent success: 1
Message sent failed: 0
----------------

Message No: 1
Status: success
Date/Time: January 01, 1970 00:00:09
Recipient: xxx@gmail.com
Subject: Test sending plain text Email
----------------

Heap : 197160

**Share code snippet to reproduce the issue:**
```yaml
    Serial.println("Heap : " + String(MailClient.getFreeHeap()));  //This is only thing I added to example

```

**Additional information and things you've tried:**
Moved it out of setup to separate function, sampled RAM at all stages, always 197160 after first send.

mobizt, 2023-07-14:
This is not an issue.

Did you ever try to use WiFi function in your code before?

Library release all used memory after session was closed and clear the sending result.

If you want to free all memory that WiFi used, disable WiFi.

closed 2023-07-14 by mobizt

dneault333, 2023-07-14:
Yes, I have seen that RAM hit using WiFi, many times, this is additional.
Noticed it in a bigger project, so tried it alone on ESP32 dev module.

If you says it's not an issue, then it ain't, many thanks

mobizt, 2023-07-14:
Sorry for jumping to conclusions. That is my fault.

This issue is now fixed in [v3.1.16](https://github.com/mobizt/ESP-Mail-Client/releases/tag/v3.1.16), please update.