mobizt/ESP-Mail-Client #273: Library crashes when WiFi connection is lost while fetching messages

dkalliv, 2023-07-14:
**Build tool used:**

- [x] Arudino
- [x] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based
- [x] ESP32-based

**Other Libraries That are used:**

**Description of problem:**
As it is possible that my project will be used with weak WiFi connections, I am conducting stability tests by deliberately interrupting the WiFi connection at various execution stages. I noticed, that the application will crash, when the connection is interrupted while fetching messages. I used the Search_And_Read.ino example (only added my WiFi and IMAP credentials) to reproduce the issue. While my application will recover after the reboot, it would be nice, if the library would catch the exception and fail gracefully. I've included the serial log and backtrace below:

**Share code snippet to reproduce the issue:**
```yaml
/**
 * This example shows how to search the unread messages and read them.
 *
 * To get the incoming mail notification, see Mailbox_Changes_Notification.ino.
 *
 * Created by K. Suwatchai (Mobizt)
 *
 * Email: suwatchai@outlook.com
 *
 * Github: https://github.com/mobizt/ESP-Mail-Client
 *
 * Copyright (c) 2023 mobizt
 *
 */

/**
 * To use library in silent mode (no debug printing and callback), please define this macro in src/ESP_Mail_FS.h.
 * #define SILENT_MODE
 */

#include <Arduino.h>
#if defined(ESP32) || defined(ARDUINO_RASPBERRY_PI_PICO_W)
#include <WiFi.h>
#elif defined(ESP8266)
#include <ESP8266WiFi.h>
#else

#endif

#include <ESP_Mail_Client.h>

#include <extras/SDHelper.h>

#define WIFI_SSID "<ssid>"
#define WIFI_PASSWORD "<password>"

#define IMAP_HOST "<host>"

#define IMAP_PORT 993

#define AUTHOR_EMAIL "<email>"
#define AUTHOR_PASSWORD "<password>"

void imapCallback(IMAP_Status status);

void printAllMailboxesInfo(IMAPSession &imap);

void printSelectedMailboxInfo(SelectedFolderInfo sFolder);

void printMessageData();

[... 378 lines cut ...]
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d2dea  PS      : 0x00060330  A0      : 0x800d2e3f  A1      : 0x3ffb1d00  
A2      : 0x000000dc  A3      : 0x00000014  A4      : 0x00000000  A5      : 0x3ffd99d0  
A6      : 0x3f41dfd1  A7      : 0x00000000  A8      : 0x00000012  A9      : 0x3ffb1cd0  
A10     : 0x3ffb1d18  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000017  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x000000e0  LBEG    : 0x4008a6a1  LEND    : 0x4008a6b1  LCOUNT  : 0xfffffffb  

Backtrace: 0x400d2de7:0x3ffb1d00 0x400d2e3c:0x3ffb1d20 0x400d2e72:0x3ffb1d40 0x400dc115:0x3ffb1d60 0x400e31c2:0x3ffb1da0 0x400e5f92:0x3ffb20d0 0x400d3c69:0x3ffb21a0 0x400ee206:0x3ffb2290
```
and here is the decoded backtrace:

```yaml
Decoding stack results
0x400d2de7: MB_String::_reserve(unsigned int, bool) at /home/dkal/Arduino/libraries/ESP_Mail_Client/src/extras/MB_String.h line 1785
0x400d2e3c: MB_String::copy(char const*, unsigned int) at /home/dkal/Arduino/libraries/ESP_Mail_Client/src/extras/MB_String.h line 1766
0x400d2e72: MB_String::operator=(char const*) at /home/dkal/Arduino/libraries/ESP_Mail_Client/src/extras/MB_String.h line 972
0x400dc115: ESP_Mail_Client::reconnect(IMAPSession*, unsigned long, bool) at /home/dkal/Arduino/libraries/ESP_Mail_Client/src/ESP_Mail_IMAP.h line 2641
0x400e31c2: ESP_Mail_Client::handleIMAPResponse(IMAPSession*, int, bool) at /home/dkal/Arduino/libraries/ESP_Mail_Client/src/ESP_Mail_IMAP.h line 2731
0x400e5f92: ESP_Mail_Client::readMail(IMAPSession*, bool) at /home/dkal/Arduino/libraries/ESP_Mail_Client/src/ESP_Mail_IMAP.h line 683
0x400d3c69: setup() at /home/dkal/Arduino/Search_And_Read_test/Search_And_Read_test.ino line 182
0x400ee206: loopTask(void*) at /home/dkal/.arduino15/packages/esp32/hardware/esp32/2.0.9/cores/esp32/main.cpp line 42
```

mobizt, 2023-07-15:
Thanks for reporting the issue. 

Now this issue was fixed in [v3.2.0](https://github.com/mobizt/ESP-Mail-Client/releases/tag/v3.2.0).

Please update the library.

closed 2023-07-15 by mobizt

dkalliv, 2023-07-16:
Thank you for fixing this, works very well!