mobizt/ESP-Mail-Client #275: Gmail Header Data not stored reliably in IMAP_Data

dkalliv, 2023-07-18:
**Build tool used:**

- [x] Arudino
- [x] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based
- [x] ESP32-based

**Other Libraries That are used:**

**Description of problem:**
For my current application I only need the message headers, so I do not fetch the message again after performing a search. As part of my testing I regularly use different E-Mail services. When testing with Gmail I noticed, that the subject field is empty (in IMAP_Data) for some messages, even though the debug log reports them as being fetched correctly (see attached log file). The affected messages and the number of affected messages changes randomly (sometimes the headers for all messages are stored correctly). I've used the attached test code with three other E-Mail services and only Gmail seems to be affected.

**Share code snippet to reproduce the issue:**
```yaml
#include <Arduino.h>
#include <WiFi.h>
#include <ESP_Mail_Client.h>

#define WIFI_SSID "##########"
#define WIFI_PASSWORD "###########"

#define IMAP_HOST "##########"
#define IMAP_PORT 993
#define AUTHOR_EMAIL "##############"
#define AUTHOR_PASSWORD "##############"

void imapCallback(IMAP_Status status);

void printAllMailboxesInfo(IMAPSession &imap);

void printSelectedMailboxInfo(SelectedFolderInfo sFolder);

void printMessageData();

IMAPSession imap;

Session_Config config;

IMAP_Data imap_data;

// Max messages in the search result
int max_result = 64;

void setup()
{

  Serial.begin(115200);

  Serial.println();

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);

  Serial.print("Connecting to Wi-Fi");

  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  imap.debug(333);
[... 962 lines cut ...]
UID: 3866	Subject: Test 12
UID: 3865	Subject: Test 11
UID: 3864	Subject: Test 10
UID: 3863	Subject: Test 9
UID: 3862 <------------ Missing Data!!!
UID: 3861	Subject: Test 7
UID: 3860	Subject: Test 6
UID: 3859	Subject: Test 5
UID: 3858	Subject: Test 4
UID: 3857	Subject: Test 3
UID: 3856	Subject: Test 2
UID: 3855	Subject: Test 1

#### Logging out...
> C: send IMAP command, LOGOUT
[ 20929][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 12 bytes...
[ 20932][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 2 bytes...
[ 21152][D][WiFiClient.cpp:546] connected(): Disconnected: RES: 0, ERR: 128

#### Log out completed
> C: log out completed
> C: cleaning SSL connection
[ 21163][V][ESP32_SSL_Client.cpp:503] stop_tcp_connection(): Cleaning SSL connection.
Free Heap: 238012
```

mobizt, 2023-07-18:
Please post your message source for uid 3869 and 3862.

dkalliv, 2023-07-19:
I've attached the requested message sources below, but please be aware, that I can run the same application with the same inbox content several times and uid 3869 and 3862 will process just fine, while other messages (or none) will show the problem...

UID 3862:
```yaml
Delivered-To: recipient@address
Received: by 2002:a59:9d8b:0:b0:3ed:4c69:3837 with SMTP id t11csp5139894vqv;
        Tue, 18 Jul 2023 12:20:19 -0700 (PDT)
X-Google-Smtp-Source: APBJJlHS3FEicZbrhxiqz5tRANQPf8fpPdKJEPxZ7SrQW7OPQR0+Mm9LVGfaWzZrP96L0sY1FBsp
X-Received: by 2002:a17:906:2dc3:b0:961:be96:b0e7 with SMTP id h3-20020a1709062dc300b00961be96b0e7mr650650eji.48.1689708019300;
        Tue, 18 Jul 2023 12:20:19 -0700 (PDT)
ARC-Seal: i=1; a=rsa-sha256; t=1689708019; cv=none;
        d=google.com; s=arc-20160816;
        b=cMg9oVZrNbgAA3RjIEN2zDkntI7b8X0sMrPOqCZwE6qT/2N6xSYdqlEhG2ZPkf/STZ
         sE23s5PYWyegt1F7m8dDEUXI/8eopJE+gZpfIVGTeH4va3LwnFwfI2DmHxplBuTFh7fL
         PaogE61C+z5v4aIZkyaN68IFghBH+C5Zhj3+hj9o0tNMfUj4URMN4y/P7Rn9HHrg3UY7
         nDEiuWxv/LCU4sR9RnVDP9IsZnHRbShA//n3vzrxtxgB7oW+R6IrzmWFEQzvSBVqo8QN
         DGSP6Bor9DHpXTBhDJ1HxyDd9zc/UzFeUpdeSEu12CcpOog5xOsGrNxD8ed5+LbIGZk6
         FfsQ==
ARC-Message-Signature: i=1; a=rsa-sha256; c=relaxed/relaxed; d=google.com; s=arc-20160816;
        h=content-transfer-encoding:subject:from:to:content-language
         :user-agent:mime-version:date:message-id;
        bh=47DEQpj8HBSa+/TImW+5JCeuQeRkm5NMpJWZG3hSuFU=;
        fh=unHbcYi4TNt9ggOOW11qRngAY3LzYoTDd75ZfoLlC4Y=;
        b=owz88PCEfm5+dUoygbNnEglpxOp4D4jaJwnfdnIlO57XrmsZeNKVhGC2y9pIYvT2CT
         80nzBKjEvvQPzphdpZTrIMnWAhma7i4U2x6Gmepn9O68upTtmw4BdnABWWWV21+Z7+V2
         FBUBecg0eOfEHeoBQgHt/0Lh84D9bfqMkWuM4RG6Dej7bs1s82TydlhjjeQBaskPxSBJ
         AKaQN9DJv82eNwll2YsNXsNfdoAo07zJikoyQjrxRpZqSTvS2G3Xj/pDpGwpiQNLXO7Y
         KhdEqexSQlPVjEwhcTotam4cAfviic3RlYtl+Ds+5xK9ZOsl8V8X3HlsfEzTFfRWZc6A
         CqMQ==
ARC-Authentication-Results: i=1; mx.google.com;
       spf=pass (google.com: domain of sender@address designates 212.25.24.36 as permitted sender) smtp.mailfrom=sender@address
Return-Path: <sender@address>
Received: from gozo.iway.ch (gozo.iway.ch. [212.25.24.36])
        by mx.google.com with ESMTPS id l18-20020a170906a41200b0099381745ba7si1605083ejz.878.2023.07.18.12.20.19
        for <recipient@address>
        (version=TLS1_2 cipher=ECDHE-RSA-AES128-GCM-SHA256 bits=128/128);
        Tue, 18 Jul 2023 12:20:19 -0700 (PDT)
Received-SPF: pass (google.com: domain of sender@address designates 212.25.24.36 as permitted sender) client-ip=212.25.24.36;
Authentication-Results: mx.google.com;
       spf=pass (google.com: domain of sender@address designates 212.25.24.36 as permitted sender) smtp.mailfrom=sender@address
Received: from gozo.iway.ch (localhost [127.0.0.1])
	by localhost (Postfix) with ESMTP id EF1963412E8
	for <recipient@address>; Tue, 18 Jul 2023 21:20:18 +0200 (CEST)
X-Iway-Path: 0
Received: from localhost (localhost [127.0.0.1])
	by localhost (ACF/28147.19247);
	Tue, 18 Jul 2023 21:20:18 +0200 (CEST)
Received: from switchplus-mail.ch (switchplus-mail.ch [212.25.8.236])
	(using TLSv1 with cipher ECDHE-RSA-AES256-SHA (256/256 bits))
	(No client certificate requested)
	by gozo.iway.ch (Postfix) with ESMTPS
	for <recipient@address>; Tue, 18 Jul 2023 21:20:18 +0200 (CEST)
Received: from [212.51.135.45] (account sender@address HELO [192.168.1.58])
  by swizzonic-mail.ch (CommuniGate Pro SMTP 7.1.0)
  with ESMTPSA id 275219216 for recipient@address; Tue, 18 Jul 2023 21:20:18 +0200
Message-ID: <bba3f2db-7a91-4658-fe7f-1df875f680e6@senderDomain>
Date: Tue, 18 Jul 2023 21:20:18 +0200
MIME-Version: 1.0
User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:102.0) Gecko/20100101
 Thunderbird/102.13.0
Content-Language: de-CH
To: recipient@address
From: "myName" <sender@address>
Subject: Test 8
Content-Type: text/plain; charset=UTF-8; format=flowed
Content-Transfer-Encoding: 7bit
```
UID 3869:
```yaml
Delivered-To: recipient@address
[... 41 lines cut ...]
X-iWay-Transport: alternative
X-Iway-Path: 1
Received: from localhost (localhost [127.0.0.1])
	by localhost (ACF/28147.21373);
	Tue, 18 Jul 2023 21:21:53 +0200 (CEST)
Received: from switchplus-mail.ch (switchplus-mail.ch [212.25.8.236])
	(using TLSv1 with cipher ECDHE-RSA-AES256-SHA (256/256 bits))
	(No client certificate requested)
	by gozo.iway.ch (Postfix) with ESMTPS
	for <recipient@address>; Tue, 18 Jul 2023 21:21:53 +0200 (CEST)
Received: from [212.51.135.45] (account sender@address HELO [192.168.1.58])
  by swizzonic-mail.ch (CommuniGate Pro SMTP 7.1.0)
  with ESMTPSA id 275219257 for recipient@address; Tue, 18 Jul 2023 21:21:53 +0200
Message-ID: <1d351c15-a9bf-5ac0-7cda-526d71b4a951@senderDomain>
Date: Tue, 18 Jul 2023 21:21:53 +0200
MIME-Version: 1.0
User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:102.0) Gecko/20100101
 Thunderbird/102.13.0
Content-Language: de-CH
To: recipient@address
From: "myName" <sender@address>
Subject: Test 15
Content-Type: text/plain; charset=UTF-8; format=flowed
Content-Transfer-Encoding: 7bit
```

mobizt, 2023-07-19:
All header fields are empty not only subject because of Gmail's server returns untagged multi-lines responses in some message fetching instead of single line untagged response which is very strange as the octets size of untagged response is the same i.e., 372 for all messages.

This causes no headers parsed from the subsequence responses. 

I will fix this and update soon.

mobizt, 2023-07-19:
Now this issue was fixed already in the [latest release](https://github.com/mobizt/ESP-Mail-Client/releases/tag/v3.2.1).

Please update.

dkalliv, 2023-07-21:
Thank you for this update. The issue seems to be fixed with Gmail, unfortunately the fix seems to cause problems wit other mail services. With version 3.2.2 and 3.2.1 I'm getting the following error (using the same code as above). 

```yaml
#### Search limit: 10
Found 79 messages
Show 10 messages

#### Fetch message 1, UID: 524
> C: send IMAP command, FETCH
> C: fetch message header
[  2735][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 249 bytes...
[  2739][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 2 bytes...

#### Get Flags...
> C: get Flags
[  2870][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 25 bytes...
[  2873][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 2 bytes...
#### Error, could not parse command
! E: could not parse command
> C: cleaning SSL connection
[  2933][V][ESP32_SSL_Client.cpp:503] stop_tcp_connection(): Cleaning SSL connection.
> C: Free Heap: 246596

#### Fetch message 2, UID: 523
> C: send IMAP command, FETCH
> C: fetch message header
#### Error, connection closed
! E: connection closed
```
When going back to version 3.2.0 I see, that the problem happens with responses that have a random line break (it seems like not only Gmail is doing this, but others do it differently):

```yaml
#### Get Flags...
> C: get Flags
[ 20413][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 22 bytes...
[ 20416][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 2 bytes...
* 53 FETCH (FLAGS (\Seen))
xmail OK Fetch completed (0.001 + 0.100 secs).
> C: Free Heap: 181664

#### Fetch message 28, UID: 497
> C: send IMAP command, FETCH
> C: fetch message header
[ 20805][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 249 bytes...
[ 20808][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 2 bytes...
* 52 FETCH (UID 497 MODSEQ (876) BODY[HEADER.FIELDS (FROM SENDER TO CC SU
BJECT DATE MESSAGE-ID RETURN-PATH REPLY-TO IN-REPLY-TO REFERENCES COMMENTS KEYWORDS BCC FLAGS CONTENT-TYPE CONTENT-TRANSFER-ENCODING CONTENT-LANGUAGE ACCEPT-LANGUAGE)] {283}
Return-Path: <sender@domain>
From: "senderName"<sender@domain>
To: "recipientName"<recipient@domain>
Subject: 0 Dritter grosser Koffer
Date: Fri, 07 Jul 2023 12:39:44 +0000
Message-ID: <auto-000274537787@mailServer>
)
xmail OK Fetch completed (0.001 + 0.213 secs).
```
Do you have any idea how to handle this? Please let me know if you need more information ...

mobizt, 2023-07-22:
I never get that such line break in fetch response in my test in various mail services many years since ESP32 Mail Client library and you are the first one who got this error, and it is very strange case.

This line break can be happened in untagged response which should be start with * instead of arbitrary line break like this.

mobizt, 2023-07-22:
Library reads the server response line by line or it is looking for line break.
The response like this can be timed out in response reading.
```yaml
* 52 FETCH (UID 497 MODSEQ (876) BODY[HEADER.FIELDS (FROM SENDER TO CC SU <------------- time out happen in library or who insert this line break?
BJECT DATE MESSAGE-ID RETURN-PATH REPLY-TO IN-REPLY-TO REFERENCES COMMENTS KEYWORDS BCC FLAGS CONTENT-TYPE CONTENT-TRANSFER-ENCODING CONTENT-LANGUAGE ACCEPT-LANGUAGE)] {283} <------------- line break at the end of untagged response.
```

I recommend you should connect your device directly through the internet ISP router without any gateway or via mobile WiFi hotspot.

If time out happened in library response reading, it would exit the function immediately and close the TCP session and release memory.

mobizt, 2023-07-22:
This line break is not issued by any mail servers as I always can get more octets length untagged response in just one line.

mobizt, 2023-07-22:
With no debug and callback set and inserting this code

```cpp
if (res.isUntaggedResponse && !res.untaggedRespCompleted)
   Serial.println("----------------- line break dectected -----------------");

Serial.println(res.response);
```

after this

https://github.com/mobizt/ESP-Mail-Client/blob/500bc26492e130bf3c79479f2cbf09b8389fa8a5/src/ESP_Mail_IMAP.h#L1808-L1811

### Here is the printing result of 240 messages from "SEARCH ALL" result in Gmail mailbox which supports condstore IMAP extension.

Note: The modseq number is 7 digits long and the untagged response length is 244 bytes (octets).

```yaml
Connecting to Wi-Fi.
Connected with IP: 192.168.137.9

Successfully logged in.

Available folders: INBOX, [Gmail], [Gmail]/All Mail, [Gmail]/Drafts, [Gmail]/Important, [Gmail]/Sent Mail, [Gmail]/Spam, [Gmail]/Starred, [Gmail]/Trash, test

Info of the selected folder
Total Messages: 8302
UID Validity: 1
Predicted next UID: 9815
Unseen Messages: No
Highest Modification Sequence: 1254052
Flags: \Answered, \Flagged, \Draft, \Deleted, \Seen, $Forwarded, $NotPhishing, $Phishing, Answered, Delete, Deleted, Flagged, NonJunk, Read, Seen
* 8302 FETCH (MODSEQ (1253990) BODY[HEADER.FIELDS (From Sender To Cc Subject Date Message-ID Return-Path Reply-To In-Reply-To References Comments Keywords Bcc Flags Content-Type Content-transfer-encoding Content-Language Accept-Language)] {486}
* 8301 FETCH (MODSEQ (1253905) BODY[HEADER.FIELDS (From Sender To Cc Subject Date Message-ID Return-Path Reply-To In-Reply-To References Comments Keywords Bcc Flags Content-Type Content-transfer-encoding Content-Language Accept-Language)] {441}
* 8300 FETCH (MODSEQ (1253783) BODY[HEADER.FIELDS (From Sender To Cc Subject Date Message-ID Return-Path Reply-To In-Reply-To References Comments Keywords Bcc Flags Content-Type Content-transfer-encoding Content-Language Accept-Language)] {370}
* 8299 FETCH (MODSEQ (1253605) BODY[HEADER.FIELDS (From Sender To Cc Subject Date Message-ID Return-Path Reply-To In-Reply-To References Comments Keywords Bcc Flags Content-Type Content-transfer-encoding Content-Language Accept-Language)] {419}
* 8298 FETCH (MODSEQ (1253641) BODY[HEADER.FIELDS (From Sender To Cc Subject Date Message-ID Return-Path Reply-To In-Reply-To References Comments Keywords Bcc Flags Content-Type Content-transfer-encoding Content-Language Accept-Language)] {420}
* 8297 FETCH (MODSEQ (1253435) BODY[HEADER.FIELDS (From Sender To Cc Subject Date Message-ID Return-Path Reply-To In-Reply-To References Comments Keywords Bcc Flags Content-Type Content-transfer-encoding Content-Language Accept-Language)] {457}
* 8296 FETCH (MODSEQ (1253391) BODY[HEADER.FIELDS (From Sender To Cc Subject Date Message-ID Return-Path Reply-To In-Reply-To References Comments Keywords Bcc Flags Content-Type Content-transfer-encoding Content-Language Accept-Language)] {420}
* 8295 FETCH (MODSEQ (1253167) BODY[HEADER.FIELDS (From Sender To Cc Subject Date Message-ID Return-Path Reply-To In-Reply-To References Comments Keywords Bcc Flags Content-Type Content-transfer-encoding Content-Language Accept-Language)] {739}
* 8294 FETCH (MODSEQ (1253245) BODY[HEADER.FIELDS (From Sender To Cc Subject Date Message-ID Return-Path Reply-To In-Reply-To References Comments Keywords Bcc Flags Content-Type Content-transfer-encoding Content-Language Accept-Language)] {420}
* 8293 FETCH (MODSEQ (1253017) BODY[HEADER.FIELDS (From Sender To Cc Subject Date Message-ID Return-Path Reply-To In-Reply-To References Comments Keywords Bcc Flags Content-Type Content-transfer-encoding Content-Language Accept-Language)] {571}
* 8292 FETCH (MODSEQ (1252663) BODY[HEADER.FIELDS (From Sender To Cc Subject Date Message-ID Return-Path Reply-To In-Reply-To References Comments Keywords Bcc Flags Content-Type Content-transfer-encoding Content-Language Accept-Language)] {473}
* 8291 FETCH (MODSEQ (1253341) BODY[HEADER.FIELDS (From Sender To Cc Subject Date Message-ID Return-Path Reply-To In-Reply-To References Comments Keywords Bcc Flags Content-Type Content-transfer-encoding Content-Language Accept-Language)] {372}
* 8290 FETCH (MODSEQ (1253095) BODY[HEADER.FIELDS (From Sender To Cc Subject Date Message-ID Return-Path Reply-To In-Reply-To References Comments Keywords Bcc Flags Content-Type Content-transfer-encoding Content-Language Accept-Language)] {444}
* 8289 FETCH (MODSEQ (1253094) BODY[HEADER.FIELDS (From Sender To Cc Subject Date Message-ID Return-Path Reply-To In-Reply-To References Comments Keywords Bcc Flags Content-Type Content-transfer-encoding Content-Language Accept-Language)] {602}
* 8288 FETCH (MODSEQ (1253093) BODY[HEADER.FIELDS (From Sender To Cc Subject Date Message-ID Return-Path Reply-To In-Reply-To References Comments Keywords Bcc Flags Content-Type Content-transfer-encoding Content-Language Accept-Language)] {481}
* 8287 FETCH (MODSEQ (1252161) BODY[HEADER.FIELDS (From Sender To Cc Subject Date Message-ID Return-Path Reply-To In-Reply-To References Comments Keywords Bcc Flags Content-Type Content-transfer-encoding Content-Language Accept-Language)] {406}
* 8286 FETCH (MODSEQ (1252053) BODY[HEADER.FIELDS (From Sender To Cc Subject Date Message-ID Return-Path Reply-To In-Reply-To References Comments Keywords Bcc Flags Content-Type Content-transfer-encoding Content-Language Accept-Language)] {448}
[... cut ...]
> [  2873][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 2 bytes...
> #### [...]

This is strange as the response of Fetch  command with flag parameter is very short like this and  it should be properly parsed unless the line break injection problem from middle proxy or gateway.
```yaml
* 22 FETCH (MODSEQ (501780) FLAGS (\Seen))  <------ untagged response 
xmail OK Success <------ tagged response
```

FYI

The debug value 333 is not supported any more. The full debug level can be higher value which you should look at the source.

mobizt, 2023-07-22:
The only issue I found is incorrect response parsing in IMAP `getFlags` function for server that supports condstore extension.

mobizt, 2023-07-22:
The getFlags issue is now fixed in [v3.2.3](https://github.com/mobizt/ESP-Mail-Client/releases/tag/v3.2.3).

dkalliv, 2023-07-22:
I am pretty confident, that the behavior I see is not related to a bad connection. I have a very fast Internet connection and the ESPs are not behind a proxy. Even the access point is configured as a bridge. Apart from this, I use a mobile phone hotspot to test stability in case of network interruptions. Using this Mail Provider (one of the larger ones in Switzerland) I see a consistent behavior across all tested network scenarios (see log). When I go back to Version 3.2.0 the 
```yaml
xmail BAD Error in IMAP command FETCH: Invalid messageset (0.001 + 0.253 + 0.253 secs).
#### Error, could not parse command
```
problem goes away, but I still have the problem with stray line breaks (as with Gmail). With the other Mail providers (tested with the same hardware in the same networks) I do not see this behavior.
If you don't mind, I will send you the credentials for the test account I used to conduct these tests. We did this the same way a few years ago and I will use the same address (hotmail) you provided me back then. I hope this is ok for you, as this goes way beyond my understanding of IMAP...
You can delete the test mails in the account or add new ones as you like. I will delete the account, when you are done.

```yaml
Connected with IP: 172.20.10.10

[  3191][V][ESP32_SSL_Client.cpp:503] stop_tcp_connection(): Cleaning SSL connection.
#### Connecting to IMAP server...
> C: ESP Mail Client v3.2.3
> C: connecting to IMAP server
> C: Host > mailServer
> C: Port > 993

> C: SSL/TLS negotiation
> C: seeding the random number generator
[  3854][V][ESP32_SSL_Client.cpp:201] connect_ssl(): Seeding the random number generator
> C: setting up the SSL/TLS structure
[  3868][V][ESP32_SSL_Client.cpp:220] connect_ssl(): Setting up the SSL/TLS structure...
! W: Skipping SSL Verification. INSECURE!
[  3880][I][ESP32_SSL_Client.cpp:244] connect_ssl(): WARNING: Skipping SSL Verification. INSECURE!
> C: setting hostname for TLS session
[  3891][V][ESP32_SSL_Client.cpp:387] connect_ssl(): Setting hostname for TLS session...
> C: perform the SSL/TLS handshake
[  3902][V][ESP32_SSL_Client.cpp:417] connect_ssl(): Performing the SSL/TLS handshake...
> C: verifying peer X.509 certificate
[  4828][V][ESP32_SSL_Client.cpp:456] connect_ssl(): Verifying peer X.509 certificate...
[  4829][V][ESP32_SSL_Client.cpp:472] connect_ssl(): Certificate verified.
[  4835][V][ESP32_SSL_Client.cpp:491] connect_ssl(): Free internal heap after TLS 167068
* OK [CAPABILITY IMAP4rev1 SASL-IR LOGIN-REFERRALS ID ENABLE IDLE LITERAL+ AUTH=PLAIN] Dovecot ready.

#### IMAP server connected
> C: IMAP server connected

#### Checking the capability...
> C: check the capability
xmail CAPABILITY
[  5502][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 16 bytes...
[  5511][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 2 bytes...
* CAPABILITY IMAP4rev1 SASL-IR LOGIN-REFERRALS ID ENABLE IDLE LITERAL+ AUTH=PLAIN
xmail OK Pre-login capabilities listed, post-login capabilities have more.

#### Logging in...
> C: send IMAP command, AUTHENTICATE PLAIN
xmail AUTHENTICATE PLAIN AHRlc3QxQGRrbWIuY2gAZGlmZmljdWx0cGFzc3dvcmQ=
[  6104][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 69 bytes...
[  6111][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 2 bytes...
* CAPABILITY IMAP4rev1 SASL-IR LOGIN-REFERRALS ID ENABLE IDLE SORT SORT=DISPLAY THREAD=REFERENCES THREAD=REFS THREAD=ORDEREDSUBJECT MULTIAPPEND URL-PARTIA
xmail OK Logged in

#### Send client identification...
> C: send IMAP command, ID
xmail ID ("name" "ESP Mail Client" "version" "3.2.3")
[  6698][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 53 bytes...
[  6702][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 2 bytes...
* ID ("name" "Dovecot")
xmail OK ID completed (0.001 + 0.303 + 0.302 secs).

Successfully logged in.
7075

#### Open the mailbox folder...
> C: selecting the INBOX folder...
xmail EXAMINE "INBOX" (CONDSTORE)
[  7087][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 33 bytes...
[... 55 lines cut ...]
#### Fetch message 2, Number: 4
> C: send IMAP command, FETCH
> C: fetch message header
#### Error, connection closed
! E: connection closed

#### Fetch message 3, Number: 3
> C: send IMAP command, FETCH
> C: fetch message header
#### Error, connection closed
! E: connection closed

#### Fetch message 4, Number: 2
> C: send IMAP command, FETCH
> C: fetch message header
#### Error, connection closed
! E: connection closed

#### Fetch message 5, Number: 1
> C: send IMAP command, FETCH
> C: fetch message header
#### Error, connection closed
! E: connection closed
Free Heap: 212840
```

mobizt, 2023-07-23:
I finally found the issue when setting debug level in IDE.

The build flags changed and revealed the issue.

The variable that stored the UID has used (read) before assigning the initial value and it contains arbitrary value when read that causes all subsequence tasks failure in the fix from v3.2.1. 

Now it fixed already in [latest release v3.2.4](https://github.com/mobizt/ESP-Mail-Client/releases/tag/v3.2.4), please update.

Now you can delete your test account.

Even the line break issue was fixed, I still never seen the line break in IMAP untagged response from the test for many years in many IMAP servers both free and paid services including my own IMAP server.

mobizt, 2023-07-23:
This is the issue when trying to send IMAP command to get UID.

> xmail FETCH 16696 UID

The message sequence number 16696 is not existed in your mailbox which this number is the arbitrary value of uint32_t that does not yet initialize before used.

dkalliv, 2023-07-23:
Thank you for taking the time to look at this. The update is a big improvement, but unfortunately this really seems to be a difficult one to fix. Initially I thought everything is working fine, but after testing with real mail data I encountered similar problems again. It took me some investigation to realize that the problem does not occur on messages sent from the same server (maybe the headers are not "complex" enough to cause the insertion of stray newlines). I reactivated the test account (with the same credentials I already sent you) and generated 64 test messages from another mail server. Now the issue can be reproduced with library v3.2.4 (feel free to use the test account as you need).

I made some interesting observations, which may help you pin down the issue:

Version 3.2.4 now nicely handles stray newlines that occur towards the end of the response:
```yaml
#### Fetch message 17, UID: 231
> C: send IMAP command, FETCH
> C: fetch message header
[ 15938][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 249 bytes...
[ 15941][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 2 bytes...
* 17 FETCH (UID 231 MODSEQ (332) BODY[HEADER.FIELDS (FROM SENDER TO CC SUBJECT DATE MESSAGE-ID RETURN-PATH REPLY-TO IN-REPLY-TO REFERENCES COMMENTS KEYWORDS BCC FLAGS CONT
ENT-TYPE CONTENT-TRANSFER-ENCODING CONTENT-LANGUAGE ACCEPT-LANGUAGE)] {289}
Return-Path: <sender@senderDomain>
From: sender@senderDomain
To: recipient@recipientDomain
Subject: This Text is to make the header subject field a bit longer to provoke potential issues Test Nr 16
Date: Sun, 23 Jul 2023 17:21:02 +0200 (CEST)
Message-Id: <E1qNatL-000FDS-Do@mailServer>
)
xmail OK Fetch completed (0.001 + 0.255 secs).
```
but fails, when the stray newline is positioned near the beginning of the response:
```yaml
#### Fetch message 48, UID: 262
> C: send IMAP command, FETCH
> C: fetch message header
[ 37248][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 249 bytes...
[ 37251][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 2 bytes...
* 48 
FETCH (UID 262 MODSEQ (386) BODY[HEADER.FIELDS (FROM SENDER TO CC SUBJECT DATE MESSAGE-ID RETURN-PATH REPLY-TO IN-REPLY-TO REFERENCES COMMENTS KEYWORDS BCC FLAGS CONTENT-TYPE CONTENT-TRANSFER-ENCODING CONTENT-LANGUAGE ACCEPT-LANGUAGE)] {289}
Return-Path: <sender@senderDomain>
From: sender@senderDomain
To: recipient@recipientDomain
Subject: This Text is to make the header subject field a bit longer to provoke potential issues Test Nr 39
Date: Sun, 23 Jul 2023 17:23:01 +0200 (CEST)
Message-Id: <E1qNaw9-000G1m-GZ@mailServer>
)
xmail OK Fetch completed (0.001 + 0.265 secs).
#### Error, Fetch completed (0.001 + 0.265 secs).
! E: Fetch completed (0.001 + 0.265 secs).
```
even worse, when there is a stray newline in the "xmail OK ..." part, there is a timeout, the connection is closed and all subsequent fetches fail
```yaml
#### Fetch message 4, UID: 218
> C: send IMAP command, FETCH
> C: fetch message header
[  7560][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 249 bytes...
[  7563][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 2 bytes...
* 4 FETCH (UID 218 MODSEQ (306) BODY[HEADER.FIELDS (FROM SENDER TO CC SUBJECT DATE MESSAGE-ID RETURN-PATH REPLY-TO IN-REPLY-TO REFERENCES COMMENTS KEYWORDS BCC FLAGS CONTENT-TYPE CONTENT-TRANSFER-ENCODING CONTENT-LANGUAGE ACCEPT-LANGUAGE)] {288}
Return-Path: <sender@senderDomain>
From: sender@senderDomain
To: recipient@recipientDomain
Subject: This Text is to make the header subject field a bit longer to provoke potential issues Test Nr 3
Date: Sun, 23 Jul 2023 17:19:55 +0200 (CEST)
Message-Id: <E1qNasG-0005KW-00@mailServer>
)
xmail OK Fetch completed (0.001 + 0.255 secs).

#### Get Flags...
> C: get Flags
[  7933][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 21 bytes...
[  7939][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 2 bytes...
* 4 FETCH (FLAGS (NonJunk))
x
mail OK Fetch completed (0.001 + 0.167 secs).
> C: cleaning SSL connection
[ 38172][V][ESP32_SSL_Client.cpp:503] stop_tcp_connection(): Cleaning SSL connection.
#### Error, session timed out
! E: session timed out
[... 23 lines cut ...]
#### Error, session timed out
! E: session timed out
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d2d1a  PS      : 0x00060330  A0      : 0x800d2d73  A1      : 0x3ffcba40  
A2      : 0x000000dc  A3      : 0x00000014  A4      : 0x00000000  A5      : 0x3ffd70c8  
A6      : 0x3f41d411  A7      : 0x00000000  A8      : 0x00000012  A9      : 0x3ffcba10  
A10     : 0x3ffcba58  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x000000e0  LBEG    : 0x4008a66d  LEND    : 0x4008a67d  LCOUNT  : 0xfffffffb  

Backtrace: 0x400d2d17:0x3ffcba40 0x400d2d70:0x3ffcba60 0x400d2da6:0x3ffcba80 0x400db58c:0x3ffcbaa0 0x400e2a66:0x3ffcbae0 0x400e53de:0x3ffcbe10 0x400d2fdf:0x3ffcbee0 0x400edf16:0x3ffcbf40
```
```yaml
Decoding stack results
0x400d2d17: MB_String::_reserve(unsigned int, bool) at /home/dkal/Arduino/libraries/ESP_Mail_Client/src/extras/MB_String.h line 1785
0x400d2d70: MB_String::copy(char const*, unsigned int) at /home/dkal/Arduino/libraries/ESP_Mail_Client/src/extras/MB_String.h line 1766
[... cut ...]
0x400e2a66: ESP_Mail_Client::handleIMAPResponse(IMAPSession*, int, bool) at /home/dkal/Arduino/libraries/ESP_Mail_Client/src/ESP_Mail_IMAP.h line 2789
0x400e53de: ESP_Mail_Client::readMail(IMAPSession*, bool) at /home/dkal/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/include/c++/8.4.0/bits/stl_vector.h line 805
0x400d2fdf: setup() at /home/dkal/Arduino/libraries/ESP_Mail_Client/src/extras/MB_String.h line 1183
0x400edf16: uartSetRxFIFOFull at /home/dkal/.arduino15/packages/esp32/hardware/esp32/2.0.10/cores/esp32/esp32-hal-uart.c line 269
```

mobizt, 2023-07-23:
There is something does not work properly on your side, may be software (ESP32 core 2.0.10 with debug built) or hardware.

For example, this server response like this is not valid and I'm pretty sure that mail server sends the complete response in single line instead of spit it to several lines like this.

```yaml
x
mail OK Fetch completed (0.001 + 0.167 secs).
```
Where xmail is the TAG or identifier belongs to the request.

This may be mbedTLS issue that may incorrectly decrypt the SSL data out.

Can you try the library with ESP8266 device?

I'm sure that all untagged response should be complete in single line per RFC standard documents, and I never see this issue on my side in all platforms.

dkalliv, 2023-07-23:
Unfortunately I do not have access to a ESP8266 at the moment, but you are correct, I was using ESP32 core 2.0.10 with debug build. However, I do not see the problems with other Mail Services, even when using ESP32 core 2.0.10 with debug build.
I anyway downgraded to 2.0.6 and built in release mode but I did see exactly the same problems apart from running faster and more free heap (which was expected).

I understand, that the server does not seem to comply with the relevant RFCs. On the other side neither is Gmail, which also inserts stray newlines in the untagged response. I assume, that there must be other providers out there doing the same thing. As I want my application to be compatible with as many mail services as possible, I am happy to work with you to get this working (not fixing your excellent library, but making it more tolerant to non compliant mail services).

Assuming this would be a short troubleshooting I used the Arduino IDE to build the test code. I now moved everything to PlatformIO and VS Code, which is my normal build environment.

The corruption with the newlines in the first half of the untagged response happens quite frequently. You should be able to reproduce it with one or two runs of the test code against the test account I provided you (at least if you use the test messages in the inbox).

The corruption in the tagged response is not so easy to reproduce. It took me about 15 runs of my test code (fetching the headers for 65 messages each time) to catch this exception:
```yaml
#### Fetch message 31, UID: 248
> C: send IMAP command, FETCH
> C: fetch message header
* 34 FETCH (UID 248 MODSEQ (363) BODY[HEADER.FIELDS (FROM SENDER TO CC SUBJECT DATE MESSAGE-ID RETURN-PATH REPLY-TO IN-REPLY-TO REFERENCES COMMENTS KEYWORDS BCC FLAGS CONTENT-TYPE CONTENT-TRANSFER-ENCODING CONTENT-LANGUAGE ACCEPT-LANGUAGE)] {289}
Return-Path: <sender@senderDomain>
From: sender@senderDomain
To: recipient@recipientDomain
Subject: This Text is to make the header subject field a bit longer to provoke potential issues Test Nr 33
Date: Sun, 23 Jul 2023 17:22:30 +0200 (CEST)
Message-Id: <E1qNaux-0006Eh-1W@mailServer>
)
xmail O
K Fetch completed (0.001 + 0.153 secs).
> C: cleaning SSL connection
#### Error, session timed out
! E: session timed out
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d2cfa  PS      : 0x00060330  A0      : 0x800d2d53  A1      : 0x3ffb1d90  
A2      : 0x000000dc  A3      : 0x00000014  A4      : 0x00000000  A5      : 0x3ffcdda0  
A6      : 0x3f419511  A7      : 0x00000000  A8      : 0x00000012  A9      : 0x3ffb1d60  
A10     : 0x3ffb1da8  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000014  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x000000e0  LBEG    : 0x4008a671  LEND    : 0x4008a681  LCOUNT  : 0xfffffffb  

Backtrace: 0x400d2cf7:0x3ffb1d90 0x400d2d50:0x3ffb1db0 0x400d2d86:0x3ffb1dd0 0x400d8df0:0x3ffb1df0 0x400e02ce:0x3ffb1e30 0x400e2c46:0x3ffb2160 0x400d300b:0x3ffb2230 0x400e6ac2:0x3ffb2290

  #0  0x400d2cf7:0x3ffb1d90 in MB_String::_reserve(unsigned int, bool) at .pio/libdeps/mhetesp32minikit/ESP Mail Client/src/extras/MB_String.h:1785
  #1  0x400d2d50:0x3ffb1db0 in MB_String::copy(char const*, unsigned int) at .pio/libdeps/mhetesp32minikit/ESP Mail Client/src/extras/MB_String.h:1766
  #2  0x400d2d86:0x3ffb1dd0 in MB_String::operator=(char const*) at .pio/libdeps/mhetesp32minikit/ESP Mail Client/src/extras/MB_String.h:972
  #3  0x400d8df0:0x3ffb1df0 in ESP_Mail_Client::reconnect(IMAPSession*, unsigned long, bool) at .pio/libdeps/mhetesp32minikit/ESP Mail Client/src/ESP_Mail_IMAP.h:2617
  #4  0x400e02ce:0x3ffb1e30 in ESP_Mail_Client::handleIMAPResponse(IMAPSession*, int, bool) at .pio/libdeps/mhetesp32minikit/ESP Mail Client/src/ESP_Mail_IMAP.h:2801
  #5  0x400e2c46:0x3ffb2160 in ESP_Mail_Client::readMail(IMAPSession*, bool) at .pio/libdeps/mhetesp32minikit/ESP Mail Client/src/ESP_Mail_IMAP.h:682
  #6  0x400d300b:0x3ffb2230 in setup() at src/main.cpp:113
  #7  0x400e6ac2:0x3ffb2290 in loopTask(void*) at /home/dkal/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42
```

mobizt, 2023-07-23:
This is not the issue about RFC compliance. Mail server should provide the basic IMAP services at least complies to RFC 3501.

We are not talk about the extension of IMAPv4 but why invalid and in expected response received in the application.

I test much enough for Gmail, Outlook, yahoo, AOL, and other Email services included free and paid, my corporate server and my own IMAP server on my server.

All are not produce sush strange issue in almost five years and no issue about this found in this repo and forum on internet, blog and YouTube ever before.

I test with your mail service with your test account and did not found any issue about this.

I iterate run through over 100,000 messages fetching test in my oldest email account both Gmail and Hotmail and I did not found this issue.

I still confirm that this issue should not exist and come from server side and library side if both refer to the standards.

To implement the fix for this issue is easy but it breaks the parsing rules as hard break and soft break have their meaning in this protocol.

The SSL buffer data is invalid in your case which I strongly confirm that it will never happen in standard IMAP server.

I test with PlatformIO, because it developed in it and it works fine.  The lastest PlatformIO ESP32 core is currently included ESP32 core v2.0.8 which is not the latest ESP32 Arduino release.

I cannot replicate your issue on my test.

Any way I will test with lastest ESP32 Arduino core with debug option and let you know.

dkalliv, 2023-07-24:
Can you confirm, that you do not see the issue using your build environment with an ESP32 and my test account with my test messages? I don't want to waste your time if the problem is caused by my build environment.
I do also normally develop in PlatformIO but was using the latest Espressif 6.3.2 which is using Arduino core 2.0.9. I experienced the problem in my application which targets ESP32-S3. To exclude all other factors I went back to the Arduino IDE and a basic ESP32. Based on your feedback I rebuild my test code on PlatformIO using Espressif 6.2.0 which uses Arduino core 2.0.8 (see platformio.ini below). Unfortunately I am getting exactly the same problems as before. I also switched to STARTTLS (port 143) but did not see a difference.
Here is my platformio.ini
```yaml
[env:mhetesp32minikit]
platform = espressif32@6.2.0
board = mhetesp32minikit
framework = arduino
lib_deps = mobizt/ESP Mail Client@^3.2.4
monitor_speed = 115200
build_type = release
monitor_filters = esp32_exception_decoder
build_flags = 
;	-DCORE_DEBUG_LEVEL=5
```

And here is the log from (in this run the problem occurs for UID 261 and 268) the test code built with PlatformIO and Espressif 6.2.0 configured for STARTTLS and using the test account I provided you:
```yaml
Connecting to Wi-Fi.
Connected with IP: 172.20.10.10

#### Connecting to IMAP server...
> C: ESP Mail Client v3.2.4
> C: connecting to IMAP server
> C: Host > mailServer
> C: Port > 143
* OK [CAPABILITY IMAP4rev1 SASL-IR LOGIN-REFERRALS ID ENABLE IDLE LITERAL+ STARTTLS AUTH=PLAIN] Dovecot ready.

#### IMAP server connected
> C: IMAP server connected

#### Checking the capability...
> C: check the capability
* CAPABILITY IMAP4rev1 SASL-IR LOGIN-REFERRALS ID ENABLE IDLE LITERAL+ STARTTLS AUTH=PLAIN
xmail OK Pre-login capabilities listed, post-login capabilities have more.

#### Sending STARTTLS command...
> C: send command, STARTTLS
xmail OK Begin TLS negotiation now.
> C: perform SSL/TLS handshake

> C: SSL/TLS negotiation
> C: seeding the random number generator
> C: setting up the SSL/TLS structure
! W: Skipping SSL Verification. INSECURE!
> C: setting hostname for TLS session
> C: perform the SSL/TLS handshake
> C: verifying peer X.509 certificate

#### Checking the capability...
> C: check the capability
* CAPABILITY IMAP4rev1 SASL-IR LOGIN-REFERRALS ID ENABLE IDLE LITERAL+ AUTH=PLAIN
xmail OK Pre-login capabilities listed, post-login capabilities have more.

#### Logging in...
> C: send IMAP command, AUTHENTICATE PLAIN
* CAPABILITY IMAP4rev1 SASL-IR LOGIN-REFERRALS ID ENABLE IDLE SORT SORT=DISPLAY THREAD=REFERENCES THREAD=REFS THREAD=ORDEREDSUBJECT MULTIAPPEND URL-PARTIAL CATENATE UNSELECT CHILDREN NAMESPACE UIDPLUS LIST-EXTENDED I18NLEVEL=1 CONDSTORE QRESYNC ESEARCH ESORT SEARCHRES WITHIN CONTEXT=SEARCH LIST-STATUS BINARY MOVE SNIPPET=FUZZY PREVIEW=FUZZY PREVIEW STATUS=SIZE SAVEDATE LITERAL+ NOTIFY SPECIAL-USE QUOTA
xmail OK Logged in

#### Send client identification...
> C: send IMAP command, ID
* ID ("name" "Dovecot")
xmail OK ID completed (0.001 + 0.546 + 0.546 secs).

Successfully logged in.
3880

#### Open the mailbox folder...
> C: selecting the INBOX folder...
* FLAGS (\Answered \Flagged \Deleted \Seen \Draft NonJunk)
[... 985 lines cut ...]
UID: 226        Subject: This Text is to make the header subject field a bit longer to provoke potential issues Test Nr 11
UID: 225        Subject: This Text is to make the header subject field a bit longer to provoke potential issues Test Nr 10
UID: 224        Subject: This Text is to make the header subject field a bit longer to provoke potential issues Test Nr 9
UID: 223        Subject: This Text is to make the header subject field a bit longer to provoke potential issues Test Nr 8
UID: 222        Subject: This Text is to make the header subject field a bit longer to provoke potential issues Test Nr 7
UID: 221        Subject: This Text is to make the header subject field a bit longer to provoke potential issues Test Nr 6
UID: 220        Subject: This Text is to make the header subject field a bit longer to provoke potential issues Test Nr 5
UID: 219        Subject: This Text is to make the header subject field a bit longer to provoke potential issues Test Nr 4
UID: 218        Subject: This Text is to make the header subject field a bit longer to provoke potential issues Test Nr 3
UID: 217        Subject: This Text is to make the header subject field a bit longer to provoke potential issues Test Nr 2
UID: 216        Subject: This Text is to make the header subject field a bit longer to provoke potential issues Test Nr 1
UID: 215        Subject: This Text is to make the header subject field a bit longer to provoke potential issues Test Nr 0

#### Logging out...
> C: send IMAP command, LOGOUT
* BYE Logging out
xmail OK Logout completed (0.001 + 0.088 + 0.088 secs).
! E: SSL - The peer notified us that the connection is going to be closed
> C: cleaning SSL connection

#### Log out completed
> C: log out completed
Free Heap: 230244

```

mobizt, 2023-07-24:
Today I try with 1600 messages fetching only header in Gmail.

I can confirm that I did not find issue on Arduino IDE verbose debug build in core 2.0.10.

My device is ESP32-WROVER. 

I don't have other series of ESP32 module in hand.

If you confirm the issue found in PlatformIO too, it may relate to core SDK variant.

Can you test with other ESP32 WROOM or WROVER?

mobizt, 2023-07-24:
Your mail service using [dovecot](https://www.dovecot.org/) which is the most RFC compliance mail server and other mail server too then it will not send truncated invalid response like this.

dkalliv, 2023-07-24:
Sorry for asking again, but just to be 100% sure: you can confirm, that using an ESP32-WROVER and my mailservice (Hostpoint with Dovecot) you can  not reproduce any of the issues (breaking of untagged responses and breaking of short tagged responses)?
I just tested with a ESP32 WROVER based module (previous test was with a WROOM based module) and can confirm that I see the same issue.
Just to clarify: After your fix in v3.2.1 I did no longer get any problems with Gmail. The problem I am seeing now concerns only the provider Hostpoint with Dovecot. What I really do not understand is why this
fails:
```yaml
#### Fetch message 11, UID: 268
> C: send IMAP command, FETCH
> C: fetch message header
* 54 FETCH (UID 268 MODSEQ (397) BODY[HEADE
R.FIELDS (FROM SENDER TO CC SUBJECT DATE MESSAGE-ID RETURN-PATH REPLY-TO IN-REPLY-TO REFERENCES COMMENTS KEYWORDS BCC FLAGS CONTENT-TYPE CONTENT-TRANSFER-ENCODING CONTENT-LANGUAGE ACCEPT-LANGUAGE)] {289}
Return-Path: <sender@senderDomain>
From: sender@senderDomain
To: recipient@recipientDomain
Subject: This Text is to make the header subject field a bit longer to provoke potential issues Test Nr 57
Date: Sun, 23 Jul 2023 17:24:34 +0200 (CEST)
Message-Id: <E1qNawt-0006j9-2F@mxin019.mail.hostpoint.ch>
)
xmail OK Fetch completed (0.001 + 0.266 secs).
#### Error, Fetch completed (0.001 + 0.266 secs).test1@dkmb.ch
! E: Fetch completed (0.001 + 0.266 secs).
```
while this works:
```yaml
#### Fetch message 45, UID: 3919
> C: send IMAP command, FETCH
> C: fetch message header
* 22 FETCH (UID 3919 MODSEQ (504448) BODY[HEADER.FIELDS (From Send
er To Cc Subject Date Message-ID Return-Path Reply-To In-Reply-To References Comments Keywords Bcc Flags Content-Type Content-transfer-encoding Content-Language Accept-Language)] {291}
Return-Path: <sender@senderDomain>
From: sender@senderDomain
To: recipient@recipientDomain
Subject: This Text is to make the header subject field a bit longer to provoke
 potential issues Test Nr 26
Date: Sun, 23 Jul 2023 17:21:54 +0200 (CEST)
Message-Id: <E1qNauB-000EBe-2G@mxin018.mail.hostpoint.ch>
)
xmail OK Success
```

mobizt, 2023-07-24:
Yes, I confirm. There is no issue.

I will do some additional test with simulate condition.

mobizt, 2023-07-25:
There are possible two cases that cause the parsing error in this library which your issue may be related to this both cases.

The response is actually spit into multiple truncated parts instead of unknown line break injection as my formerly assumption. 

1.  Sometimes data is not available due to network condition, client does not get the data even there are remaining data to read.

This case is mostly happened (especially in your case) which causes the [readLine](https://github.com/mobizt/ESP-Mail-Client/blob/779c305be0666b98e46f611858e2770175de22a2/src/ESP_Mail_Client.cpp#L1093) function exits before response reading to be completed with line break.

You will see the data truncation even in small response reading.
  
2. When the response size is bigger than 1024 bytes (the default response buffer size), that causes the response splitting into multiple truncated parts.

This usually occurred in case very long message ID or long header string contains in header fetching response. 

When this occurs, the false error like this will be shown.
` ! E: Fetch completed`

Now both cases are now fixed in the latest update, [v3.2.5](https://github.com/mobizt/ESP-Mail-Client/releases/tag/v3.2.5).

Please update the library. 

Hope this will fix your issue.

dkalliv, 2023-07-25:
Thank you for the update. My initial tests didn't show any stray line breaks, however I can not test extensively, as for some reason the connection is closed consistently after fetching 4 messages, all subsequent fetches fail (see log).

```yaml
#### Searching messages...
> C: searching messages
[  3278][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 20 bytes...
[  3286][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 2 bytes...
> C: [#               ] 9 %
> C: [##              ] 15 %
> C: [###             ] 20 %
> C: [####            ] 26 %
> C: [#####           ] 32 %
> C: [######          ] 38 %
> C: [#######         ] 49 %
> C: [########        ] 55 %
> C: [#########       ] 60 %
> C: [##########      ] 66 %
> C: [###########     ] 72 %
> C: [############    ] 78 %
> C: [#############   ] 83 %
> C: [##############  ] 89 %
> C: [############### ] 95 %
> C: [################] 100 %

#### Search limit: 64
Found 65 messages
Show 64 messages

#### Fetch message 1, UID: 279
> C: send IMAP command, FETCH
> C: fetch message header
[  3436][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 249 bytes...
[  3441][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 2 bytes...
* 65 FETCH (UID 279 MODSEQ (419) BODY[HEADER.FIELDS (FROM SENDER TO CC SUBJECT DATE MESSAGE-ID RETURN-PATH REPLY-TO IN-REPLY-TO REFERENCES COMMENTS KEYWORDS BCC FLAGS CONTENT-TYPE CONTENT-TRANSFER-ENCODING CONTENT-LANGUAGE ACCEPT-LANGUAGE)] {365}
Return-Path: <sender@senderDomain>
Message-ID: <8b4c5b97-13f8-c4b4-0567-5631dc0b72a6@senderDomain>
Date: Tue, 25 Jul 2023 00:35:39 +0200
Content-Language: de-CH
To: recipient@recipientDomain
From: "senderName" <sender@senderDomain>
Subject: neu
Content-Type: text/plain; charset=UTF-8; format=flowed
Content-Transfer-Encoding: 7bit
)
xmail OK Fetch completed (0.001 + 0.061 secs).

#### Get Flags...
> C: get Flags
[  3602][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 22 bytes...
[  3607][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 2 bytes...
* 65 FETCH (FLAGS (\Seen NonJunk))
xmail OK Fetch completed (0.001 + 0.211 secs).
> C: Free Heap: 199804

#### Fetch message 2, UID: 278
> C: send IMAP command, FETCH
> C: fetch message header
[  3855][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 249 bytes...
[  3860][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 2 bytes...
* 64 FETCH (UID 278 MODSEQ (413) BODY[HEADER.FIELDS (FROM SENDER TO CC SUBJECT DATE MESSAGE-ID RETURN-PATH REPLY-TO IN-REPLY-TO REFERENCES COMMENTS KEYWORDS BCC FLAGS CONTENT-TYPE CONTENT-TRANSFER-ENCODING CONTENT-LANGUAGE ACCEPT-LANGUAGE)] {289}
Return-Path: <sender@senderDomain>
From: sender@senderDomain
To: recipient@recipientDomain
Subject: This Text is to make the header subject field a bit longer to provoke potential issues Test Nr 63
Date: Sun, 23 Jul 2023 17:25:05 +0200 (CEST)
Message-Id: <E1qNaxr-0006yz-2a@mxin019.serverDomain>
)
xmail OK Fetch completed (0.001 + 0.070 secs).

#### Get Flags...
> C: get Flags
[  4014][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 22 bytes...
[  4025][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 2 bytes...
* 64 FETCH (FLAGS (\Seen NonJunk))
xmail OK Fetch completed (0.001 + 0.105 secs).
> C: Free Heap: 199168

#### Fetch message 3, UID: 277
> C: send IMAP command, FETCH
> C: fetch message header
[  4175][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 249 bytes...
[  4179][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 2 bytes...
* 63 FETCH (UID 277 MODSEQ (416) BODY[HEADER.FIELDS (FROM SENDER TO CC SUBJECT DATE MESSAGE-ID RETURN-PATH REPLY-TO IN-REPLY-TO REFERENCES COMMENTS KEYWORDS BCC FLAGS CONTENT-TYPE CONTENT-TRANSFER-ENCODING CONTENT-LANGUAGE ACCEPT-LANGUAGE)] {289}
Return-Path: <sender@senderDomain>
From: sender@senderDomain
To: recipient@recipientDomain
Subject: This Text is to make the header subject field a bit longer to provoke potential issues Test Nr 59
Date: Sun, 23 Jul 2023 17:24:44 +0200 (CEST)
Message-Id: <E1qNaxq-0006yJ-3C@mxin019.serverDomain>
)
xmail OK Fetch completed (0.001 + 0.059 secs).

#### Get Flags...
> C: get Flags
[  4325][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 22 bytes...
[  4337][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 2 bytes...
* 63 FETCH (FLAGS (\Seen NonJunk))
xmail OK Fetch completed (0.001 + 0.067 secs).
> C: Free Heap: 198232

#### Fetch message 4, UID: 276
> C: send IMAP command, FETCH
> C: fetch message header
[  4447][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 249 bytes...
[  4452][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 2 bytes...
* 62 FETCH (UID 276 MODSEQ (416) BODY[HEADER.FIELDS (FROM SENDER TO CC SUBJECT DATE MESSAGE-ID RETURN-PATH REPLY-TO IN-REPLY-TO REFERENCES COMMENTS KEYWORDS BCC FLAGS CONTENT-TYPE CONTENT-TRANSFER-ENCODING CONTENT-LANGUAGE ACCEPT-LANGUAGE)] {289}
Return-Path: <sender@senderDomain>
From: sender@senderDomain
To: recipient@recipientDomain
Subject: This Text is to make the header subject field a bit longer to provoke potential issues Test Nr 62
Date: Sun, 23 Jul 2023 17:24:59 +0200 (CEST)
Message-Id: <E1qNaxA-000MqW-Sh@mxin016.serverDomain>
)
> C: cleaning SSL connection
[  4599][V][ESP32_SSL_Client.cpp:503] stop_tcp_connection(): Cleaning SSL connection.
xmail OK
#### Error, connection closed
! E: connection closed

#### Fetch message 5, UID: 275
> C: send IMAP command, FETCH
> C: fetch message header
#### Error, connection closed
! E: connection closed

#### Fetch message 6, UID: 274
> C: send IMAP command, FETCH
> C: fetch message header
#### Error, connection closed
! E: connection closed
```

mobizt, 2023-07-25:
It can be network traffic issue.

dkalliv, 2023-07-25:
I don't think so, I tested several times over two different connections with exactly the same result. When going back to v3.2.0 I can fetch all the messages (with the obvious problems due to line breaks). v3.2.4 also fails after fetching 4 messages and triggers an unhandled exception:

```yaml
#### Fetch message 4, UID: 276
> C: send IMAP command, FETCH
> C: fetch message header
[  5388][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 249 bytes...
[  5393][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 2 bytes...
* 62 FETCH (UID 276 MODSEQ (416) BODY[HEADER.FIELDS (FROM SENDER TO CC SUBJECT DATE MESSAGE-ID RETURN-PATH REPLY-TO IN-REPLY-TO REFERENCES COMMENTS KEYWORDS BCC FLAGS CONTENT-TYPE CONTENT-TRANSFER-ENCODING CONTENT-LANGUAGE ACCEPT-LANGUAGE)] {289}
Return-Path: <sender@senderDomain>
From: sender@senderDomain
To: recipient@recipientDomain
Subject: This Text is to make the header subject field a bit longer to provoke potential issues Test Nr 62
Date: Sun, 23 Jul 2023 17:24:59 +0200 (CEST)
Message-Id: <E1qNaxA-000MqW-Sh@mxin016.serverDomain>
)
xmail OK
 Fetch completed (0.001 + 0.105 secs).
> C: cleaning SSL connection
[ 35762][V][ESP32_SSL_Client.cpp:503] stop_tcp_connection(): Cleaning SSL connection.
#### Error, session timed out
! E: session timed out
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d2cfa  PS      : 0x00060330  A0      : 0x800d2d53  A1      : 0x3ffb1d90  
A2      : 0x000000dc  A3      : 0x00000014  A4      : 0x00000000  A5      : 0x3ffce478  
A6      : 0x3f41aa89  A7      : 0x00000000  A8      : 0x00000012  A9      : 0x3ffb1d60  
A10     : 0x3ffb1da8  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000014  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x000000e0  LBEG    : 0x4008a671  LEND    : 0x4008a681  LCOUNT  : 0xfffffffb  

Backtrace: 0x400d2cf7:0x3ffb1d90 0x400d2d50:0x3ffb1db0 0x400d2d86:0x3ffb1dd0 0x400da088:0x3ffb1df0 0x400e1566:0x3ffb1e30 0x400e3ede:0x3ffb2160 0x400d300b:0x3ffb2230 0x400e833e:0x3ffb2290

  #0  0x400d2cf7:0x3ffb1d90 in MB_String::_reserve(unsigned int, bool) at .pio/libdeps/mhetesp32minikit/ESP Mail Client/src/extras/MB_String.h:1785
  #1  0x400d2d50:0x3ffb1db0 in MB_String::copy(char const*, unsigned int) at .pio/libdeps/mhetesp32minikit/ESP Mail Client/src/extras/MB_String.h:1766
  #2  0x400d2d86:0x3ffb1dd0 in MB_String::operator=(char const*) at .pio/libdeps/mhetesp32minikit/ESP Mail Client/src/extras/MB_String.h:972
  #3  0x400da088:0x3ffb1df0 in ESP_Mail_Client::reconnect(IMAPSession*, unsigned long, bool) at .pio/libdeps/mhetesp32minikit/ESP Mail Client/src/ESP_Mail_IMAP.h:2617
  #4  0x400e1566:0x3ffb1e30 in ESP_Mail_Client::handleIMAPResponse(IMAPSession*, int, bool) at .pio/libdeps/mhetesp32minikit/ESP Mail Client/src/ESP_Mail_IMAP.h:2801
  #5  0x400e3ede:0x3ffb2160 in ESP_Mail_Client::readMail(IMAPSession*, bool) at .pio/libdeps/mhetesp32minikit/ESP Mail Client/src/ESP_Mail_IMAP.h:682
  #6  0x400d300b:0x3ffb2230 in setup() at src/main.cpp:117
  #7  0x400e833e:0x3ffb2290 in loopTask(void*) at /home/dkal/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42
```

mobizt, 2023-07-25:
This can be the reason why that you got this issue that I never discoverd.

There are many reasons that cause the network issue e.g. shared devices under the same network, signal strength, your power supply and ISP.

When you use TCP keep alive feature via the the keepAlive function, the client connected function will show nearly realtime server connecting status.

dkalliv, 2023-07-25:
Using the same network connections, the same device and v3.2.5 I can retrieve the same test messages via Gmail without any issues. I even tried different boards and they all fail after fetching the 4th message from the Hostpoint test account. For good measure I also added the keepAlive() function to my test code, but it made no difference. It is therefore highly unlikely, that this has to do anything with my ISP or HW setup, especially as I am currently working from a completely different environment as when I originally reported the issue. I am currently on vacation and use a different ISP and power my devices from my notebook (I hope you don't think I'm crazy for carrying a bunch of microcontrollers with me during vacation, after all I'm an engineer and there are some rainy days...)

mobizt, 2023-07-25:
If you see such this message, make sure you use new version library because it will not see in the latest version.

```
xmail OK
 Fetch completed (0.001 + 0.105 secs).
```

Deleting the Arduino built cache files and deleting all versions of this library in Arduino libraries folder are recommended.

I use both 5G network and Home Fibre internet to test and it works fine.

The TCP keep alive can't improve or fix anything. It is network probing mechanism that helps you to know how and when your device is disconnected from server or when session was closed while the session was kept reuse or open.

Thanks to share your situation. 

I understand your situation and don't want you to quickly reply or response.

And also, I'm one of busy engineer. 

And this is not a forum that no one needs to care your message. I spend the time and take the most responsibility here because you are posting in my workplace for help.

mobizt, 2023-07-25:
In version 3.2.4 and 3.2.5 you still get the null pointer error after 4 messages fetching. 

The version 3.2.5 is entirely fixed the unexpected behavior in response buffering that can cause subsequence failure of null pointer exception as in previous version.

Then you should clean install the library to test.

mobizt, 2023-07-25:
This is line number 2617 from 3.2.4 from `ESP_Mail_IMAP.h` from trace decoded result which not guard of using `cHeader(imap)` that can be returned the null pointer when current header object was not yet allocated due to header response parsing error.
https://github.com/mobizt/ESP-Mail-Client/blob/779c305be0666b98e46f611858e2770175de22a2/src/ESP_Mail_IMAP.h#L2617
This is line number 2617 from version 3.2.5
https://github.com/mobizt/ESP-Mail-Client/blob/469f6b69f741ce81d757a4d354f3c452de663e29/src/ESP_Mail_IMAP.h#L2617

dkalliv, 2023-07-25:
I apologize if I caused confusion. The Backtrace I've sent you is from v3.2.4 and not v3.2.5 (maybe I did not point this out clearly enough). I thought it might help to identify the issue...
I followed your advice and did a "full Clean" in PlatfrormIO and even manually deleted the library from .pio/libdeps. Nevertheless I still get exactly the same behavior, when using the Hostpoint test account I provided you (using a ESP32 WROVER module). The connection is terminated after fetching the 4th message:
```yaml
#### Fetch message 4, UID: 276
> C: send IMAP command, FETCH
> C: fetch message header
[  7693][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 249 bytes...
[  7700][V][ESP32_SSL_Client.cpp:534] send_ssl_data(): Writing request with 2 bytes...
* 62 FETCH (UID 276 MODSEQ (416) BODY[HEADER.FIELDS (FROM SENDER TO CC SUBJECT DATE MESSAGE-ID RETURN-PATH REPLY-TO IN-REPLY-TO REFERENCES COMMENTS KEYWORDS BCC FLAGS CONTENT-TYPE CONTENT-TRANSFER-ENCODING CONTENT-LANGUAGE ACCEPT-LANGUAGE)] {289}
Return-Path: <sender@senderDomain>
From: sender@senderDomain
To: recipient@recipientDomain
Subject: This Text is to make the header subject field a bit longer to provoke potential issues Test Nr 62
Date: Sun, 23 Jul 2023 17:24:59 +0200 (CEST)
Message-Id: <E1qNaxA-000MqW-Sh@mxin016.serverDomain>
)
> C: cleaning SSL connection
[  8866][V][ESP32_SSL_Client.cpp:503] stop_tcp_connection(): Cleaning SSL connection.
xmail OK
#### Error, connection closed
! E: connection closed
```
When originally identifying the issue I was connected over a 10GBit/s fiber connection and was also using a 5G connection for testing. Currently I'm testing with a 50MBit/s ADSL connection as well as a 5G connection.

I do highly appreciate your commitment and dedication (your troubleshooting and fixing is extremely competent and fast) as well as your contribution to the open source community. I fully understand if you no longer want to invest time in resolving an issue you cannot reproduce. Can you confirm, that using the test code I originally submitted with an ESP32 module and the Hostpoint test account I provided, you can fetch all messages without disconnect?

mobizt, 2023-07-25:
Don't forget to call `setInsecure()` to `WiFiClientSecure` to bypass server certificate verification.

mobizt, 2023-07-26:
In case your laptop is your board power supply, I recommend you should replace your USB cable with new one with shorter length because you got the same internet connection issue on different boards.

dkalliv, 2023-07-26:
Thank you for your recommendation. As one of my boards has a micro USB connector and the other boards have USB-C connectors I use different cables, all of them very short (10cm). I also tested with a 1m cable but found no difference.
I am very confident, that this is not a problem with power supply stability, interference, or internet traffic / stability as these are all random processes and the fetch always fails after the 4th message (across all boards and with both internet connections I have currently available).
I put together a very simple IMAP client using WiFiClientSecure to verify my assumptions and found that I can fetch all messages without any problem (and quite fast). I can therefore confidently exclude a hardware issue. Maybe it is a problem with MbedTLS, but then again you should have had the same issue on your side. I have to decide how to proceed from here. Thank you anyway for your work...

dkalliv, 2023-07-26:
Thank you, this sounds good!

mobizt, 2023-07-27:
Now the library update is available with WiFiClient replacement by using `lwIP TCP Client`.

The speed and stability improved.

In the version [3.3.0](https://github.com/mobizt/ESP-Mail-Client/releases/tag/v3.3.0), you can test with/without `WiFiClient` implementation to compare.

To test with slowest speed WiFiClient with its socket, define the macro in `ESP_Mail_FS.h` or build flag, `ESP_MAIL_ESP32_USE_WIFICLIENT_SOCKET_TEST` 

To test with slow speed WiFiClient, define the macro in `ESP_Mail_FS.h` or build flag, `ESP_MAIL_ESP32_USE_WIFICLIENT_TEST`

To test with fastest speed `lwIP TCP Client`, just undefine both `ESP_MAIL_ESP32_USE_WIFICLIENT_SOCKET_TEST` and `ESP_MAIL_ESP32_USE_WIFICLIENT_TEST`.

dkalliv, 2023-07-27:
Thank you for the update! I have just finished my testing:
Using the macros ESP_MAIL_ESP32_USE_WIFICLIENT_SOCKET_TEST and ESP_MAIL_ESP32_USE_WIFICLIENT_TEST I can fetch all messages without any problem (apart from being extremely slow).
Without the macros I get exactly the same behavior as before: fetching fails at the 4th message.
Does this mean there is a timing issue when getting data from the SSL read buffer?

mobizt, 2023-07-27:
No timing issue, data handles by mbedTLS entirely.

It is lwIP socket interfaces with mbedTLS the same way, same method and same code as in in latest WiFiClientSecure when handling transportation since TLS handshake.

I will verify the socket options when I'm at my PC but at this time it's standard setup.

dkalliv, 2023-07-27:
I have to add, that out of habit I used my application (not my test code) and an ESP32-S3 for the test. I now also used a ESP32 WROVER and my test code and got different results.
Without the macros I get an exception:
```yaml
Connecting to Wi-Fi..
Connected with IP: 172.20.10.4

#### Connecting to IMAP server...
> C: ESP Mail Client v3.3.0
> C: connecting to IMAP server
> C: Host > imap.mail.hostpoint.ch
> C: Port > 993
> C: Free internal heap before TLS 247292
> C: Starting socket

> C: SSL/TLS negotiation
> C: seeding the random number generator
> C: setting up the SSL/TLS structure
! W: Skipping SSL Verification. INSECURE!
> C: setting hostname for TLS session
> C: perform the SSL/TLS handshake
> C: verifying peer X.509 certificate

assert failed: heap_caps_free heap_caps.c:360 (heap != NULL && "free() target pointer is outside heap areas")

Backtrace: 0x40083705:0x3ffb1e80 0x4008ce41:0x3ffb1ea0 0x40092761:0x3ffb1ec0 0x40083c42:0x3ffb1ff0 0x40092791:0x3ffb2010 0x400d2aeb:0x3ffb2030 0x400d2b49:0x3ffb2050 0x400e4252:0x3ffb2070 0x400e4288:0x3ffb2090 0x400e42c5:0x3ffb20b0 0x40162db2:0x3ffb20d0 0x40162e31:0x3ffb2110 0x400e3f3e:0x3ffb2130 0x400d98b5:0x3ffb2150 0x400df95e:0x3ffb2190 0x400dfb58:0x3ffb21f0 0x400e2252:0x3ffb2210 0x400d2d83:0x3ffb2230 0x400e770e:0x3ffb2290

  #0  0x40083705:0x3ffb1e80 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
  #1  0x4008ce41:0x3ffb1ea0 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
  #2  0x40092761:0x3ffb1ec0 in __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85
  #3  0x40083c42:0x3ffb1ff0 in heap_caps_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/heap_caps.c:360
      (inlined by) heap_caps_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/heap_caps.c:345
  #4  0x40092791:0x3ffb2010 in free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/heap.c:39
  #5  0x400d2aeb:0x3ffb2030 in MB_String::allocate(unsigned int, bool) at .pio/libdeps/mhetesp32minikit/ESP Mail Client/src/extras/MB_String.h:1709
  #6  0x400d2b49:0x3ffb2050 in MB_String::clear() at .pio/libdeps/mhetesp32minikit/ESP Mail Client/src/extras/MB_String.h:1208
  #7  0x400e4252:0x3ffb2070 in ESP32_WCS::stop() at .pio/libdeps/mhetesp32minikit/ESP Mail Client/src/wcs/esp32/ESP32_WCS.cpp:128
      (inlined by) ESP32_WCS::stop() at .pio/libdeps/mhetesp32minikit/ESP Mail Client/src/wcs/esp32/ESP32_WCS.cpp:121
  #8  0x400e4288:0x3ffb2090 in ESP32_WCS::~ESP32_WCS() at .pio/libdeps/mhetesp32minikit/ESP Mail Client/src/wcs/esp32/ESP32_WCS.cpp:84
  #9  0x400e42c5:0x3ffb20b0 in ESP32_WCS::~ESP32_WCS() at .pio/libdeps/mhetesp32minikit/ESP Mail Client/src/wcs/esp32/ESP32_WCS.cpp:96
  #10 0x40162db2:0x3ffb20d0 in WiFiClient::setSocketOption(int, int, void const*, unsigned int) at /home/dkal/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiClient.cpp:311
  #11 0x40162e31:0x3ffb2110 in WiFiClient::setOption(int, int*) at /home/dkal/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiClient.cpp:338
  #12 0x400e3f3e:0x3ffb2130 in ESP32_WCS::setOption(int, int*) at .pio/libdeps/mhetesp32minikit/ESP Mail Client/src/wcs/esp32/ESP32_WCS.h:425
      (inlined by) ESP32_TCP_Client::connect(bool, bool) at .pio/libdeps/mhetesp32minikit/ESP Mail Client/src/wcs/esp32/ESP32_TCP_Client.cpp:341
      (inlined by) ESP32_TCP_Client::connect(bool, bool) at .pio/libdeps/mhetesp32minikit/ESP Mail Client/src/wcs/esp32/ESP32_TCP_Client.cpp:273
  #13 0x400d98b5:0x3ffb2150 in ESP_Mail_Client::beginConnection(esp_mail_session_config_t*, void*, bool, bool) at .pio/libdeps/mhetesp32minikit/ESP Mail Client/src/ESP_Mail_Client.cpp:1344
  #14 0x400df95e:0x3ffb2190 in IMAPSession::connect(bool&) at .pio/libdeps/mhetesp32minikit/ESP Mail Client/src/ESP_Mail_IMAP.h:5079
  #15 0x400dfb58:0x3ffb21f0 in IMAPSession::handleConnection(esp_mail_session_config_t*, esp_mail_imap_data_config_t*, bool&) at .pio/libdeps/mhetesp32minikit/ESP Mail Client/src/ESP_Mail_IMAP.h:5019
      (inlined by) IMAPSession::handleConnection(esp_mail_session_config_t*, esp_mail_imap_data_config_t*, bool&) at .pio/libdeps/mhetesp32minikit/ESP Mail Client/src/ESP_Mail_IMAP.h:4993
  #16 0x400e2252:0x3ffb2210 in IMAPSession::connect(esp_mail_session_config_t*, esp_mail_imap_data_config_t*, bool) at .pio/libdeps/mhetesp32minikit/ESP Mail Client/src/ESP_Mail_IMAP.h:4807
  #17 0x400d2d83:0x3ffb2230 in setup() at src/main.cpp:98
  #18 0x400e770e:0x3ffb2290 in loopTask(void*) at /home/dkal/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42
```
I will provide more results in the evening (local time) as I will be away from my computer for a while...

mobizt, 2023-07-27:
If this error occurred in all board revisions, you should provide the simple code that reproduces the error.

If may related to pointer handling.

If this error is board specific, it may related to MMU and selected board compatibility.

mobizt, 2023-07-27:
Now the server disconnected issue is now fixed with [this update](https://github.com/mobizt/ESP-Mail-Client/releases/tag/v3.3.1).

The root cause is related to internet condition and improper implement of `stop()` to close session when data is not `available` and when `write()` and `read()` function return error.

The timeout will never be happened because of session was immediately closed when these three situations occurred.

This issue was only found on ESP32 and I never find this issue when internal WiFiClient was used as time out was implement differently.

dkalliv, 2023-07-27:
Thank you for this update!
Things are now working nice and fast and I can fetch all the messages without defining the macros. I also figured out the cause for the exceptions: I actually did one small change to my test code, I added a keepAlive() upon your suggestion. Unfortunately I did not pay attention and added it before opening the connection. While ```WiFiClient``` didn't mind, ```lwIP TCP Client``` threw an exception.
Thank you for your time and persistence and figuring out where the problem is...

mobizt, 2023-07-28:
I also thanks for your time too for testing and prove that this is actually issue.

mobizt, 2023-07-28:
[This version](https://github.com/mobizt/ESP-Mail-Client/releases/tag/v3.3.3) fixed `keepAlive` issue.