mobizt/ESP-Mail-Client #286: UNO R4  wifi board

BerieF2023, 2023-08-08:
**Build tool used:**

- [x ] Arudino using IDE v2.1.1 on Windows 11
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based
- [ ] ESP32-based

**Other Libraries That are used:**

**Description of problem:**
Hi 
I  just purchased   the Arduino R4 WIFI  board to use  to send  email alerts . 
I then downloaded  the sample send sketch
(https://github.com/mobizt/ESP-Mail-Client/tree/master/examples)/[SMTP](https://github.com/mobizt/ESP-Mail-Client/tree/master/examples/SMTP)/[Send_Text]

This did  not  compile/verify. gives the following  errors

**_WARNING: library ESP Mail Client claims to run on esp8266, esp32, sam, samd, stm32, STM32F1, STM32F4, teensy, avr, megaavr, mbed_nano, mbed_rp2040, rp2040 architecture(s) and may be incompatible with your current board which runs on renesas_uno architecture(s).
C:\Users\CV\AppData\Local\Temp\.arduinoIDE-unsaved202378-1896-1yqo019.i9duj\sketch_aug8a\sketch_aug8a.ino: In function 'void setup()':
C:\Users\CV\AppData\Local\Temp\.arduinoIDE-unsaved202378-1896-1yqo019.i9duj\sketch_aug8a\sketch_aug8a.ino:112:3: error: 'WiFi' was not declared in this scope
   WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
   ^~~~
C:\Users\CV\AppData\Local\Temp\.arduinoIDE-unsaved202378-1896-1yqo019.i9duj\sketch_aug8a\sketch_aug8a.ino:121:27: error: 'WL_CONNECTED' was not declared in this scope
   while (WiFi.status() != WL_CONNECTED)
                           ^~~~~~~~~~~~exit status 1
Compilation error: 'WiFi' was not declared in this scope
exit status 1
Compilation error: 'WiFi' was not declared in this scope_**

**Additional information and things you've tried:**
_I had  already  compiled a sketch  that connected  to my wifi. Then as a trial I added  the line 
#include <ESP_Mail_Client.h>
this  failed to compile also  with WARNING: library ESP Mail Client claims to run on esp8266, esp32, sam, samd, stm32, STM32F1, STM32F4, teensy, avr, megaavr, mbed_nano, mbed_rp2040, rp2040 architecture(s) and may be incompatible with your current board which runs on renesas_uno architecture(s).
C:\Users\CV\OneDrive\Documents\Arduino\libraries\ESP_Mail_Client\src\extras\RFC2047.cpp: In member function 'void RFC2047_Decoder::rfc2047DecodeWord(char*, const char*, size_t)':
C:\Users\CV\OneDrive\Documents\Arduino\libraries\ESP_Mail_Client\src\extras\RFC2047.cpp:81:5: error: 'strsep' was not declared in this scope
     strsep(&end, "?");
     ^~~~~~
C:\Users\CV\OneDrive\Documents\Arduino\libraries\ESP_Mail_Client\src\extras\RFC2047.cpp:81:5: note: suggested alternative: 'strstr'
     strsep(&end, "?");
     ^~~~~~
     strstr
In file included from C:\Users\CV\OneDrive\Documents\Arduino\libraries\ESP_Mail_Client\src\ESP_Mail_Client.cpp:41:0:
C:\Users\CV\OneDrive\Documents\Arduino\libraries\ESP_Mail_Client\src\ESP_Mail_IMAP.h: In member function 'void ESP_Mail_Client::parseFoldersResponse(IMAPSession*, char*, bool)':
C:\Users\CV\OneDrive\Documents\Arduino\libraries\ESP_Mail_Client\src\ESP_Mail_IMAP.h:3704:13: error: 'strsep' was not declared in this scope
             strsep(&end, " ");
             ^~~~~~
C:\Users\CV\OneDrive\Documents\Arduino\libraries\ESP_Mail_Client\src\ESP_Mail_IMAP.h:3704:13: note: suggested alternative: 'strstr'
             strsep(&end, " ");
             ^~~~~~
             strstr
C:\Users\CV\OneDrive\Documents\Arduino\libraries\ESP_Mail_Client\src\ESP_Mail_IMAP.h: In member function 'bool ESP_Mail_Client::getFlags(IMAPSession*, char*, esp_mail_imap_response_types)':
C:\Users\CV\OneDrive\Documents\Arduino\libraries\ESP_Mail_Client\src\ESP_Mail_IMAP.h:3969:13: error: 'strsep' was not declared in this scope
             strsep(&end, " ");
             ^~~~~~
C:\Users\CV\OneDrive\Documents\Arduino\libraries\ESP_Mail_Client\src\ESP_Mail_IMAP.h:3969:13: note: suggested alternative: 'strstr'
             strsep(&end, " ");
             ^~~~~~
             strstr
C:\Users\CV\OneDrive\Documents\Arduino\libraries\ESP_Mail_Client\src\ESP_Mail_Client.cpp: In member function 'void ESP_Mail_Client::splitToken(const char*, std::vector<MB_String>&, const char*)':
C:\Users\CV\OneDrive\Documents\Arduino\libraries\ESP_Mail_Client\src\ESP_Mail_Client.cpp:816:5: error: 'strsep' was not declared in this scope
     strsep(&end, delim);
     ^~~~~~
C:\Users\CV\OneDrive\Documents\Arduino\libraries\ESP_Mail_Client\src\ESP_Mail_Client.cpp:816:5: note: suggested alternative: 'strstr'
     strsep(&end, delim);
     ^~~~~~
     strstr_
**Share code snippet to reproduce the issue:**
```yaml
PASTE .cpp / .ino code here

```

mobizt, 2023-08-08:
Arduino UNO R4 WiFi that uses Renesas MCU is not supported by current release of library (v3.3.7 and earlier).

You should try this unreleased [v3.4.0](https://github.com/mobizt/ESP-Mail-Client/tree/v3.4.0) which supports all Arduino devices included Arduino UNO R4 WiFi.

Because of [v3.4.0](https://github.com/mobizt/ESP-Mail-Client/tree/v3.4.0) is not yet released and is under development, to install [v3.4.0](https://github.com/mobizt/ESP-Mail-Client/tree/v3.4.0), you need to uninstall the older version first and download the zip file and install using zip file.

Arduino UNO R4 WiFi  uses `WiFiS3.h` library and you should include it to your code when using WiFi.
https://github.com/mobizt/ESP-Mail-Client/blob/195c67b3292b4d8f8603bcd710c2cd230c189291/examples/SMTP/Send_Text/Send_Text.ino#L35-L46

BerieF2023, 2023-08-08:
Wow. Thank you very much for the prompt and helpful response.

I'll try this out tomorrow. I'm on a big learning curve here with these
devices.

So again THANK YOU.
Cheers
Bernie

BerieF2023, 2023-08-09:
Hi Just to say I got it working fine thank you.
Had to create new sketches from scratch but it's send ING emails.

Many thanks again.

closed 2023-08-09 by mobizt