mobizt/ESP-Mail-Client #291: Compile flags and not unsets

guestisp, 2023-08-16:
Please use compile flags (like -D ENABLE_IMAP) and not "everything enabled and just disable what you want by editing a core file) to set which component has to be incuded.

This would be much easier for platformio users. Also, it's impossible to include `Custom_ESP_Mail_FS.h` in the same directory as `ESP_Mail_FS.h` if you are using a library manager like platformio, because the `ESP_Mail_FS.h` is outside the project directory.

So:

A. please use compile flag so that we can choose what to compile, by chaging the platformio config file
B. please don't search for `Custom_ESP_Mail_FS.h` in the same directory of ESP_Mail_FS.h but use the project directory. (but solution "A" is much better)

guestisp, 2023-08-16:
Yes, tried and build size is always the same regardless `build_unflgas`

```

build_unflags =
	-D ENABLE_IMAP
	-D ESP_MAIL_DEFAULT_SD_FS
	-D ESP_MAIL_DEFAULT_FLASH_FS
	-D ENABLE_NTP_TIME
```

i would expect a much smaller library, but the firmware is always the same.

mobizt, 2023-08-16:
Update:

Use build flags -D and -U can't take effect when that preprocessor macros are already defined.

You should remove preprocessor macro from `ESP_Mail_FS.h` then you can use build flags to def and undef.

If you use PlatformIO, please read PlatformIO documentation first.

`-D` followed by preprocessor name without space.

mobizt, 2023-08-16:
You can folk the library to work in your customization version.

By default, in `ESP_Mail_FS.h` file it can be only following.

```cpp

#include <FS.h> //<- use in v3.4.0
#define ESP_MAIL_DEFAULT_DEBUG_PORT Serial

```

guestisp, 2023-08-16:
Forking is what i'll do if i can't use the build_flag/unflag from pio