mobizt/ESP-Mail-Client #300: Defect: Divide/Modulo by zero

RavaszEmber, 2023-09-01:
DO NOT DELETE ANY TEXT from this template! Otherwise the issue may be closed without a comment.
-->

**Build tool used:**

- [ ] Arudino
- [x ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based
- [ x] ESP32-based

**Other Libraries That are used:**

lib_deps = 
	ottowinter/ESPAsyncWebServer-esphome@^3.0.0
	adafruit/DHT sensor library@^1.4.4
	mobizt/ESP Mail Client@^3.4.9

**Description of problem:**
I did an inspection of a project that included your library in platformio. It said there was a defect (divide/modulo by 0). 
This is the line: #if defined(ESP32) && __has_include(<sys/stat.h>)
The error seems to be the / you have in sys/stat.h.
[image]

Adjacent to this issue, the library itself seems to take a huge amount of flash memory (200 KB).
[image]
Is there a trimmed/optimized version I should be using instead? I don't believe I need SSL client, which seems to be half the library.
[image]

Thx for making this, it's very helpful!

**Share code snippet to reproduce the issue:**
```yaml
PASTE .cpp / .ino code here

```

**Additional information and things you've tried:**

mobizt, 2023-09-01:
## DO NOT OPEN THE ISSUE HERE

It's not library related issue. It is due to the bugs of the [cppcheck](https://sourceforge.net/projects/cppcheck/) tool that PlatformIO used for code/memory inspection.

This is the [known bugs](https://sourceforge.net/p/cppcheck/discussion/general/thread/f6587f0d65/) in [cppcheck](https://sourceforge.net/projects/cppcheck/) tool that can't parse __has_include syntax correctly when it found `<>` symbols.
`__has_include(<sys/stat.h>)`
It can parse only in case
`__has_include("sys/stat.h")`

And it is already the [issue](https://github.com/platformio/platformio-core/issues/4688) opened in PlatformIO repos.

And also you are currently using the out dated PlatformIO, when you update to latest version you will get another error "`Invalid Param`" which is the [bugs](https://github.com/platformio/platformio-vscode-ide/issues/3805) from [cppcheck](https://sourceforge.net/projects/cppcheck/) tool.

closed 2023-09-01 by mobizt

mobizt, 2023-09-01:
## PLEASE READ LIBRARY DOCUMENTATION BEFORE POSTING

> Adjacent to this issue, the library itself seems to take a huge amount of flash memory (200 KB).

This is comprehensive library of Email client.

The 100k flash space used by BearSSL and Client and the rest used by the code and PROGMEM debug string.

If you just simple use the library to send attachment message and don't care about the standards, please use other library because this library provides more features plus RFC compliances.

> Is there a trimmed/optimized version I should be using instead? I don't believe I need SSL client, which seems to be half the library.

If you are using only non-ssl ports, you can exclude the SSL Client.

You can build the library with some options for only used features.

https://github.com/mobizt/ESP-Mail-Client#library-build-options