mobizt/ESP-Mail-Client #303: Email rejected from IONOS if time is set through NTP (OK without setting time at all: 1st January 1970")

gsicilia82, 2023-09-16:
**Build tool used:**

- [ ] Arudino
- [X] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based
- [X] ESP32-based

**Other Libraries That are used:**

No other libs are installed

**Description of problem:**

---
1. When I use example code from **Send_Text.ino**, my email will be not sent from my email hoster IONOS. It will be rejected due to policy restrictions. I have investigated a little bit and it seems, that something is wrong with date in header from email. If NTP time will not be set at all in my code (local copy from your source code and undefined  "ENABLE_NTP_TIME"), it works well. This could be a workaround, but I don't know how long it would work...

  - About IONOS policies, I found this comment:
  "Email header must include valid Date related to RFC2822"
[https://datatracker.ietf.org/doc/html/rfc2822#section-3.3](https://datatracker.ietf.org/doc/html/rfc2822#section-3.3)

  - I would like to do further analysis, but your code is over my C++ skills...
  In short: No date at all works, set actual date over NTP fails.
  BTW: I configured correct GMT offsets to get actual German time.

---
2. I wanted to test with manual time settings. Your example code from file **Set_Time.ino** is not working.
 _Parameter `timestamp` is unknown in class `SMTP_Message`_:

```
    /* The time format of timestamp to inject into subject or content as using in strftime C++ function */
    message.timestamp.tag = "#esp_mail_current_time";

    /* The tag that will be replaced with current timestamp */
    message.timestamp.format = "%B %d, %Y %H:%M:%S";
```

**Share code snippet to reproduce the issue:**
[https://github.com/mobizt/ESP-Mail-Client/blob/master/examples/SMTP/Send_Text/Send_Text.ino](https://github.com/mobizt/ESP-Mail-Client/blob/master/examples/SMTP/Send_Text/Send_Text.ino)
[https://github.com/mobizt/ESP-Mail-Client/blob/master/examples/SMTP/Set_Time/Set_Time.ino](https://github.com/mobizt/ESP-Mail-Client/blob/master/examples/SMTP/Set_Time/Set_Time.ino)

**Additional information and things you've tried:**
To be sure, that NTP time +offsets will not be in the future, I tested different offsets. Always rejected from IONOS. With SMTP server from GMAIL it worked well.

mobizt, 2023-09-16:
No issue you mentioned in this library, especially in the latest version.

Before open the issue, you should update the library and start with the discussion first especially about the options and features.

The date header sent to server is valid RFC2822 date format and library is RFC standards compliance.

The Date header was set since [v1.5.10](https://github.com/mobizt/ESP-Mail-Client/releases/tag/v1.5.10).

Since your device time was set in `smtp.connect`, you can print the date with time that library will be set to Date header field with `MailClient.Time.getDateTimeString()` which will print the valid RFC2822 date.

`Sat, 16 Sep 2023 22:44:34 +0300`

After message sent, you can see the Date header that sent with the header fields from the message source that can be viewed from your Email client app.

```yaml
Date: Sat, 16 Sep 2023 22:36:01 +0300
Content-Type: text/plain; charset="UTF-8";
Content-transfer-encoding: 7bit
```

If you can't compile with `message.timestamp.tag`, that is because you installed the old version of library that is not supported.

The `message.timestamp.tag` does nothing with message header. It is for find and replace the tag with date time in the message title and body.

closed 2023-09-16 by mobizt

mobizt, 2023-09-17:
After investigation, you are right, the non-zero time zone in config causes only the time zone advancement while the time is still GMT (+0) when using `MailClient.Time.getDateTimeString()`.

I will fix this and update.

mobizt, 2023-09-17:
Latest investigation.

The issue is only occurred when you set your device time manually before calling `smtp.connect`, you need to set the time zone in the library config to match with your time zone.

`config.time.gmt_offset = xxx;`

If you don't set it, the default GMT offset (0) was used and you will see `+0000` in the date string from `MailClient.Time.getDateTimeString()` which may cause invalid date when your actual device time's time zone was set differently.