mobizt/ESP-Mail-Client #305: Error send mail

Michelbe30111, 2023-10-18:
**Build tool used:**

- [ x] Arudino
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ x] ESP8266-based
- [x ] ESP32-based

**Other Libraries That are used:**

**Description of problem:**
example Send_Reuse_Session_mail.ino 
time ok with esp8266
time error with esp32

send mail ok on esp8266
send mail ok for first send  with esp32  and for next send error:
Error, Status Code: 0, Error Code: -2, Reason: data sending failed
Error, Status Code: 0, Error Code: -100, Reason: unable to connect to server
Connection error, Status Code: 0, Error Code: -401, Reason: response read timed out

**Share code snippet to reproduce the issue:**
```yaml
//base32.ino
#include "ini.h"

#ifdef ESP32
#define nom "base32"
#include <WiFi.h>
#elif defined(ESP8266)
#define nom "base8266"
#include <ESP8266WiFi.h>
#endif

bool WifiEtat = true;
#include <FS.h>

#include <ArduinoOTA.h>
#include <LittleFS.h>
#include <FTPServer.h>
FTPServer ftpSrv(LittleFS);

#include <ESP_Mail_Client.h>
SMTPSession smtp;
Session_Config config;
void smtpCallback(SMTP_Status status);
#include "HeapStat.h"
HeapStat heapInfo;

unsigned long sentMillis = 0, tpsmillis = 0;

#include <time.h>
time_t maintenant;
struct tm* timeinfo;
int jour, jouractu;
int heure, heureactu;
char DateHeure[20];

void setup() {
  startserial();
  startWifi();
  if (WifiEtat) {
    //startMDNS();
    startOTA();  
    ftpSrv.begin("esp8266", "esp8266");

[... 303 lines cut ...]
      else
        Serial.println("Connected with no Auth.");
    }
  }
  if (erreur) EnreEvenement(Journal1 + smtp.errorReason());
  if (!MailClient.sendMail(&smtp, &message, false) && !erreur)
    ESP_MAIL_PRINTF("Error, Status Code: %d, Error Code: %d, Reason: %s", smtp.statusCode(), smtp.errorCode(), smtp.errorReason().c_str());
}

//ini.h
const char* ssid = "xxx";
const char* password = "xxx";
//const char* nom = "Mail8266";
const char* poste = "Essai esp32";

#define ficheEve "evene.csv"
#define SMTP_HOST "smtp.laposte.net"
#define SMTP_PORT 465
#define AUTHOR_EMAIL "xxx@laposte.net"
#define AUTHOR_PASSWORD "Axxx1!"
#define RECIPIENT_EMAIL "xxx@laposte.net"
#define AUTHOR_NOM "Test"
```

**Additional information and things you've tried:**

mobizt, 2023-10-18:
If the internet is not the problem, you should test by sending email without attachment.

If you can send email without attachment, this means the issue is memory allocation failure in filesystems that leads to file reading timeout (always zero byte read returns from filesystem).

Filesystem memory allocation failure in Core library issue starts from ESP32 Core v2.0.1.

mobizt, 2023-10-18:
[This example](https://github.com/mobizt/ESP-Mail-Client/blob/master/examples/SMTP/Send_Reuse_Session/Send_Reuse_Session.ino) shows that you can send and resend email many times with single log in.

You can also log in every time when sending the email by closing the session after email sent.

Michelbe30111, 2023-10-18:
This example works very well with an esp8266 but with an esp32 only sends one email, the following ones are in error

mobizt, 2023-10-18:
It's your device or network problem.

I test with many ERP32 modules (WROOM, WROVER) with all networks (built-in WiFi, Ethernet and GSM modem) and all work fine.

closed 2023-10-18 by mobizt