mobizt/ESP-Mail-Client #306: UTF-8 strings in sender/recipient name and subject items.

hunter-nl, 2023-11-05:
**Build tool used:**

- [ ] Arudino
- [ ] Platformio (Visual Studio Code plugin)
- [ X] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based
- [X ] ESP32-based

**Other Libraries That are used:**

**Description of problem:**
When sending a mail message with an UTF-8 string in Recipient name, Sender name or subject items. Then some SPAM filters act on it as it has not been correctly encoded.
How can I solve this?

Example:
    message.subject = F("Test sending Email from René");
    message.addRecipient("André", "a.test@test.com");

**Share code snippet to reproduce the issue:**
```yaml
Use Send_Text.ino example and replace the two lines given in the example above.
```

**Additional information and things you've tried:**

mobizt, 2023-11-05:
Your sketch file (.ino) or your source file (.cpp) should be saved as UTF-8 too.

closed 2023-11-05 by mobizt

hunter-nl, 2023-11-05:
That's saved in UTF-8. I found it should be conform [RFC2047](https://www.rfc-editor.org/rfc/rfc2047).

mobizt, 2023-11-05:
I think your sketch file is not UTF-8 encoded. I don't have any problem to hard code the string literal that contains UTF-8 or unicode characters.

Library does nothing with this as the UTF-8 character will be convert to variable bytes char arrays during code compilation by compiler.

mobizt, 2023-11-05:
Library is natively support [RFC2047](https://github.com/mobizt/ESP-Mail-Client/blob/master/src/extras/RFC2047.h) for parsing 
 for UTF8 for IMAP but encoding to UTF-8 is not necessary on the client as I described.

The SMTP server will handle UTF-8 encoding before sending.

In your case is invalid characters in your sketch which compiler can't be recognized the proper encoding as your sketch is not UTF-8 file.

hunter-nl, 2023-11-05:
The problem is that the mail headers is sending in UTF-8 and if the characters are not standard ASCII this will be wrong in the headers. This is not compliant with RFC2047 and SPAM filters can block these messages.
Example:
```message.addRecipient("René", "someaddress@test.com)");```

will result into the header To: item
```To: "RenÃ©" <someaddress@test.com>```

It should be (encoded in iso-8859-1, it can be in different encoding, this is just an example)
```To: =?iso-8859-1?b?UmVu6Q==?= <someaddress@test.com>```

mobizt, 2023-11-05:
As I said, your sketch is not actually UTF-8 encoded.

It's not library issue unless your issue.

Please keep in mind that this is the SMTP server responsibility instead of library and Email app.
`To: =?iso-8859-1?b?UmVu6Q==?= <someaddress@test.com>`

This is your problem of unicode character `é` in your header.
`message.addRecipient("René", "someaddress@test.com)");`

The GCC compiler knows how to convert `é` to multi-byte UTF-8 character.

Email app and library do not need to encoded this unicode char to sequences of `encoded-word` like `=?iso-8859-1?b?UmVu6Q==?=`.

Your problem is GCC does not know how to treat or convert your non-ascii chars that contains in your sketch into proper encoding bytes because your sketch is not UTF-8 encoded.

You should open your sketch file with good text editor e.g. Editplus or Notepad++ and save it as UTF-8 without BOM instead.

mobizt, 2023-11-05:
FYI.

It's not possible to include hundred megabytes of international character encoding tables in this library to convert non-ascii chars to `encoded-word`.

hunter-nl, 2023-11-05:
It is the responsibility of the client not server. Client has to conform the RFC 2047.
I'm using Visual Studio Code with Platformio and correct GCC. And file is saved correctly in UTF-8.
Python has a good mail header encoder. https://docs.python.org/3/library/email.header.html Just to illustrate you don't need hundred megabytes of international character encoding tables ;-)

mobizt, 2023-11-05:
You are misunderstanding about this.

The OS handles this instead of application.

hunter-nl, 2023-11-05:
Now the OS is the cause? Come on... I'm going to abandon your library.

mobizt, 2023-11-05:
It's up to you and your bok.
This library developed under PIO and I can send any Chinese, Russian and all languages email to SMTP server without problem.