mobizt/ESP-Mail-Client #309: Arduino Uno R4; Failure to connect to server using provided examples

AustinFoss, 2023-12-04:
**Build tool used:**

- [x] Arudino
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based
- [ ] ESP32-based
- [x] Arduino Uno R4 (ESP32*?)

**Other Libraries That are used:**

None

**Description of problem:**
When trying to get familiar with this library using my R4 I'm unable to get past the first hurdle, connecting to an IMAP server, in this case Gmail. For instance I went and copied the code from `examples/IMAP/Get_Quota` into a fresh sketch: 

**Share code snippet to reproduce the issue:**
```yaml
#include <Arduino.h>
#if defined(ESP32) || defined(ARDUINO_RASPBERRY_PI_PICO_W)
#include <WiFi.h>
#elif defined(ESP8266)
#include <ESP8266WiFi.h>
#elif __has_include(<WiFiNINA.h>)
#include <WiFiNINA.h>
#elif __has_include(<WiFi101.h>)
#include <WiFi101.h>
#elif __has_include(<WiFiS3.h>)
#include <WiFiS3.h>
#endif

#include <ESP_Mail_Client.h>

#define WIFI_SSID "<ssid>"
#define WIFI_PASSWORD "<wifi_password>"

#define IMAP_HOST "imap.gmail.com"
#define IMAP_PORT 993

#define AUTHOR_EMAIL "<my_email@gmail.com>"
#define AUTHOR_PASSWORD "<app_password>"

IMAPSession imap;

#if defined(ARDUINO_RASPBERRY_PI_PICO_W)
WiFiMulti multi;
#endif

void setup()
{

    Serial.begin(115200);

#if defined(ARDUINO_ARCH_SAMD)
    while (!Serial)
        ;
#endif

    Serial.println();

#if defined(ARDUINO_RASPBERRY_PI_PICO_W)
    multi.addAP(WIFI_SSID, WIFI_PASSWORD);
    multi.run();
#else
    WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
#endif

    Serial.print("Connecting to Wi-Fi");

#if defined(ARDUINO_RASPBERRY_PI_PICO_W)
    unsigned long ms = millis();
#endif

    while (WiFi.status() != WL_CONNECTED)
    {
        Serial.print(".");
        delay(300);
#if defined(ARDUINO_RASPBERRY_PI_PICO_W)
        if (millis() - ms > 10000)
            break;
#endif
    }
    Serial.println();
    Serial.print("Connected with IP: ");
    Serial.println(WiFi.localIP());
    Serial.println();

    MailClient.networkReconnect(true);

#if defined(ARDUINO_RASPBERRY_PI_PICO_W)
    MailClient.clearAP();
    MailClient.addAP(WIFI_SSID, WIFI_PASSWORD);
#endif

    imap.debug(1);
    Session_Config config;

    /* Set the session config */
    config.server.host_name = IMAP_HOST;
    config.server.port = IMAP_PORT;
    config.login.email = AUTHOR_EMAIL;
    config.login.password = AUTHOR_PASSWORD;

    IMAP_Data imap_data;

    if (!imap.connect(&config, &imap_data))
    {
        MailClient.printf("Connection error, Error Code: %d, Reason: %s\n", imap.errorCode(), imap.errorReason().c_str());
        return;
    }

    if (imap.isAuthenticated())
        Serial.println("Successfully logged in.");
    else
        Serial.println("Connected with no Auth.");

    IMAP_Quota_Root_Info info;

    Serial.println("\nGet quota root...");

    if (!imap.getQuota("", &info))
    {
        Serial.println("Get quota root failed");
    }
    else
    {
        MailClient.printf("Quota root: %s, Resource name: %s, Usage (k): %d, Limit (k): %d\n", info.quota_root.c_str(), info.name.c_str(), (int)info.usage, (int)info.limit);
    }

    IMAP_Quota_Roots_List quota_roots;

    Serial.println("\nGet quota roots list for INBOX folder...");

    if (!imap.getQuotaRoot("INBOX", &quota_roots))
    {
        Serial.println("\nGet quota roots list failed");
    }
    else
    {
        for (size_t i = 0; i < quota_roots.size(); i++)
            MailClient.printf("%d, Quota root: %s, Resource name: %s, Usage (k): %d, Limit (k): %d\n", (int)i + 1, quota_roots[i].quota_root.c_str(), quota_roots[i].name.c_str(), (int)quota_roots[i].usage, (int)quota_roots[i].limit);
    }

    Serial.println("\nSet quota root...");

    info.name = "STORAGE";
    info.limit = 1024 * 1024; // 1048576 k or 1 G.

    if (!imap.setQuota("", &info))
    {
        Serial.println("\nSet quota root failed");
    }
    else
    {
        Serial.println("\nSet quota root success");
    }

    MailClient.printf("Free Heap: %d\n", MailClient.getFreeHeap());
}

void loop()
{
}
```

**Additional information and things you've tried:**
I've switched things up by changing the port from 993 to 143, google's app password is 16 characters in 4 chunks separated by spaces and so I've tried with and without spaces, tried my normal log in password as well, yes 2FA is activated for my gmail account and so is IMAP.

The output in the console from `imap.debug(1);` looks like:
```
> C: ESP Mail Client v3.4.14
> C: connecting to IMAP server
> C: Host > imap.gmail.com
> C: Port > lu
! E: unable to connect to server
Connection error, Error Code: -200, Reason: unable to connect to server
```

mobizt, 2023-12-04:
Gmail does not support port 143 for IMAP, only 993 was supported.

mobizt, 2023-12-04:
Arduino UNO R4 WiFi has limited memory, and the stack overflow error can be occurred when connecting to IMAP server because the large memory allocation (16k+) required in `BearSSL` for full frame SSL supports.