mobizt/ESP-Mail-Client #312: Having a problem coding with this library

luciepell, 2024-01-05:
**Build tool used:**

- [ ] Arudino 
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based
**- [ ] ESP32-based**
I used the ESP32 board

**Other Libraries That are used:**
Paste contents of platformio.ini / libraries or list some of the libraries used.
-->
#include <ESP32MailHTTPClient.h>
#include <WiFi.h>

**Description of problem:**
Essentially, I am trying to code so that when a button is pressed (I have a setup on a breadboard) an email is sent to the contact. I am trying to use ESP32 Mail Client to accomplish this. However, the same error message keeps appearing. 
This is the error message I keep receiving: 

In file included from /Users/luciepelletier/Documents/Arduino/libraries/ESP32_Mail_Client/src/ESP32_MailClient.cpp:37:
/Users/luciepelletier/Documents/Arduino/libraries/ESP32_Mail_Client/src/ESP32_MailClient.h:39:10: fatal error: ETH.h: No such file or directory
 #include <ETH.h>
          ^~~~~~~
compilation terminated.

exit status 1

Compilation error: exit status 1

I try to add the  #include <ETH.h>, and the code is still not working and the same error message regarding  #include <ETH.h> keeps appearing. 
I read that this library supports wifi so it is unusual that the  #include <ETH.h>
error keeps occurring. 

**Share code snippet to reproduce the issue:**
```yaml
PASTE .cpp / .ino code here

#include <ESP32MailHTTPClient.h>
#include <WiFi.h>

// Wi-Fi credentials
const char* ssid = "my SSID";
const char* password = "my SSID password";

// SMTP server settings
const char* smtpServer = "smtp.gmail.com";
const int smtpPort = 587;  // Change the port if needed
const char* smtpUsername = "my SMPT email";
const char* smtpPassword = "my SMPT password";
const char* senderEmail = "my sender email";
const char* recipientEmail = "my receiving email";  // Change to the recipient's email address

// Pin for the button
const int buttonPin = 5;  // Change to the pin where your button is connected

void setup() {
  Serial.begin(115200);

  // Connect to Wi-Fi
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.println("Connecting to WiFi...");
  }
  Serial.println("Connected to WiFi");

  // Setup button
  pinMode(buttonPin, INPUT_PULLUP);

  Serial.println("Setup complete");
}

void loop() {
  if (digitalRead(buttonPin) == LOW) {
    Serial.println("Button pressed, sending email...");

    // Customize your email content here
    String emailSubject = "subject content";
    String emailMessage = "content";

    // Create an instance of the ESP32MailHTTPClient
    ESP32MailHTTPClient mail;

    // Set SMTP server details
    mail.setSMTPServer(smtpServer, smtpPort);
    mail.setLogin(smtpUsername, smtpPassword);

    // Set email parameters
    mail.setSender(senderEmail);
    mail.setRecipient(recipientEmail);
    mail.setSubject(emailSubject);
    mail.setMessage(emailMessage);

    // Send the email
    if (mail.send()) {
      Serial.println("Email sent successfully");
    } else {
      Serial.println("Error sending email");
    }

    // Wait for the button to be released to avoid multiple emails for one press
    while (digitalRead(buttonPin) == LOW) {
      delay(100);
    }
  }

  delay(100);  // Adjust the delay based on your requirements
}

**Additional information and things you've tried:**
I've honestly been trying to figure out the #include <ETH.h> error message for a while, I keep getting dragged in circles where the  #include <ETH.h> doesn't work but the rest of my code won't work without it. I've downloaded the Ethernet library but the code doesn't support one or the other or both at a time. 

<!-- LEAVE THIS LINE AS-IS AND DON'T DELETE IT, OTHERWISE THE ISSUE WILL BE CLOSED AUTOMATICALLY. -->

mobizt, 2024-01-05:
That library has been deprecated for many years ago and no further support.

closed 2024-01-05 by mobizt

luciepell, 2024-01-05:
Which mail client library should I use that is compatible with #include <ETH.h> ?

mobizt, 2024-01-05:
You are mentioned the deprecated library (ancestor of this library) which is not supported and should not ask here

mobizt, 2024-01-05:
See [this](https://github.com/mobizt/ESP32-Mail-Client/issues/81).