mobizt/ESP-Mail-Client #313: Some methods appear to be private unnecessarily

jksemple, 2024-01-05:
**Build tool used:**

Arduino

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP32-based

**Other Libraries That are used:**

Not relevant

**Description of problem:**

A number of useful methods declared in ESP_Mail_Client.h appear to have been declared as private unnecessarily.
In particular (for me), ESP_Mail_Client::mimeFromFile(const char* name, MB_String &mime);

The private access specifier at line#1072 has been applied to all subsequent methods.

**Share code snippet to reproduce the issue:**
```yaml
Not relevant

```

**Additional information and things you've tried:**

I forked the repo and added a public: access specifier at line 1334 and I can now use ::mimeFromFile() as required.

I hesitate to propose a PullRequest as I don't know where public / private specifiers ought to be added to open up the interface as was probably intended.

mobizt, 2024-01-05:
Yes, that method is only intended for private (internal) use.

It takes `MB_String` as parameter which is library specific string class and may not be suitable to make it public access.

Instead of editing the library, you can create your own use function with Arduino `String` like this.

```cpp
void mimeFromFile(const char *name, String &mime)
{
    String ext = name;
    int p = ext.lastIndexOf('.');
    if (p != -1)
    {
        ext = ext.substring(p, ext.length() - p);
        if (ext.length() > 0)
            getMIME(ext.c_str(), mime);
    }
}

void getMIME(const char *ext, String &mime)
{
    mime = "";
    for (int i = 0; i < esp_mail_file_extension_maxType; i++)
    {
        if (strcmp_P(ext, mimeinfo[i].endsWith) == 0)
        {
            mime = mimeinfo[i].mimeType;
            break;
        }
    }
}
```

jksemple, 2024-01-05:
Why are the MB_String versions of these methods in the library at all then? They don't get used internally as far as I can see. I wouldn't need to re-implement them externally if the library used them internally to infer mime types automatically from supplied file names.

mobizt, 2024-01-05:
It used here.
https://github.com/mobizt/ESP-Mail-Client/blob/008b4f1ed3c88c484b37946b0d4976c1b8028d3f/src/ESP_Mail_SMTP.h#L2505

I don't know what is your problem?

closed 2024-01-05 by mobizt

mobizt, 2024-01-05:
You can see the `MB_String` header description what it is used for.

### I don't have to describe all as it is internally use and this library already provided full of code comment and documentation which if you interest it, you should read it.