mobizt/ESP-Mail-Client #316: SMTP send mail failed M5Stack W5500 unit

RIES-endo, 2024-01-26:
**Build tool used:**

- [* ] Arudino
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based
- [ *] ESP32-based

**Other Libraries That are used:**
M5Core2, Ethernet

**Description of problem:**
M5Stack reboots before sending email. 
it seems to connection of Ethernet and SMTP sever are OK.
Here's output of serial monitor:

! I: Resetting Ethernet Board...
! I: Starting Ethernet connection...
! I: Connected with IP 192.168.11.202
#### Connecting to SMTP server...
> C: ESP Mail Client v3.4.15
> C: connecting to SMTP server
> C: Host > smtp.*.jp
> C: Port > 587

#### SMTP server connected
> C: SMTP server connected, wait for greeting...
< S: 220 smtp1.*.jp ESMTP Welcome to HINES mailgateway
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40093994  PS      : 0x00060c33  A0      : 0x80094113  A1      : 0x3ffca4f0  
A2      : 0x3f800014  A3      : 0xffffffff  A4      : 0xfffffffc  A5      : 0xbaad5678  
A6      : 0x00060020  A7      : 0x00000001  A8      : 0x3f800bd4  A9      : 0x00000004  
A10     : 0x3f800278  A11     : 0x00000000  A12     : 0x3f800bdc  A13     : 0x3f800078  
A14     : 0x00000005  A15     : 0x00000005  SAR     : 0x0000001b  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x0000000b  LBEG    : 0x4008bd0d  LEND    : 0x4008bd1d  LCOUNT  : 0xffffffee  

Backtrace: 0x40093991:0x3ffca4f0 0x40094110:0x3ffca510 0x4009424a:0x3ffca530 0x40084301:0x3ffca550 0x40094679:0x3ffca570 0x400db989:0x3ffca590 0x400e3e8a:0x3ffca5b0 0x400eb297:0x3ffca690 0x400eb321:0x3ffca6d0 0x400eb368:0x3ffca6f0 0x400d2aaa:0x3ffca710 0x40102776:0x3ffcaab0

**Share code snippet to reproduce the issue:**
```yaml
PASTE .cpp / .ino code here
#include <Ethernet.h>
#include <M5Core2.h>
#include <ESP_Mail_Client.h>

#define WIZNET_RESET_PIN 13  // Connect W5500 Reset pin to GPIO 26 of ESP32
#define WIZNET_CS_PIN 26     // Connect W5500 CS pin to GPIO 5 of ESP32
#define WIZNET_MISO_PIN 19   // Connect W5500 MISO pin to GPIO 19 of ESP32
#define WIZNET_MOSI_PIN 23   // Connect W5500 MOSI pin to GPIO 23 of ESP32
#define WIZNET_SCLK_PIN 18   // Connect W5500 SCLK pin to GPIO 18 of ESP32

byte Eth_MAC[] = { 0x02, 0xF0, 0x0D, 0xBE, 0xEF, 0x01 };
void smtpCallback(SMTP_Status status);
//ESP_Mail_Session config;
Session_Config config;

void setup() {
  Serial.begin(115200);
  M5.begin();
  config.server.host_name = "smtp.*.jp";
  config.server.port = 587;
  config.login.email = "***";
  config.login.password = "***";
  config.login.user_domain = "";

  // Declare the SMTP_Message class variable to handle to message being transport
  SMTP_Message message;
  SMTPSession smtp;

  // Set the message headers
  message.sender.name = F("ESP");
  message.sender.email = F("***");
  message.subject = F("ESP Test Email");
  message.addRecipient(F(""), F("***"));

  // Set the message content
  message.text.content = F("This is simple plain text message");

  // Set debug option
  smtp.debug(1);

  // Set the callback function to get the sending results
  smtp.callback(smtpCallback);

  EthernetClient eth_client;

  smtp.setEthernetClient(&eth_client, Eth_MAC, WIZNET_CS_PIN, WIZNET_RESET_PIN);

  // Connect to the server with the defined session and options
  smtp.connect(&config);

  M5.Lcd.println(Ethernet.localIP());

  // Start sending Email and close the session

  MailClient.sendMail(&smtp, &message);

  if (!MailClient.sendMail(&smtp, &message))
    Serial.println("Error sending Email, " + smtp.errorReason());
}

void loop() {
}

void smtpCallback(SMTP_Status status) {

  Serial.println(status.info());

  if (status.success()) {
    M5.Lcd.println("OK");
  }
}

```

**Additional information and things you've tried:**

mobizt, 2024-01-26:
Before posting issue, you should debug it first.

Why you think to use EthernetClient locally inside the setup?

```
EthernetClient eth_client;

smtp.setEthernetClient(&eth_client, Eth_MAC, WIZNET_CS_PIN, WIZNET_RESET_PIN);
```

The function `smtp.setEthernetClient` takes the pointer of `eth_client` which will be later used, and it will not be existed outside the setup.

closed 2024-01-26 by mobizt

RIES-endo, 2024-01-26:
Thank you for your reply.
I try to use your example code, but still I can not connect SMTP sever. why ?

```yaml
/**
 * Created by K. Suwatchai (Mobizt)
 *
 * Email: suwatchai@outlook.com
 *
 * Github: https://github.com/mobizt/ESP-Mail-Client
 *
 * Copyright (c) 2023 mobizt
 */

/**
 * This example shows how to send Email using EthernetClient.
 *
 * This example used ESP32 and WIZnet W5500 Ethernet module.
 *
 * For ESP32 and LAN8720 see examples/SMTP/Ethernet/ESP32/Send_Text.ino.
 *
 * ESP32 Arduino SDK native Ethernet using ETH.h is currently support Ethernet PHY chips included the following
 * LAN8720, TLK101, RTL8201, DP83848, DM9051, KSZ8041 and KSZ8081.
 *
 */

/** Note for library update from v2.x.x to v3.x.x.
 *
 *  Struct data names changed
 *
 * "ESP_Mail_Session" changes to "Session_Config"
 * "IMAP_Config" changes to "IMAP_Data"
 *
 * Changes in the examples
 *
 * ESP_Mail_Session session;
 * to
 * Session_Config config;
 *
 * IMAP_Config config;
 * to
 * IMAP_Data imap_data;
 */

// https://github.com/arduino-libraries/Ethernet
#include <Ethernet.h>
// For using other Ethernet library that works with other Ethernet module, 
// the following build flags or macros should be assigned in src/ESP_Mail_FS.h or your Custom_ESP_Mail_FS.h.
// ESP_MAIL_ETHERNET_MODULE_LIB and ESP_MAIL_ETHERNET_MODULE_CLASS
// See src/ESP_Mail_FS.h for detail.

#include <M5Core2.h>
#include <ESP_Mail_Client.h>

#define SMTP_HOST "smtp.*"
#define SMTP_PORT 587

#define AUTHOR_EMAIL "***"
#define AUTHOR_PASSWORD "****"
#define RECIPIENT_EMAIL "****"

#define WIZNET_RESET_PIN 13 // Connect W5500 Reset pin to GPIO 26 of ESP32
#define WIZNET_CS_PIN 26     // Connect W5500 CS pin to GPIO 5 of ESP32
#define WIZNET_MISO_PIN 19  // Connect W5500 MISO pin to GPIO 19 of ESP32
#define WIZNET_MOSI_PIN 23  // Connect W5500 MOSI pin to GPIO 23 of ESP32
#define WIZNET_SCLK_PIN 18  // Connect W5500 SCLK pin to GPIO 18 of ESP32

unsigned long sentMillis = 0;

const int analog_pin = 34;
[... 126 lines cut ...]
> C: connecting to SMTP server
> C: Host > smtp.hines.hokudai.ac.jp
> C: Port > 587
ets Jul 29 2019 12:21:46

rst:0x8 (TG1WDT_SYS_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4

> C: Port > 587
ets Jul 29 2019 12:21:46

rst:0x8 (TG1WDT_SYS_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4

mobizt, 2024-01-26:
Your device was reset by unknown reason.

The hardware reset was not triggered by wdt and library.

I don't have any problem with my own test.

The Ethernet library tested is Arduino Ethernet library v2.0.2.

If you are using third party Ethernet library, change it to use official Ethernet library that provided by Arduino and ESP32 Arduino dev team.

mobizt, 2024-01-26:
I see that you open the issue at M5Core2 repo.

The code you provided there and the error you post there are not relevant.

The error `Guru Meditation Error: Core 1 panic'ed (StoreProhibited). Exception was unhandled.` is because dangling pointer due to you defined the `EthernetClient eth_client` locally in the sutup.

While your current issue after trying with the library example code that you provide on M5Core2 repo is because wdt reset and you should post the debug message below at that repo instead.

```
rst:0x8 (TG1WDT_SYS_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
```

dinosauria123, 2024-01-26:
Thank you for your comment.
I have added new debug message.

mobizt, 2024-01-26:
Note that, the library was well wdt feed inside the library.

The possibility problem is hardware issue that causes the too long blocking Ethernet library operation which may causes the wdt reset.

Another possibility is your ESP32 power failure.

Anyway you have to update ESP32 Arduino Core SDK to latest version.