mobizt/ESP-Mail-Client #319: PlatformIO error for String operators: more than one operator "+" matches these operands

Nagymadar, 2024-02-01:
Using PlatformIO it compiles successfully, but String operators indicates error for any String operators due to MB_String:

**Code:**
```yaml
  String str1 = "string1";
  String str2 = "They are a simple Strings " + str1;
```

**Error message:**
```
more than one operator "+" matches these operands
function "operator+(const StringSumHelper &lhs, const String &rhs)" (declared at line 185 of "C:\Users\x\.platformio\packages\framework-arduinoespressif32\cores\esp32\WString.h")
function "operator+(MB_String &&lhs, const MB_String &rhs)" (declared at line 2048 of "C:\Users\detre\.platformio\packages\framework-arduinoespressif32\libraries\ESP-Mail-Client-master\src\extras\MB_String.h")
operand types are: const char [27] + String
function "operator+(MB_String &&lhs, char rhs)" does not match because argument #2 does not match parameter
...
```

**Platformio.ini** (no Global lib copy):

```yaml
[platformio]
default_envs = upload

[env:upload]
platform = espressif32
board = esp32dev
framework = arduino
monitor_speed = 115200
monitor_filters = direct, esp32_exception_decoder
build_flags = 
    -mfix-esp32-psram-cache-issue

lib_deps =
  mobizt/ESP Mail Client

build_type = release
upload_port = COM6
upload_speed = 921600

```
I used "disable error squiggles" to make errors disappear. I have a rather large code, where it gives tons of errors. Recommendation?

mobizt, 2024-02-01:
It's VSCode issue not library issue (just IntelliSense syntax error) and not PlatformIO issue and it is related to Strung class instead of MB_String.

You can search PlatformIO community forum or search google and for this issue.

closed 2024-02-01 by mobizt

Nagymadar, 2024-02-01:
It is not only me. Concerning IntelliSense I found this, also related to MB_String: 

https://community.platformio.org/t/false-problem-in-intellisense-in-vscode-more-than-one-operator-matches-these-operands/32347/3

I can't find any resolution to this. Could you help?

mobizt, 2024-02-01:
I can't remember that this issue existed on ESP32 or ESP8266 PIO platforms.

For me it gone for a while after the PlatformIO and the Arduino core updated but can't remember when.

You should update both PIO and Arduino Core.

Nagymadar, 2024-02-02:
I am still struggling with this. Removing ESP-Mail-Client solves these errors in my project. So it is the library does something.

```
MB_String.h:879:56: error: call of overloaded 'appendNum(int&)' is ambiguous
MB_String.h:881:57: error: call of overloaded 'appendNum(short unsigned int&)' is ambiguous
MB_String.h:889:53: error: call of overloaded 'appendNum(bool&)' is ambiguous
MB_String.h:1530:17: error: ambiguous overload for 'operator+=' (operand types are 'MB_String' and 'int')
MB_Time.h:298:13: error: ambiguous overload for 'operator+=' (operand types are 'MB_String' and 'int')
etc...
```

mobizt, 2024-02-02:
I don't have any problem with that as library developed in PlatformIO and intensively tested in many platforms.

It looks like your PlatformIO installation problem. Please try to uninstall and reinstall PlatformIO and the platform SDK.

Nagymadar, 2024-02-03:
I reinstalled it. Even tried Arduino as EspIdf component. Seems IDF is more strict in C++ error checking. Same errors. In total 20-30, some examples:

```
MB_String.h[119]: expected nested-name-specifier before 'enable_if'
MB_String.h[553]: call of overloaded 'appendNum(bool&)' is ambiguous
MB_String.h[781]: call of overloaded 'appendNum(bool&)' is ambiguous
...
MB_String.h[2037]:optimization attribute on 'MB_String operator+(const MB_String&, const MB_String&)' follows definition but the attribute doesn't match [-Wattributes]
...
```

Nagymadar, 2024-02-03:
The overload candidates for most of them:
```
.pio/libdeps/esp32dev/ESP Mail Client/src/extras/MB_Time.h:298:13: error: ambiguous overload for 'operator+=' (operand types are 'MB_String' and 'int')
       tzStr += 0;
       ~~~~~~^~~~
In file included from .pio/libdeps/esp32dev/ESP Mail Client/src/./extras/MB_FS.h:43,
                 from .pio/libdeps/esp32dev/ESP Mail Client/src/extras/RFC2047.h:13,
                 from .pio/libdeps/esp32dev/ESP Mail Client/src/ESP_Mail_Client.h:39,
                 from include/email.h:3,
                 from include/menus\menu_debug.h:10,
                 from include/menus\menu_main.h:10,
                 from src/main.cpp:12:
.pio/libdeps/esp32dev/ESP Mail Client/src/./extras/MB_String.h:692:16: note: candidate: 'MB_String& MB_String::operator+=(StringSumHelper)'
     MB_String &operator+=(StringSumHelper rval)
                ^~~~~~~~
.pio/libdeps/esp32dev/ESP Mail Client/src/./extras/MB_String.h:700:16: note: candidate: 'MB_String& MB_String::operator+=(const __FlashStringHelper*)'
     MB_String &operator+=(const __FlashStringHelper *str)
                ^~~~~~~~
.pio/libdeps/esp32dev/ESP Mail Client/src/./extras/MB_String.h:728:16: note: candidate: 'MB_String& MB_String::operator+=(const string&)'
     MB_String &operator+=(const std::string &rhs)
                ^~~~~~~~
.pio/libdeps/esp32dev/ESP Mail Client/src/./extras/MB_String.h:734:16: note: candidate: 'MB_String& MB_String::operator+=(const String&)'
     MB_String &operator+=(const String &rhs)
                ^~~~~~~~
.pio/libdeps/esp32dev/ESP Mail Client/src/./extras/MB_String.h:753:16: note: candidate: 'MB_String& MB_String::operator+=(const MB_String&)'
     MB_String &operator+=(const MB_String &rhs)
                ^~~~~~~~
.pio/libdeps/esp32dev/ESP Mail Client/src/./extras/MB_String.h:759:16: note: candidate: 'MB_String& MB_String::operator+=(const char*)'
     MB_String &operator+=(const char *cstr)
                ^~~~~~~~
.pio/libdeps/esp32dev/ESP Mail Client/src/./extras/MB_String.h:773:16: note: candidate: 'MB_String& MB_String::operator+=(char)'
     MB_String &operator+=(char cstr)
                ^~~~~~~~
.pio/libdeps/esp32dev/ESP Mail Client/src/./extras/MB_String.h:779:16: note: candidate: 'MB_String& MB_String::operator+=(bool)'
     MB_String &operator+=(bool value)
                ^~~~~~~~
.pio/libdeps/esp32dev/ESP Mail Client/src/./extras/MB_String.h:791:16: note: candidate: 'MB_String& MB_String::operator+=(mb_string::MB_StringPtr)'
     MB_String &operator+=(MB_StringPtr ptr)
```

mobizt, 2024-02-03:
I will say this for the **last time** and will limit the conversation on this user error.

I don't have this such MB_String compilation error from intensively test in 5 years for AVR, STM32, SAMD, Teensy, ESP8266, ESP32 and RaspberryPi Pico on PlatformIO and Arduino IDE (full error and warnings compile level)

The library has no error when compile with the [CI test](https://github.com/mobizt/ESP-Mail-Client/actions) too.

If you know how to debug the information from the compiler errors especially "ambiguous", you will know why. 

The different sources path of the same headers "MB_String,h" in this case from different location leads this compilation error.

The library examples compilation must be ok. If you still get the error when just compiling the library examples, your IDE cache will be the problem.

The library and IDE installation should not be in the sandbox, virtual folder, OS strict location or cloud storage folder (OneDrive, Google Drive).

mobizt, 2024-02-03:
If you don't understand what I mean for **"ambiguous"**

Please look at the following folders from the compilation errors.

`.pio/libdeps/esp32dev/ESP Mail Client/src/extras`
`.pio/libdeps/esp32dev/ESP Mail Client/src/./extras`

How you create the later one (`.pio/libdeps/esp32dev/ESP Mail Client/src/./extras`) as it is does not created by PIO?