mobizt/ESP-Mail-Client #323: ESP crashes

BrianG-coder, 2024-03-02:
**Build tool used:**

- [X ] Arudino
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based
- [ X] ESP32-based

**Other Libraries That are used:**

**Description of problem:**
ESP crashes with the following backtrace
Backtrace: 0x40092727:0x3ffe1f20 0x400d25d1:0x3ffe1f40 0x400e2f55:0x3ffe1f60 0x400e8761:0x3ffe2040 0x400eaf71:0x3ffe2080 0x400eb9cd:0x3ffe2140 0x400eba31:0x3ffe2180 0x400d4663:0x3ffe21a0

From the backtrace the problem is coming from line 1057 (ESP_MAIL_CLIENT.cpp): 
    if (len > 0 && (ovf || ovfBuf.length() > 0))
      ovfBuf += buf;

**Share code snippet to reproduce the issue:**
At present, I have not found a simple reproducible snippet.  I've only observed this in a complicated multi-threaded environment.  
1) The crashes coincide with the following error:
 #### Error, fail to set up the SSL/TLS structure
2) The crashing seems to get worse when there is I2C communicate happening at the same time as the SMTP. 
3) It only seems to happen after repeated messages.  

I believe there is something other then the ESP_Mail_Client library causing a communication error.  

**Additional information and things you've tried:**
Changing the code at line 1056 to the following stops the esp from crashing:
    if (len > 0 && (ovf || ovfBuf.length() > 0))
      ovfBuf += buf;
     Serial.print("ovfBuf Length"); 
     Serial.println(ovfBuf.length()); 
     if (ovfBuf.length()>10000)
     {
          Serial.print("AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA"); 
        return false;
     }

mobizt, 2024-03-02:
It's not a library issue.

You have to manage your memory usage.

closed 2024-03-02 by mobizt

BrianG-coder, 2024-03-03:
Thanks, for your quick reply.  I'll do some more investigation.  Can you please tell me what the ovfBuf is used for?    I see it creates an overflow buffer error, but why is the overflow data saved?

mobizt, 2024-03-03:
It's not what you should concern.

It used when response buffer size (1024) is too small and should be expanded later and overflow buffer is temporary data to be copied.

You have to manage memory usage especially stack.

The error is about wdt reset because of memory allocation failure or stack overflow and nothing is related to the code you are looking for.

BrianG-coder, 2024-03-05:
The library is trying to read an empty buffer.  The result is that if keeps writing 0xFFs to memory until the ESP crashes.

In my test I modified the library with some debugging code.   Here are some snippets to show what was added.

int ESP_Mail_Client::readLine(ESP_Mail_TCPClient *client, char *buf, int bufLen, bool withLineBreak, int &count, bool &ovf, unsigned long timeoutSec, bool &isTimeout)
{
  int ret = -1;
  char c = 0;
  char _c = 0;
  int idx = 0;
  ovf = idx >= bufLen;
  bool lineBreak = false;
  isTimeout = false;

  unsigned long ms = millis();
      Serial.print("client->available()");
      Serial.println(client->available()); 

bool ESP_Mail_Client::readResponse(T sessionPtr, char *buf, int bufLen, int &readLen, bool withLineBreak, int &count, MB_String &ovfBuf)
{
  bool ovf = false, isTimeout = false;
  unsigned long timeoutSec = TCP_CLIENT_DEFAULT_TCP_TIMEOUT_SEC;
      Serial.print("bufLen: ");
      Serial.println(bufLen);
  do
  {
    timeoutSec = sessionPtr->client.tcpTimeout();
    int len = readLine(&(sessionPtr->client), buf, bufLen, withLineBreak, count, ovf, timeoutSec, isTimeout);
    Serial.print("len: ");
    Serial.println(len);
    readLen += len;
    if (len > 0 && (ovf || ovfBuf.length() > 0))
      ovfBuf += buf;

      Serial.print("ovfBuf size: ");
      Serial.println(ovfBuf.length());
      Serial.print("buf: ");
      //Serial.println(buf);
      Serial.print(buf[0], HEX);
      Serial.print(buf[1], HEX);
      Serial.print(buf[2], HEX);
      Serial.print(buf[3], HEX);
      Serial.print(buf[4], HEX);
      Serial.println(buf[5], HEX);

Here is my output:
18:56:51.644 -> #### Sending message header...
18:56:51.644 -> 0
18:56:51.644 -> error code: 0
18:56:51.677 -> DEBUG Line: 144 577
18:56:51.710 -> bufLen: 1024
18:56:51.743 -> client->available()23
18:56:53.799 -> len: 1023
18:56:53.799 -> ovfBuf size: 3
18:56:53.799 -> buf: 153301236
18:56:53.799 -> client->available()0
18:56:55.854 -> len: 1023
18:56:55.854 -> ovfBuf size: 1026
18:56:55.854 -> buf: FFFFFFFFFFFF
18:56:55.854 -> client->available()0
18:56:57.910 -> len: 1023
18:56:57.910 -> ovfBuf size: 2049
18:56:57.910 -> buf: FFFFFFFFFFFF
18:56:57.910 -> client->available()0
18:56:59.967 -> len: 1023
18:56:59.967 -> ovfBuf size: 3072
18:56:59.967 -> buf: FFFFFFFFFFFF
18:56:59.967 -> client->available()0
18:57:01.990 -> len: 1023
18:57:01.990 -> ovfBuf size: 4095
18:57:01.990 -> buf: FFFFFFFFFFFF
18:57:01.990 -> client->available()0
18:57:04.045 -> len: 1023
18:57:04.045 -> ovfBuf size: 5118
18:57:04.078 -> buf: FFFFFFFFFFFF
18:57:04.078 -> client->available()0
18:57:06.101 -> len: 1023
18:57:06.101 -> ovfBuf size: 6141
18:57:06.134 -> buf: FFFFFFFFFFFF
18:57:06.134 -> client->available()0
18:57:08.158 -> len: 1023
18:57:08.158 -> ovfBuf size: 7164
18:57:08.158 -> buf: FFFFFFFFFFFF
18:57:08.191 -> client->available()0
18:57:10.214 -> len: 1023
18:57:10.214 -> ovfBuf size: 8187
18:57:10.248 -> buf: FFFFFFFFFFFF
18:57:10.248 -> client->available()0
18:57:12.270 -> len: 1023
18:57:12.270 -> ovfBuf size: 9210
18:57:12.270 -> buf: FFFFFFFFFFFF
18:57:12.270 -> client->available()0
18:57:14.292 -> len: 1023
18:57:14.325 -> ovfBuf size: 10233
18:57:14.325 -> buf: FFFFFFFFFFFF
18:57:14.325 -> client->available()0
18:57:16.347 -> len: 1023
18:57:16.380 -> ovfBuf size: 11256
18:57:16.380 -> buf: FFFFFFFFFFFF
18:57:16.380 -> client->available()0
18:57:18.404 -> len: 1023
18:57:18.404 -> ovfBuf size: 12279
18:57:18.437 -> buf: FFFFFFFFFFFF
18:57:18.437 -> client->available()0
18:57:20.462 -> len: 1023
18:57:20.462 -> ovfBuf size: 13302
18:57:20.462 -> buf: FFFFFFFFFFFF
18:57:20.495 -> client->available()0
18:57:22.519 -> len: 1023
18:57:22.519 -> ovfBuf size: 14325
18:57:22.519 -> buf: FFFFFFFFFFFF
18:57:22.519 -> client->available()0
18:57:24.608 -> len: 1023
18:57:24.608 -> ovfBuf size: 15348
18:57:24.608 -> buf: FFFFFFFFFFFF
18:57:24.608 -> client->available()0
18:57:26.631 -> len: 1023
18:57:26.631 -> ovfBuf size: 16371
18:57:26.631 -> buf: FFFFFFFFFFFF

This output continues until the ESP crashes.

I think this line in int ESP_Mail_Client::readLine may be the problem:
  while (idx < bufLen && client->connected() && (!lineBreak || client->available() /* data may not available sometimes */))

In the case where there is no line break, it reads forever.

did you mean this?:
  while (idx < bufLen && client->connected() && !lineBreak && client->available() /* data may not available sometimes */))

mobizt, 2024-03-05:
There is no issue in [all my platforms and devices](https://github.com/mobizt/ESP-Mail-Client?tab=readme-ov-file#supported-devices) testing.

What you are trying to do is not related to library issue.

I will say this finally that you have the memory problem, and you should solve it yourself.

mobizt, 2024-03-05:
From your information provided I only know that the sever you are trying to connect is not SMTP server or its response are not valid in smtp protocols.

The response of server challenge in SMTP protocol will never be large as 16k.

mobizt, 2024-03-05:
There is no valid SMTP status code found from your server response and it continue to read the garbage from TCP client buffer and fill it to response buffer which continued to grow up with junk then crash in never ending loop until memory allocation failure.

BrianG-coder, 2024-03-05:
Did you see that client->available()= 0 while it's reading the "garbage".  The library is trying to pull data from a buffer that's empty.  So it keeps getting 0xFFs from the read().  It keeps saving the 0xFFs until it runs out of memory.

mobizt, 2024-03-05:
You have to understand the context of what you are talking for and you have to know the basic understanding of the working protocols.

At least the 0 from available in Client (Stream) does not mean the end of data unless we get what we expected.

BrianG-coder, 2024-03-05:
I don't doubt that there is an invalid SMTP message in the buffer.    But is there a another way to handle the invalid message without getting stuck in an endless loop?

mobizt, 2024-03-05:
You should read what I mention above carefully.

Your server that you are trying to connect is not SMTP server.

mobizt, 2024-03-05:
We don't expect the normal SMTP protocol response to be bigger than 1k unless the base64 server challenge that can be taken place with status code 334 and 4xx.

BrianG-coder, 2024-03-05:
I sent 22 emails through the same server before the crash.

mobizt, 2024-03-05:
https://github.com/mobizt/ESP-Mail-Client/blob/5c6e231e181ff863a325bba38eb692e88c087abb/examples/SMTP/Send_Text/Send_Text.ino#L404-L405

BrianG-coder, 2024-03-05:
already done.

mobizt, 2024-03-05:
You should follow the issue template by providing the code instead of wasting the time of guessing your problem.

You don't have to clear the sending result if you are using the latest or recent version of library because the only last sending result will be stored.

mobizt, 2024-03-05:
You have to debug memory usage or memory leaks in your code too.

The library is memory leaks free using memory management class.

mobizt, 2024-03-05:
The library was intensively test 24/7 for sending and reading email from million messages inbox and it works fine.

mobizt, 2024-03-05:
For the aid of memory usage log, you can use this [simple helping class](https://github.com/mobizt/ESP-Mail-Client/blob/master/src/HeapStat.h) to verify your memory usage.