mobizt/ESP-Mail-Client #331: Restart after triggering an exception

yunhai20092009, 2024-03-18:
**Build tool used:**

- [x] Arudino
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based
- [x] ESP32-based

**Other Libraries That are used:**

> C: ESP Mail Client v3.4.14
> C: connecting to SMTP server
> C: Host > smtp-mail.outlook.com
> C: Port > 587
[4766547][V][WiFiClientImpl.h:317] tcpConnect(): Starting socket

> C: SMTP server connected, wait for greeting...
< S: 220 TYAPR01CA0068.outlook.office365.com Microsoft ESMTP MAIL Service ready at Mon, 18 Mar 2024 05:06:27 +0000

> C: send SMTP command, EHLO
< S: 250-TYAPR01CA0068.outlook.office365.com Hello [120.225.254.181]
< S: 250-SIZE 157286400
< S: 250-PIPELINING
< S: 250-DSN
< S: 250-ENHANCEDSTATUSCODES
< S: 250-STARTTLS
< S: 250-8BITMIME
< S: 250-BINARYMIME
< S: 250-CHUNKING
< S: 250 SMTPUTF8

> C: send command, STARTTLS
< S: 220 2.0.0 SMTP server ready
> C: perform SSL/TLS handshake

abort() was called at PC 0x4018dcb3 on core 1

Backtrace: 0x400837ed:0x3ffb1970 0x4008e9c5:0x3ffb1990 0x40093d3d:0x3ffb19b0 0x4018dcb3:0x3ffb1a30 0x4018dcfa:0x3ffb1a50 0x4018dc5b:0x3ffb1a70 0x4018dfb2:0x3ffb1a90 0x401002f8:0x3ffb1ab0 0x40100623:0x3ffb1b10 0x401006e8:0x3ffb1b40 0x40100705:0x3ffb1b60 0x400e1fa1:0x3ffb1b80 0x400e7b43:0x3ffb1bc0 0x400ee2e6:0x3ffb1c20 0x400d49fd:0x3ffb1c40 0x400d7061:0x3ffb20e0 0x401071fd:0x3ffb2290

ELF file SHA256: 9fa2c669003baabb

Rebooting...
ets Jun  8 2016 00:22:57

**Share code snippet to reproduce the issue:**
```yaml
#include <HTTPClient.h>
#include <ESP_Mail_Client.h>

int sendEmail0(const char* SMTP_HOST, int SMTP_PORT, const char* AUTHOR_EMAIL, const char* AUTHOR_PASSWORD, String arr[], int length){
  int ret = 1;
  const char* SMTP_HOST_i = SMTP_HOST;
  int SMTP_PORT_i = SMTP_PORT;
  const char* AUTHOR_EMAIL_i = AUTHOR_EMAIL;
  const char* AUTHOR_PASSWORD_i = AUTHOR_PASSWORD;
 
  String subject = "stage3";
  String sender_name = "Me";
  String textMsg = subject;
  
  SMTPSession smtp0;
  SMTP_Message message;
  message.sender.name = sender_name;
  message.sender.email = AUTHOR_EMAIL_i;
  message.subject = subject;
  message.text.content = textMsg;
  message.text.transfer_encoding = "enc_7bit"; // recommend for non-ASCII words in message.
  message.text.charSet = F("utf-8"); // recommend for non-ASCII words in message.
  message.priority = esp_mail_smtp_priority::esp_mail_smtp_priority_low;
  message.addHeader(F(AUTHOR_EMAIL_i));

  for ( int i = 0 ; i < length ; i++){
    Serial.println(arr[i]);
    if ( arr[i] != "" )
      message.addRecipient("i", arr[i]);
  }
  
//  MailClient.networkReconnect(true);
  
  smtp0.debug(1);
  Session_Config config;
  config.server.host_name = SMTP_HOST_i;
  config.server.port = SMTP_PORT_i;
  config.login.email = AUTHOR_EMAIL_i;
  config.login.password = AUTHOR_PASSWORD_i; 
  config.login.user_domain = F("127.0.0.1");
  config.time.ntp_server = F("pool.ntp.org,time.nist.gov");
  config.time.gmt_offset = 8;
  config.time.day_light_offset = 0;

  SMTP_Attachment att[1];
  int attIndex = 0;
  att[attIndex].descr.filename = F("data.7z");
  att[attIndex].descr.mime = F("application/octet-stream"); // binary data
  att[attIndex].descr.description = F("data");
  att[attIndex].file.path = F("/data.7z");
  att[attIndex].file.storage_type = esp_mail_file_storage_type_flash;
  att[attIndex].descr.transfer_encoding = Content_Transfer_Encoding::enc_base64;
  message.addAttachment(att[attIndex]);
  
  if (!smtp0.connect(&config))
  {
    MailClient.printf("Connection error, Status Code: %d, Error Code: %d, Reason: %s\n", smtp0.statusCode(), smtp0.errorCode(), smtp0.errorReason().c_str());
    ret = 0;
    smtp0.sendingResult.clear();  
    Serial.println("----------------------------");
    return ret;
  }
  
  if (!smtp0.isLoggedIn())
  {
    Serial.println("Not yet logged in.");
  }
  else
  {
    if (smtp0.isAuthenticated())
      Serial.println("Successfully logged in.");
    else
      Serial.println("Connected with no Auth.");
  }
  
  /* Start sending Email and close the session */
  if (!MailClient.sendMail(&smtp0, &message)){
    ret = 0;
    MailClient.printf("Error, Status Code: %d, Error Code: %d, Reason: %s\n", smtp0.statusCode(), smtp0.errorCode(), smtp0.errorReason().c_str());  
  }
    

  smtp0.sendingResult.clear();
  Serial.println("----------------------------");
  return ret;
}

```
loop interval 10min;
one loop：
  10 Recipients；every 20s trigger sendEmail0，once trigger sendEmail0 has 4 Recipients;

mobizt, 2024-03-18:
Your device is running out of memory that causes the memory allocation failure and system task will fail to feed the watchdog.

mobizt, 2024-03-18:
You should check your code for memory leaks.

This library is memory leaks free.

closed 2024-03-18 by mobizt

mobizt, 2024-03-18:
After inspection your code, there are incorrectly use of objects that pass to the function by pointer.

You have to follow the examples to avoid dangling pointer issue.

Some objects need to define global as it contains the information or session throughout the operation.

mobizt, 2024-03-18:
The SMTPSession should be defined globally as it contains TCP and SSL sessions and user data that some of this will be used in the callback.