mobizt/ESP-Mail-Client #337: Mail client triggers tasks watchdog in FreeRTOS

JoskeVdB, 2024-04-02:
**Build tool used:**

- [ ] Arudino
- [X] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based
- [X] ESP32-based

**Other Libraries That are used:**

**Description of problem:**

**Share code snippet to reproduce the issue:**
```yaml
PASTE .cpp / .ino code here

```

**Additional information and things you've tried:**

JoskeVdB, 2024-04-02:
When it takes too long for an action to complete, the tasks watchdog is triggered when Mail Client is ran as a taks.
This can be solved by putting "delay(10);" (in stead of delay(0);) in file "ESP_Mail_Const.h" where "static void __attribute__((used))
yield_impl()" is defined.

mobizt, 2024-04-02:
Both yield and delay(0) allow wdt feeding.

But delay with any value other than 0 is different scenario and operation and it not the solution for this case.

closed 2024-04-02 by mobizt

JoskeVdB, 2024-04-02:
Thank you very much for this information. It was indeed running on core 0. Problem solved when running on core 1.