mobizt/ESP-Mail-Client #342: IMAP and SMPT example combined produce a crash with ESP8266

pberna67, 2024-05-07:
Dear Developer

It seems that combining the SMTP **Send_text.ino** and the IMAP **Mailbox_Changes_Notification.ino** in a ESP8266 platform, into a single project (to have the capability to send and receive mails) the combined project (attached) crashed, however each project runs independently correctly. At the moment is not yet clear if the problem is located somewhere in your code on in the below libraries

I've noticed that the same projects (combined version and separated version), compiled under the ES32 environment work correctly.

Arduino Configuration

[image]

Serial console report

:15:26.489 -> scandone
13:15:26.489 -> 
13:15:26.631 -> Connecting to Wi-Fi..........scandone
13:15:30.556 -> state: 0 -> 2 (b0)
13:15:30.556 -> .state: 2 -> 3 (0)
13:15:30.556 -> state: 3 -> 5 (10)
13:15:30.556 -> add 0
13:15:30.556 -> aid 1
13:15:30.556 -> cnt 
13:15:30.556 -> 
13:15:30.556 -> connected with WIND-WiFi-C69F, channel 4
13:15:30.603 -> dhcp client start...
13:15:30.836 -> ........ip:192.168.10.182,mask:255.255.255.0,gw:192.168.10.150
13:15:33.220 -> 
13:15:33.220 -> Connected with IP: 192.168.10.182
13:15:33.266 -> 
13:15:33.266 -> #### Connecting to IMAP server...
13:15:33.266 -> > C: ESP Mail Client v3.4.19
13:15:33.266 -> > C: Reading time from NTP server..
13:15:33.312 -> > C: connecting to IMAP server
13:15:33.312 -> > C: Host > imapmail.libero.it
13:15:33.312 -> > C: Port > 993
13:15:34.576 -> 
13:15:34.576 -> #### IMAP server connected
13:15:34.576 -> > C: IMAP server connected
13:15:34.622 -> 
13:15:34.622 -> #### Checking the capability...
13:15:34.622 -> > C: check the capability
13:15:34.716 -> 
13:15:34.716 -> #### Logging in...
13:15:34.716 -> > C: send IMAP command, AUTHENTICATE PLAIN
13:15:34.856 -> 
13:15:34.856 -> #### Send client identification...
13:15:34.856 -> > C: send IMAP command, ID
13:15:35.043 -> Successfully logged in.
13:15:35.043 -> 
13:15:35.043 -> #### Reading the list of mailboxes...
13:15:35.043 -> > C: send IMAP command, LIST
13:15:35.182 -> 
13:15:35.182 -> Available folders: Posta inviata, Posta eliminata, Posta indesiderata, trash, outbox, Spam, draft, INBOX
13:15:35.182 -> 
13:15:35.182 -> #### Open the mailbox folder...
13:15:35.182 -> > C: selecting the INBOX folder...
13:15:35.323 -> 
13:15:35.323 -> Info of the selected folder
13:15:35.323 -> Total Messages: 329
13:15:35.323 -> UID Validity: 1713786152
13:15:35.323 -> Predicted next UID: 1951
13:15:35.323 -> First Unseen Message Number: 5
13:15:35.323 -> Highest Modification Sequence: 2246
13:15:35.323 -> Flags: \Answered, \Flagged, \Deleted, \Seen, \Draft
13:15:35.323 -> #### Connecting to SMTP server...
13:15:35.323 -> > C: ESP Mail Client v3.4.19
13:15:35.323 -> > C: connecting to SMTP server
13:15:35.323 -> > C: Host > smtp.libero.it
13:15:35.323 -> > C: Port > 465
13:15:37.430 -> 
13:15:37.430 -> #### SMTP server connected
[... 161 lines cut ...]
13:15:39.069 -> 3fff15c0:  00000000 00000000 3fff68d8 00000000  
13:15:39.069 -> 3fff15d0:  3fff6914 3ffe8c4c 3fff6ae8 3fff742c  
13:15:39.069 -> 3fff15e0:  000000cc 3fff750c 000000fc 00000000  
13:15:39.069 -> 3fff15f0:  00000000 00000000 00000000 00000000  
13:15:39.069 -> 3fff1600:  00000000 3ffe8c4c 3fff68d8 40100962  
13:15:39.116 -> 3fff1610:  00000000 00000001 00000000 40100962  
13:15:39.116 -> 3fff1620:  00000001 00000000 00000001 3fff00f8  
13:15:39.116 -> 3fff1630:  3fff0068 00000941 3ffe8c4c 40100a6a  
13:15:39.116 -> 3fff1640:  3fff6914 00000000 00000000 00000000  
13:15:39.116 -> 3fff1650:  3ffefdfc 00000020 3fff690c 0000079f  
13:15:39.116 -> 3fff1660:  3ffefdfc 3ffefffc 3fff0440 40206a81  
13:15:39.116 -> 3fff1670:  0000014a 00000000 66264d28 0000079f  
13:15:39.116 -> 3fff1680:  00000005 3fff4c94 00000008 00000000  
13:15:39.116 -> 3fff1690:  00000000 00002f3f 00000101 00000001  
13:15:39.116 -> 3fff16a0:  0000014a 00000000 00000000 3fff6b2c  
13:15:39.116 -> 3fff16b0:  3fff6b54 3fff6b54 00000000 00000000  
13:15:39.163 -> 3fff16c0:  00000000 00000001 0000014a 00000000  
13:15:39.163 -> 3fff16d0:  00000000 3ffe97dd 3fff054c 3fff177c  
13:15:39.163 -> 3fff16e0:  3fff0414 3ffefffc 3fff054c 40206beb  
13:15:39.163 -> 3fff16f0:  4022ce48 b60aa8c0 feefeffe 3fff177c  
13:15:39.163 -> 3fff1700:  3fffdad0 00000000 3fff1750 4022982c  
13:15:39.163 -> 3fff1710:  feefeffe feefeffe 3fffdab0 401011e1  
13:15:39.163 -> <<<stack<<<

[ImapSmptComboDebug_github.zip](https://github.com/mobizt/ESP-Mail-Client/files/15234855/ImapSmptComboDebug_github.zip)

mobizt, 2024-05-07:
It is normal as expected because out of memory.

You have to use Raspberry Pi Pico or ESP32 instead.

closed 2024-05-07 by mobizt

pberna67, 2024-05-07:
correct, with the right memory setting it works. Thanks