mobizt/ESP-Mail-Client #343: Broken TinyGSM detection logic

fatemehsni, 2024-05-25:
**Build tool used:**

- [x] Arudino
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based
- [x] ESP32-based

**Other Libraries That are used:**

TinyGSM

**Description of problem:**
The ESP-Mail-Client fails to detect availability of TinyGSM correctly and define `ESP_MAIL_TINYGSM_IS_AVAILABLE` and `ESP_MAIL_GSM_MODEM_IS_AVAILABLE` as needed. The logic in [extras/Networks.h](https://github.com/mobizt/ESP-Mail-Client/blob/78b8e7b4270199dab5f14265ad565acd80f10465/src/extras/Networks.h#L179-L208) seems to be broken to detect the usage of TinyGSM correctly  for 2 reasons:
1. `TINY_GSM_MODEM_*` `#define`s in the `main.ino` file aren't available to use (check with `#if defined(...)`) in the ESP-Mail-Client library, at least in the latest version of Arduino IDE (2.3.2).
2. New modems such as `TINY_GSM_MODEM_A7670` are missing from the list.

This causes all commands (e.g., `smtp.connect()`) to fail with `connection closed` because [.setGSMClient](https://github.com/mobizt/ESP-Mail-Client/blob/78b8e7b4270199dab5f14265ad565acd80f10465/src/client/ESP_Mail_TCPClient.h#L103-L111) silently fails. [Another issue](https://forum.arduino.cc/t/trouble-with-lilygo-t-sim7000g-and-sending-e-mails/1168461) on Arduino forum by someone else seems to be pointing out the same issue.

To fix the issue I had to manually add `#define TINY_GSM_MODEM_A7670` in the `extras/Networks.h` file and also add `defined(TINY_GSM_MODEM_A7670)` to the list of `#if defined(...)`s.

My guess is that [THIS](https://forum.arduino.cc/t/what-happened-to-my-define/1011013/5) is happening. And the include path `client/ESP_Mail_TCPClient.h -> ESP_Mail_Const.h -> extras/Networks.h` gets put in another translation unit that doesn't see the `#define`s in the `main.ino` file.

**Share code snippet to reproduce the issue:**
Example for using A7670 LTE modem in the README.md (using Arduino IDE 2.3.2 and the T-A7670 Board).

**Additional information and things you've tried:**
Tried defining `#define TINY_GSM_MODEM_A7670`, `#define TINY_GSM_MODEM_A7` (in the supported list), and even directly `#define ESP_MAIL_GSM_MODEM_IS_AVAILABLE` in the `main.ino` file but none worked. None of the `#define`s in the `main.ino` file seem to be visible in the `extras/Networks.h` file included by `client/ESP_Mail_TCPClient.h`.

mobizt, 2024-05-25:
You should read the library documentation thoroughly.
https://github.com/mobizt/ESP-Mail-Client?tab=readme-ov-file#ttgo-t-a7670-lte-with-tinygsm

mobizt, 2024-05-25:
https://github.com/mobizt/ESP-Mail-Client/blob/78b8e7b4270199dab5f14265ad565acd80f10465/examples/SMTP/External_Client/GSMClient/ESP32_SIMA7670/ESP32_SIMA7670.ino#L32-L36

closed 2024-05-25 by mobizt

mobizt, 2024-05-25:
> [!IMPORTANT]  
> The GSM module macros e.g. `TINY_GSM_MODEM_SIM7600` should be defined in two locations.
> 1. In [src/ESP_Mail_FS.h](https://github.com/mobizt/ESP-Mail-Client/blob/master/src/ESP_Mail_FS.h) or in your own defined config at [src/Custom_ESP_Mail_FS.h](https://github.com/mobizt/ESP-Mail-Client/blob/master/src/Custom_ESP_Mail_FS.h) or adding `TINY_GSM_MODEM_SIM7600` in compiler build flags.
> 2. In your sketch.

fatemehsni, 2024-05-25:
Thanks! I hadn't noticed that. It'd be great if you also support `TINY_GSM_MODEM_A7670` though.

mobizt, 2024-05-25:
Why? It wrote in the example comment clearly.

mobizt, 2024-05-25:
With A7670 you can do the same with the define `TINY_GSM_MODEM_A7670`.

You should understand which GSM module macro was used in your case to adapt from example.

mobizt, 2024-05-25:
This issue will be deleted as user error.