mobizt/ESP-Mail-Client #345: ESP8266 goes into WDReset

Devilscave, 2024-06-16:
**Build tool used:**

- [ ] Arudino
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [X] ESP8266-based
- [ ] ESP32-based

**Other Libraries That are used:**

**Description of problem:**

**Share code snippet to reproduce the issue:**
```yaml
PASTE .cpp / .ino code here

```

**Additional information and things you've tried:**

I simply added a small Serial.print to each code section so that I always know which section it is currently in.
And it always hangs at line: ``if (!imap.connect(&config, &imap_data))``
I can see this by the fact that it writes ``imap.connect`` beforehand, does nothing more and then a WDR appears
```
wdt reset
load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v00081be0
~ld```

my login datas are correct.
What else could be the problem?

<!-- LEAVE THIS LINE AS-IS AND DON'T DELETE IT, OTHERWISE THE ISSUE WILL BE CLOSED AUTOMATICALLY. -->

mobizt, 2024-06-16:
You should read the library documentation and follow it.

mobizt, 2024-06-16:
https://github.com/mobizt/ESP-Mail-Client?tab=readme-ov-file#memory-options-for-esp8266

closed 2024-06-16 by mobizt