mobizt/ESP-Mail-Client #347: EmailClient crashes combinint IMAP as well es SMPT libraries

pberna67, 2024-06-23:
Dear   mobizt 

I've already opened such Case some months ago, but for strange reason it has been removed, even from the closed case. You have already solved the problem related to the memory model, not correct, but unfortunately the Case has been cancelled. 

Can you please remind me which memory model must be used to avoid the crashing problem combing both IMAP as well as SMPT libs  using NINA-W101 (ESP32 with 2Mbyte flash) see the below memory options.

Thanks

[image]

mobizt, 2024-06-23:
You are mentioned in device crashes (run time error) but posting the flash memory that stored the user code (firmware) which are not relevant.

You have to read the Espressif documentation for debugging.

For memory issue, you have to debug and monitor your device free heap and stack. 

If you design your own hardware with ESP32 module, you should read the Espressif or module maker design guide documentation.

closed 2024-06-23 by mobizt

mobizt, 2024-06-23:
I see your old post [here](https://github.com/mobizt/ESP-Mail-Client/discussions/332).

And one of your attach images showed that you enable the PSRAM in the option but your u-blox NINA ESP32 has no PSRAM and lead to crash when it boots.

mobizt, 2024-06-23:
I delete my first reply because it is not relevant based on your provided information which is not clear.

You are using U-blox evaluation kit (don't say only u-blox module which you have to design the board to use it) which in the u-blox module contains the ESP32 chip and it does not have PSRAM connected inside the module then you can't choose the PSRAM from IDE option.

That is why no PSRAM option when you choose the board u-blox.

If your board is immediately crashed after boot, I guess it is from you have chosen PSRAM option in board ESP32 Dev Module 

If the crash occurred during running (runtime error), the memory allocation or memory leak may be the cause. In this case, you have to debug your code and monitor your device free heap.