mobizt/ESP-Mail-Client #348: Specifying multiple recipients results in the TO header have incorrect comma placement

LenStruttmann, 2024-07-15:
**Build tool used:**

- [ ] Arudino
- [X ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based
- [X ] ESP32-based

**Other Libraries That are used:**

Specifying multiple recipients, like this:

```yaml
  message.addRecipient( F("Len Struttmann"), F("LenStruttmann@gmail.com"));
  message.addRecipient( F("Len Struttmann"), F("business@struttfolk.com"));
```
... results in the TO header have incorrect comma placement:

to: Len Struttmann Len Struttmann <business@struttfolk.com>,
LenStruttmann@gmail.com

...or...

To: "=?utf-8?B?TGVuIFN0cnV0dG1hbm4=?=" <LenStruttmann@gmail.com>"=?utf-8?B?TGVuIFN0cnV0dG1hbm4=?=" ,<business@struttfolk.com>

This issue is in ESP_Mail_Client::appendAddressHeaderField().  Since appendString() places the comma at the beginning of the buffer...

Lines 812-819:

```yaml
    appendString(buf, encode ? encodeBUTF8(source.name.c_str()).c_str() : source.name.c_str(), false, false, esp_mail_string_mark_type_double_quote);
    // Add white space after name for SMTP to fix iCloud Mail Service IMAP search compatibility issue #278
    // This is not restricted by rfc2822.
    appendSpace(buf);
  }

  appendString(buf, source.email.c_str(), comma, newLine, esp_mail_string_mark_type_angle_bracket);
```

...should be...

```yaml
    appendString(buf, encode ? encodeBUTF8(source.name.c_str()).c_str() : source.name.c_str(), comma, false, esp_mail_string_mark_type_double_quote);
    // Add white space after name for SMTP to fix iCloud Mail Service IMAP search compatibility issue #278
    // This is not restricted by rfc2822.
    appendSpace(buf);
  }

  appendString(buf, source.email.c_str(), false, newLine, esp_mail_string_mark_type_angle_bracket);
```

**Share code snippet to reproduce the issue:**
```yaml
  message.addRecipient( F("Len Struttmann"), F("LenStruttmann@gmail.com"));
  message.addRecipient( F("Len Struttmann"), F("business@struttfolk.com"));
```

**Additional information and things you've tried:**

mobizt, 2024-07-15:
Thanks for reporting this issue.

The issue was fixed in [v3.4.20](https://github.com/mobizt/ESP-Mail-Client/releases/tag/v3.4.20), please update.

closed 2024-07-15 by mobizt

proasnet, 2025-02-11:
Dear @mobizt , thank you for your work. Its nice.
I have a similar problem. I have an option in UI for 2 recipients.

```
  if ( exchange_data.em1_en ) message.addRecipient(F("Email 1"), exchange_data.email_1 ); 
  if ( exchange_data.em2_en ) message.addRecipient(F("Email 2"), exchange_data.email_2 );

  if ( !MailClient.sendMail( &smtp, &message ))
    MailClient.printf( "Error, Status Code: %d, Error Code: %d, Reason: %s\n", smtp.statusCode(), smtp.errorCode(), smtp.errorReason().c_str());
```

This working only for the first recipient and mail is received only by the first recipient.
The second recipient not receive any mail.

mobizt, 2025-02-11:
@proasnet 

This issue was fixed, and it will not exist anymore. 

In your case, you have to check your code and your mail server policy.