mobizt/ESP-Mail-Client #350: Microaoft disabling 3rd party apps?

dineshgyl, 2024-07-21:
**Build tool used:**

- [ x] Arudino
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ x] ESP8266-based
- [ ] ESP32-based

**Other Libraries That are used:**

**Description of problem:**

**Share code snippet to reproduce the issue:**
```yaml
PASTE .cpp / .ino code here

```

**Additional information and things you've tried:**

got this email from microsoft as i am using hotmail email id to send emails using this library. Want to know will this be impacted based on below email?

The safety and security of your information is top priority for Microsoft. To help keep your account secure, Microsoft will no longer support the use of third-party email and calendar apps which ask you to sign in with only your Microsoft Account username and password. To keep you safe you will need to use a mail or calendar app which supports Microsoft’s modern authentication methods. If you do not act, your third-party email apps will no longer be able to access your Outlook.com, Hotmail or Live.com email address on September 16th.

 

What do you need to do?

If you are receiving this email, you are currently using an email or calendar app that uses a less secure authentication method to connect to your Outlook.com email account. You will need to upgrade your third-party mail and calendar app to a version which supports modern authentication methods. 

 

Microsoft provides free versions of Outlook for your PC, Mac, iOS, and Android devices which can be easily downloaded and connect to your email account. Using an updated version of an Outlook application will ensure you are connecting in the most secure way.

 

How can you set up your Gmail, Apple Mail, or other third-party mail application?

Various non-Microsoft applications will have their own steps for connecting to your Outlook.com email account using modern authentication methods. See our help article - Modern Authentication Methods now needed to continue syncing Outlook Email in non-Microsoft email apps. However, you may need to contact the creators of those applications to provide you with instructions. In many cases, simply removing and re-adding your account with the latest version of that application will configure it to use modern authentication methods.

mobizt, 2024-07-21:
You should ask Microsoft instead of asking here.

This is Email Client library that implements the IMAP and SMTP protocols using supports auth that does not require user consent in non-UI device.

https://github.com/mobizt/ESP-Mail-Client?tab=readme-ov-file#features

There is no user consent auth flow support, and it is not possible to do in embedded devices.

closed 2024-07-21 by mobizt

mobizt, 2024-07-21:
This library supports OAUth2.0 authentication which you need to create this access token yourself.

This library does not provide the authentication management functions because it is out of library scope, and it may require user consent in the javascript or web browser capable devices.

mobizt, 2024-07-21:
https://learn.microsoft.com/en-us/entra/identity-platform/v2-oauth2-auth-code-flow

mobizt, 2024-07-21:
If you do not enable Two-factor authentication for your Microsoft account, you should enable it which it is a kind of modern authentication method.

You may still use your Microsoft account credentials to sign in with this library but I'm not sure and you should ask the Microsoft support agent.