mobizt/ESP-Mail-Client #352: esp8266 random requests to girbberish domains

SpyrosLiontos, 2024-08-16:
**Build tool used:**

- [X] Arudino
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [X] ESP8266-based
- [ ] ESP32-based

**Other Libraries That are used:**

**Description of problem:**
Hello I started testing this library with a nodemcu esp8266 board with the arduino studio. I used the basic send email message code from  https://github.com/mobizt/ESP-Mail-Client?tab=readme-ov-file#send-email-message. I have a pihole that all of my network traffic goes through as my dhcp server uses it for my dns server, so naturally i can see what the esp requests when i send a message with your library. In my case as a mail is sent at setup i can see a request at the ntp server and smtp server (all correct as this far). Now the problem is this : the pihole reports after a while that the esp sends dns requests for websites that are gibberish or huge strings of characters with symbols letters and numbers, in other instanses it is a request to t.gov which i suspect is somewhat correlated  to the time.nist.gov server? and in other cases it is a request to a simple letter (eg v) or a symbol (.,*/) , all while the code specifies that only one mail in the beginning is set to be sent (although i did also observe this with multiple mails within time intervals being sent from the void loop). This behaviour was tested with a bare esp-12e and a nodemcu board and neither of them reported any soft resets (or any watchdog violations) or any ntp or smtp miscommunications. the amount it took to sent one mail is around 1.7-2.2 seconds measured by a difference with the micros function.

**Share code snippet to reproduce the issue:**
```yaml
// Include WiFi library
#include <Arduino.h>
#if defined(ESP32) || defined(ARDUINO_RASPBERRY_PI_PICO_W)
#include <WiFi.h>
#elif defined(ESP8266)
#include <ESP8266WiFi.h>
#elif  __has_include(<WiFiNINA.h>)
#include <WiFiNINA.h>
#elif __has_include(<WiFi101.h>)
#include <WiFi101.h>
#endif

// Include ESP Mail Client library (this library)
#include <ESP_Mail_Client.h>

// Declare the global used SMTPSession object for SMTP transport
SMTPSession smtp;

// Declare the global used Session_Config for user defined session credentials
Session_Config config;

// Callback function to get the Email sending status
void smtpCallback(SMTP_Status status);

void setup()
{

  Serial.begin(115200);

  WiFi.begin("<ssid>", "<password>");
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  // Set the session config
  config.server.host_name = "smtp.office365.com"; // for outlook.com
  config.server.port = 587; // for TLS with STARTTLS or 25 (Plain/TLS with STARTTLS) or 465 (SSL)
  config.login.email = "your Email address"; // set to empty for no SMTP Authentication
  config.login.password = "your Email password"; // set to empty for no SMTP Authentication
  
  // For client identity, assign invalid string can cause server rejection
  config.login.user_domain = "client domain or public ip";  

  /*
   Set the NTP config time
   For times east of the Prime Meridian use 0-12
   For times west of the Prime Meridian add 12 to the offset.
   Ex. American/Denver GMT would be -6. 6 + 12 = 18
   See https://en.wikipedia.org/wiki/Time_zone for a list of the GMT/UTC timezone offsets
   */
  config.time.ntp_server = "pool.ntp.org,time.nist.gov";
  config.time.gmt_offset = 3;
  config.time.day_light_offset = 0;

  // Declare the SMTP_Message class variable to handle to message being transport
  SMTP_Message message;

  // Set the message headers
  message.sender.name = "My Mail";
  message.sender.email = "sender or your Email address";
  message.subject = "Test sending Email";
  message.addRecipient("name1", "email1");

  // Set the message content
  message.text.content = "This is simple plain text message";

  // Set debug option
  smtp.debug(1);

  // Set the callback function to get the sending results
  smtp.callback(smtpCallback);

  // Connect to the server
  smtp.connect(&config);

  // Start sending Email and close the session
  if (!MailClient.sendMail(&smtp, &message))
    Serial.println("Error sending Email, " + smtp.errorReason());

}

void smtpCallback(SMTP_Status status)
{
 
  Serial.println(status.info());

  if (status.success())
  {
    // See example for how to get the sending result
  }
}

void loop(){}

```

**Additional information and things you've tried:**

mobizt, 2024-08-16:
What is your issue?

I did not see any issue related to the library.

closed 2024-08-16 by mobizt

mobizt, 2024-08-16:
You should learn about SMTP and related standards.

mobizt, 2024-08-16:
You should use some network protocol analysis tool e.g. Wireshark instead to check what you want in your network.

SpyrosLiontos, 2024-08-17:
My issue is that for some reason the code you provide as an example in your repo for this exact library is producing the effect i wrote above. The pihole is showing exactly what the network is doing. and in this case the esp tries sends multiple requests to various sites. for example @ô#@àì'@0Üþ?ïë·. I disagree that this is not related to your library. The code I provided above is your basic example and the effect is observed with it. Have you performed any tests? and also how can you label this as user error since it is not even produced by any additional code by me?

mobizt, 2024-08-17:
I don't know where you get the non-ascii text from `@ô#@àì'@0Üþ?ï�ë�·��.`  but it is not the website or server that you are mentioned.

If it is from your device Serial, it is not the correct baud rate between IDE console and in your code.

In this case you should debug the trace.

Please make sure you read the library documentation thoroughly. 

If this is from the tool you use, it is your problem to contact its developer.

I recommend the network protocol analyzer in this case to know where this packet is from.

SpyrosLiontos, 2024-08-17:
[image]
look at the image above. Those are requests i see on the pi by the esp. the esp is sending those random requests to the pi. this is not via serial! those are requests for servers! they are random characters and they are being sent continuosly. that is the effect i am observing. in addition i am also observing the serial monitor but no soft resets or watchdog violations are reported! i have tested this and confirmed it with two different boards. two different pis and with multiple positions of the esp (as close to the wifi antena as possible).

mobizt, 2024-08-17:
> Have you performed any tests? and also how can you label this as user error since it is not even produced by any additional code by me?

This library was intensively tested 24/7 over five years and tested by hundred thousand users.

User error is reasonable.

SpyrosLiontos, 2024-08-17:
> > Have you performed any tests? and also how can you label this as user error since it is not even produced by any additional code by me?
> 
> This library was intensively tested 24/7 over five years and tested by hundred thousand users.
> 
> User error is reasonable.

how can it be user error if this is the code provided in the instructions of your repo? i intentionally stripped everything down to it and tested the simplest example you have as to eliminate any error from any code from me.

mobizt, 2024-08-17:
Use standard tool like Wireshark instead to analyze the packet not a buggy tool.

mobizt, 2024-08-17:
> I have a pihole that all of my network traffic goes through as my dhcp server uses it for my dns server

That is the problem, you should try to use library normally without your pihole and use Wireshark to capture all your network packets.

mobizt, 2024-08-17:
There are many library [examples](https://github.com/mobizt/ESP-Mail-Client/tree/master/examples/SMTP) to test you can test it.

Avoid copying the code from web browser, it can be hidden characters copied.

If you are suspected spam or malware or spyware, this is the opensource library which you can verify the source, and all traffics from network protocol analyzer.