mobizt/ESP-Mail-Client #353: esp8266 random DNS requests

SpyrosLiontos, 2024-08-17:
**Build tool used:**

- [x] Arudino
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [x] ESP8266-based
- [ ] ESP32-based

**Other Libraries That are used:**

**Description of problem:**
I see the issue that I had posted yesterday got deleted. Why did it got deleted is beyond me (but this is a foul play from your part). I am reposting this along with your requested wireshark capture as you did not believe that pihole was enough. 
[image]
I will state again that i am using the example code found at https://github.com/mobizt/ESP-Mail-Client?tab=readme-ov-file#send-email-message and nothing more. The problem is this: I keep seeing that the esp after a certain amount of time is requesting random characters as DNS lookups. The fact that you initially accused me of user error and then after providing screenshots of the actual requests of the sites from pihole server and then deleting altogether the issue is really bad. I am reporting this as a bug and you keep denying it. I hope the wireshark capture is enough to convince you.

**Share code snippet to reproduce the issue:**
```yaml
PASTE .cpp / .ino code here

```

**Additional information and things you've tried:**

closed 2024-08-17 by mobizt

mobizt, 2024-08-17:
ESP8266
[image]

ESP32
[image]

SpyrosLiontos, 2024-08-17:
did you let the capture run for many hours? As i said it is random. I waited around ~1-2 hours before I saw the random requests. If you can see the initial requests from me are correct as well. The effect appears afterwards.

mobizt, 2024-08-17:
Are you kidding? It is nonsense at all.

Nothing from library is running in the loop and TCP session was already close.

You should further analyze the origine of other dns requests.

mobizt, 2024-08-17:
Such this irrelevant topic will be deleted.

SpyrosLiontos, 2024-08-17:
this is exactly why i created an issue, I stated this in the original issue you deleted! that while only one mail was being sent from the setup function and none was sent from the loop i saw requests from the esp.

SpyrosLiontos, 2024-08-17:
you can go ahead and delete it if it bothers you. i tried two times to report to you a behaviour i saw. it is up to you to do whatever you want.

mobizt, 2024-08-17:
There is no issue related to library.

Yes, it bothers me too much because you don't realize the origine of that dns request but just blaming something nonsense.

SpyrosLiontos, 2024-08-17:
the origin is the esp. checked by the mac address of the esp that i had loaded the code. and the ip it reported it took from the dhcp server.

mobizt, 2024-08-17:
You should open the issue in the esp8266 community repos instead for precompiled lwip and its dependencies.

mobizt, 2024-08-17:
There is nothing relate to application (library).

SpyrosLiontos, 2024-08-17:
In any case do whatever you want. As I said I only tried to report a bug to you. It seems i better spend my time on something else that to try to prove to you that i am not an elephant! The effect was observed with only your library and no others. good day!

mobizt, 2024-08-17:
As I ever told you that this is opensource library which you can verify.

SpyrosLiontos, 2024-08-17:
I never blamed you for anything i just wanted to report a bug

mobizt, 2024-08-17:
This is not a bug. That's it.