mobizt/ESP-Mail-Client #360: **more than one operator "+" matches these operands:C/C++(350)**

GoRemoteKC, 2024-09-19:
**Build tool used:**

- [ ] Arudino
- [ X] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based
- [X ] ESP32-based

**Other Libraries That are used:**

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps = mobizt/ESP Mail Client@^3.4.21

**Description of problem:**

I've seen discussions about this issue on a few forums from the last year or so, but none have found a solution.

I recently installed the most recent releases of VScode and PlatformIO on a new PC (Windows 10). Now in any project that includes the ESP MAIL CLIENT, any code that uses String operators results in the IDE (Intellisense?) flagging string operators with a warning/error.  Red squiggles everywhere.   The error is noted below:

> **more than one operator "+" matches these operands:C/C++(350)**
> 
> main.cpp(10, 13): function "operator+(const StringSumHelper &lhs, const String &rhs)" (declared at line 185 of "C:\Users\John\.**platformio\packages\framework-arduinoespressif32\cores\esp32\WString.h")**
> 
> main.cpp(10, 13): function "operator+(MB_String &&lhs, const MB_String &rhs)" (declared at line 2048 of "D:\Users\John\Documents\PlatformIO\Projects\test\.**pio\libdeps\esp32dev\ESP Mail Client\src\extras\MB_String.h")**
> 
> main.cpp(10, 13): operand types are: String + String
> main.cpp(10, 13): function "operator+(MB_String &&lhs, char rhs)" does not match because argument #2 does not match parameter
> main.cpp(10, 13): function "operator+(char lhs, MB_String &rhs)" does not match because argument #1 does not match parameter
> main.cpp(10, 13): function "operator+(MB_String &lhs, char rhs)" does not match because argument #1 does not match parameter
> main.cpp(10, 13): function "operator+(MB_String &lhs, MB_String &&rhs)" does not match because argument #1 does not match parameter

The very basic sample project below only includes the ESP_MAIL_CLIENT and demonstrates the problem.

Every line of code that uses the String "+" concatenation operator will be flagged by the IDE (Intellisense?) with the error shown above.  And of course they also show as "problems" in the IDE file explorer.

*Share code snippet to reproduce the issue:**
```yaml
PASTE .cpp / .ino code here

#include <Arduino.h>
#include <ESP_Mail_Client.h>

void setup() {
String str_1 = "One-";
String str_2 = "Two";
String Msg;
Msg = str_1 + str_2;                               // The "+" operator will be flagged.
Serial.println("Message : " + str_1 + str_2);       // The first "+" operator will be flagged.
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

**Additional information and things you've tried:**

The code does compile, link and execute correctly, but all the red squiggles and flags in the IDE is a bit disconcerting.

I am by no means an expert on overloading or namespace scope, but I believe MB_strings.h in the library overloads the String operators and several of the library header files contain the line "using namespace mb_string;" which I'm led to believe makes the scope global.  Perhaps that is what is fooling the IDE (Intellisense)?  Or perhaps I'm completely wrong and it's something else entirely.

Thanks for all you efforts.

mobizt, 2024-09-19:
There is nothing related to the library.

mobizt, 2024-09-19:
That is intelisense error, please google it.

mobizt, 2024-09-19:
The first error is from ESP32 Arduino Core WString.h.

You can test your sketch without any library.

closed 2024-09-19 by mobizt

GoRemoteKC, 2024-09-19:
OK. I just removed "lib_deps = mobizt/ESP Mail Client@^3.4.21" from PlatformIO.ini and "#include <ESP_Mail_Client.h>" from main.cpp of the simple test show above and the errors immediately go away.  I have projects that use many different libraries, and only ESP_Mail_Client" is causing this issue.

Isn't the error posted above saying it is finding two "+" operators. One in Wstring and one in MB_strings in the Mail Client library?

I've Googled everything I can think of, which is what led me to believe it's something associated with the overloading of the String operators in MB_string.h and the "using namespace mb_string;" in the library header files.

Intellisense is seeing "more than one operator "+" matches these operands:C/C++(350)" and listing them as being in Wstring.h and MB_string.h in the following lines.

mobizt, 2024-09-19:
Did you known this is not new and many people found this only in ESP32 Arduino Core?

You should search in PlatformIO forum, and you will find this VSCode intellisense error.

mobizt, 2024-09-19:
> Intellisense is seeing "more than one operator "+" matches these operands:C/C++(350)" and listing them as being in Wstring.h and MB_string.h in the following lines.

You should focus only the first error, the followed error is because of the first error.

mobizt, 2024-09-19:
I recommend you update the PlatformIO, VSCode, and ESP32 core.

This error was only found in outdated core or IDE in the past.