mobizt/ESP-Mail-Client #362: esp-littlefs error when sending mail

BozoTClown, 2024-09-26:
**Build tool used:**

- [ ] Arudino - 1.8.19
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**
   Board - ESP32-Wroom-32

- [ ] ESP8266-based
- [ ] ESP32-based - Yes

**Other Libraries That are used:**

#include <HTTPClient.h>
#include "SPIFFS.h"
#include <ESP_Mail_Client.h>
#include <WiFiUdp.h>
#include <NTPClient.h>
#include <WebServer.h>
#include <WiFi.h>
#include <PubSubClient.h>
#include <Time.h>
#include <string.h>
#include <WebOTA.h>
#include <NTPClient.h>

**Description of problem:**

**Share code snippet to reproduce the issue:**
```yaml
PASTE .cpp / .ino code here

void sendEmail() {

  config.server.host_name = SMTP_server; // for outlook.com
  config.server.port = SMTP_Port; // for TLS with STARTTLS or 25 (Plain/TLS with STARTTLS) or 465 (SSL)
  config.login.email = sender_username; // set to empty for no SMTP Authentication
  config.login.password = sender_password; // set to empty for no SMTP Authentication
  
  // For client identity, assign invalid string can cause server rejection
  config.login.user_domain = "127.0.0.1";  

  /*
   Set the NTP config time
   For times east of the Prime Meridian use 0-12
   For times west of the Prime Meridian add 12 to the offset.
   Ex. American/Denver GMT would be -6. 6 + 12 = 18
   See https://en.wikipedia.org/wiki/Time_zone for a list of the GMT/UTC timezone offsets
   */
  config.time.ntp_server = "pool.ntp.org,time.nist.gov";
  config.time.gmt_offset = 5;
  config.time.day_light_offset = -1800;

  // Declare the SMTP_Message class variable to handle to message being transport
  SMTP_Message message;

  // Set the message headers
  message.sender.name = "Garage Door";
  message.sender.email = sender_username;
  message.subject = emailSub;
  message.addRecipient("name1", Recipient_email);
//  message.addRecipient("name2", Recipient_email);

/*  message.addCc("email3");
  message.addBcc("email4");  */

  // Set the message content
  message.text.content = emailMSG;

  // Set debug option
  smtp.debug(0);

  // Set the callback function to get the sending results
  smtp.callback(smtpCallback);

  // Connect to the server
  smtp.connect(&config);

  // Start sending Email and close the session
  if (!MailClient.sendMail(&smtp, &message))
    Serial.println("Error sending Email, " + smtp.errorReason());

  emailMSG = "The garage door"; 
  
}  

```

**Additional information and things you've tried:**
I am using SPIFFS for other string stored on the device.
What I have noticed is whenever I send an email the error occurrs priot to getting connected to the SMTP server. After the error message my device becomes unstable because it seems as though my SPIFF files are no longer present. After re-uploading them the device will function.

The complete error messgae is -
   #### Connecting to SMTP server...
  E (2164310) esp_littlefs: ./components/esp_littlefs/src/littlefs/lfs.c:1225:error: Corrupted dir pair at {0x0, 0x1}

  E (2164311) esp_littlefs: mount failed,  (-84)
  E (2164314) esp_littlefs: Failed to initialize LittleFS

mobizt, 2024-09-26:
The library works normally for all filesystems. You have to check your code.

closed 2024-09-26 by mobizt

mobizt, 2024-09-27:
SPIFFS is deprecated and not recommended.

Please use LittleFS instead.

BozoTClown, 2024-09-28:
So now what you are saying that it does not work with ALL filesystems!

 

Dale

mobizt, 2024-09-28:
Please google.

BozoTClown, 2024-09-29:
Spent days googling no mention of SPIFFs being outdated. Also no mention as to why you sendmail need littleFS.

mobizt, 2024-09-29:
SPIFFS is deprecated in ESP32 Arduino Core v3.x.x and LittleFS was used by default.

You can change the default filesystem via [src/ESP_Mail_FS.h](https://github.com/mobizt/ESP-Mail-Client/blob/master/src/ESP_Mail_FS.h#L73-L92).

https://github.com/mobizt/ESP-Mail-Client/blob/566db6d446a8ed67d3e81746346852850e2a81fa/src/ESP_Mail_FS.h#L73-L92

You will see that LittleFS was set for the macro `ESP_MAIL_DEFAULT_FLASH_FS` if ESP32 Core version is greater than v2.x.x on line no. 82-83.

If you want to use SPIFFS, you have to replace line mo.82 and 83 with following which is similar to line no. 87-88.

```cpp
#include <SPIFFS.h> 
#define ESP_MAIL_DEFAULT_FLASH_FS SPIFFS 
```

BozoTClown, 2024-09-29:
Thank you very much for you reply. It was extremely helpful and I will try to mod you suggested. I hope it works as I am not extremely proficient in coding these things, as you can probably tell!

 

Thanks again

 

https://github.com/mobizt/ESP-Mail-Client/blob/566db6d446a8ed67d3e81746346852850e2a81fa/src/ESP_Mail_FS.h#L73-L92

BozoTClown, 2024-09-29:
Made the mod you suggested. Still receiving the following messages, but the email does go though.

 

#### Connecting to SMTP server...

E (117705) esp_littlefs: ./components/esp_littlefs/src/littlefs/lfs.c:1225:error: Corrupted dir pair at {0x0, 0x1}

 

E (117706) esp_littlefs: mount failed,  (-84)

E (117709) esp_littlefs: Failed to initialize LittleFS

#### Error, NTP server time reading timed out

 

#### SMTP server connected

 

When I reboot the device it goes into am endless reboot intil I unplug it. When I reload the SPIFFs data everything works fine until I send another email.

 

ets Jul 29 2019 12:21:46

 

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:1

load:0x3fff0030,len:1344

load:0x40078000,len:13964

load:0x40080400,len:3600

entry 0x400805f0

E (266) psram: PSRAM ID read error: 0xffffffff

E (292) SPIFFS: mount failed, -10025

An Error has occurred while mounting SPIFFS

 

assert failed: tcpip_send_msg_wait_sem IDF/components/lwip/lwip/src/api/tcpip.c:455 (Invalid mbox)

 

 

Backtrace: 0x40083879:0x3ffcc430 0x4008fb5d:0x3ffcc450 0x40094fdd:0x3ffcc470 0x40120111:0x3ffcc5a0 0x40130bc5:0x3ffcc5d0 0x40130c25:0x3ffcc5f0 0x4011f8e9:0x3ffcc640 0x400d8ac1:0x3ffcc660 0x400d8b50:0x3ffcc6b0 0x401012d5:0x3ffcc6d0 0x40103ce7:0x3ffcc6f0 0x4010401d:0x3ffcc750 0x4010406c:0x3ffcc770 0x40104083:0x3ffcc790 0x400d5de0:0x3ffcc7b0 0x401072cd:0x3ffcc7d0

 

 

 

 

ELF file SHA256: 1631cebf28db2569

 

https://github.com/mobizt/ESP-Mail-Client/blob/566db6d446a8ed67d3e81746346852850e2a81fa/src/ESP_Mail_FS.h#L73-L92

mobizt, 2024-09-29:
Don't reply via email. You should go to github to post your comment.

mobizt, 2024-09-29:
Where is your code that initiates SPIFFS?