mobizt/ESP-Mail-Client #366: Send Email over GSM error

Jvf01, 2024-11-14:
**Build tool used:**

- [x ] Arudino
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based
- [ x] ESP32-based

**Other Libraries That are used:**

I bought a LILLY TTGO A7670e card and I'm trying to send an email via GSM (4G?).
It gives the error "#### Error, connection closed
! E: connection closed" which I can't get past.

I think that the error occurs with the statement :

  smtp.connect(&config);

[Log_Email_error.docx](https://github.com/user-attachments/files/17753162/Log_Email_error.docx)

**Share code snippet to reproduce the issue:**
// For TTGO T-A7670
#define TINY_GSM_MODEM_A7670 // SIMA7670 Compatible with SIM7600 AT instructions

// Set serial for debug console (to the Serial Monitor, default speed 115200)
#define SerialMon Serial

// Set serial for AT commands (to the module)
// Use Hardware Serial on Mega, Leonardo, Micro
#define SerialAT Serial1

#include <ESP_Mail_Client.h>
#include <TinyGsmClient.h>

// See all AT commands, if wanted
// #define DUMP_AT_COMMANDS

// Define the serial console for debug prints, if needed
#define TINY_GSM_DEBUG SerialMon

#define TINY_GSM_USE_GPRS true
#define TINY_GSM_USE_WIFI false

// set GSM PIN, if any
#define GSM_PIN ""

#define SMTP_HOST "smtp.gmail.com"
#define SMTP_PORT 465

//  The sign in credentials 
#define AUTHOR_EMAIL "routertico@gmail.com"
#define AUTHOR_PASSWORD "xxxxxxxxxxxx"
// #define AUTHOR_PASSWORD "Gatinha1%"

//  Recipient's email
#define RECIPIENT_EMAIL "joaquim.v.fernandes@gmail.com"

SMTPSession smtp;

// Callback function to get the Email sending status
void smtpCallback(SMTP_Status status);

// Your GPRS credentials, if any
const char apn[] = "umts";
const char gprsUser[] = "";
const char gprsPass[] = "";
[... 257 lines cut ...]

  Serial.println();
  Serial.printf("Registration Status:%d\n", status);
  delay(1000);

  String ueInfo;
  if (modem.getSystemInformation(ueInfo)) {
    Serial.print("Inquiring UE system information:");
    Serial.println(ueInfo);
  }

  if (!modem.enableNetwork()) {
    Serial.println("Enable network failed!");
  }

  delay(5000);
  // enviawapp("");
}

```

**Additional information and things you've tried:**
[Log_Email_error.docx](https://github.com/user-attachments/files/17753231/Log_Email_error.docx)

<!-- LEAVE THIS LINE AS-IS AND DON'T DELETE IT, OTHERWISE THE ISSUE WILL BE CLOSED AUTOMATICALLY. -->

mobizt, 2024-11-14:
The error shows that library cannot operate your GSM module with TinyGSM library because it was not properly setup, please follow and read the comments in [this example](https://github.com/mobizt/ESP-Mail-Client/blob/master/examples/SMTP/External_Client/GSMClient/ESP32_SIMA7670/ESP32_SIMA7670.ino) thoroughly.

mobizt, 2024-11-14:
https://github.com/mobizt/ESP-Mail-Client/blob/86cab269d4b3eb58de1de680e959c3d82482bb5d/examples/SMTP/External_Client/GSMClient/ESP32_SIMA7670/ESP32_SIMA7670.ino#L32-L34

closed 2024-11-14 by mobizt

mobizt, 2024-11-14:
There is no macro that you define `TINY_GSM_MODEM_A7670` in TinyGSM library, where did you get it from?https://github.com/vshymanskyy/TinyGSM/blob/master/src/TinyGsmClient.h

Please spend the time to learn and play with TinyGSM library to make sure that you use it correctly.

As you see in this library example comment, you just to follow it because `SIMA7670` and `SIM7600` AT instructions are compatible.
https://github.com/mobizt/ESP-Mail-Client/blob/86cab269d4b3eb58de1de680e959c3d82482bb5d/examples/SMTP/External_Client/GSMClient/ESP32_SIMA7670/ESP32_SIMA7670.ino#L36

Jvf01, 2024-11-14:
In my project i´m using the libraries from "https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main/lib" that support teh borad TTGO A7670E that i bought.

I´m sorry but I don't quite understand what it means " // To allow TinyGSM library integration, the following macro should be defined in src/ESP_Mail_FS.h or 
 // your custom config file src/Custom_ESP_Mail_FS.h. 
 //  #define TINY_GSM_MODEM_SIM7600 "

ok, thank you and best regards

mobizt, 2024-11-14:
You have to use this [TinyGSM](https://github.com/vshymanskyy/TinyGSM) from the developer repo instead of modified version, otherwise you cannot use this Email library.

Did you read this line of example comment?
https://github.com/mobizt/ESP-Mail-Client/blob/86cab269d4b3eb58de1de680e959c3d82482bb5d/examples/SMTP/External_Client/GSMClient/ESP32_SIMA7670/ESP32_SIMA7670.ino#L12

It means that it uses TTGO T-A7670 which is the same board as yours.

https://github.com/mobizt/ESP-Mail-Client/blob/86cab269d4b3eb58de1de680e959c3d82482bb5d/examples/SMTP/External_Client/GSMClient/ESP32_SIMA7670/ESP32_SIMA7670.ino#L32-L34

The above comment in the example shows that, you have to edit [this config file](https://github.com/mobizt/ESP-Mail-Client/blob/86cab269d4b3eb58de1de680e959c3d82482bb5d/src/ESP_Mail_FS.h#L168) by adding this code `#define TINY_GSM_MODEM_SIM7600` in that file and also in your sketch.

Jvf01, 2024-11-15:
Thank you very much for your help..... i will try it and give you my feed
back.
Best regards

Jvf01, 2024-11-17:
It's solved.... it's working

Thank you