mobizt/ESP-Mail-Client #367: Compile problem ESP32

HPChard, 2024-12-04:
**Build tool used:**
- [ ] Arduino 2.3.4

**Board used (ESP32/ESP8266/Arduino):**
- [ ] ESP32 w SDK 3.1.0-RC3:  Arduino board "ESP32 Wrover Kit (all versions)"

**Other Libraries That are used:**
 <Arduino.h>
 <WiFi.h>

**Description of problem:**
Compile is generating following:
In file included from c:/Users/Admin/Desktop/ARDUINO/libraries/ESP_Mail_Client/src/client/SSLClient/ESP_SSLClient.h:39,
                 from c:/Users/Admin/Desktop/ARDUINO/libraries/ESP_Mail_Client/src/client/ESP_Mail_TCPClient.h:58,
                 from c:\Users\Admin\Desktop\ARDUINO\libraries\ESP_Mail_Client\src/ESP_Mail_Client.h:97,
                 from C:\Users\Admin\Desktop\ARDUINO\ESP\Test_Mail_Client\Test_Mail_Client.ino:6:
c:/Users/Admin/Desktop/ARDUINO/libraries/ESP_Mail_Client/src/client/SSLClient/client/BSSL_TCP_Client.h:440:21: error: cannot declare field 'BSSL_TCP_Client::_ssl_client' to be of abstract type 'BSSL_SSL_Client'
  440 |     BSSL_SSL_Client _ssl_client;
      |                     ^~~~~~~~~~~
In file included from c:/Users/Admin/Desktop/ARDUINO/libraries/ESP_Mail_Client/src/client/SSLClient/client/BSSL_TCP_Client.h:58:
c:/Users/Admin/Desktop/ARDUINO/libraries/ESP_Mail_Client/src/client/SSLClient/client/BSSL_SSL_Client.h:93:7: note:   because the following virtual functions are pure within 'BSSL_SSL_Client':
   93 | class BSSL_SSL_Client : public Client
      |       ^~~~~~~~~~~~~~~
In file included from C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0-RC3\cores\esp32/Arduino.h:197,
                 from C:\Users\Admin\Desktop\ARDUINO\ESP\Test_Mail_Client\Test_Mail_Client.ino:2:
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0-RC3\cores\esp32/Client.h:29:16: note:     'virtual int Client::connect(IPAddress, uint16_t, int32_t)'
   29 |    virtual int connect(IPAddress ip, uint16_t port, int32_t timeout) = 0;
      |                ^~~~~~~
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0-RC3\cores\esp32/Client.h:31:16: note:     'virtual int Client::connect(const char*, uint16_t, int32_t)'
   31 |    virtual int connect(const char *host, uint16_t port, int32_t timeout) = 0;

**Share code snippet to reproduce the issue:**

PASTE .cpp / .ino code here
// Include WiFi library
#include <Arduino.h>
#include <WiFi.h>

// Include ESP Mail Client library (this library)
#include <ESP_Mail_Client.h>

// Declare the global used SMTPSession object for SMTP transport
SMTPSession smtp;

// Declare the global used Session_Config for user defined session credentials
Session_Config config;

// Callback function to get the Email sending status
void smtpCallback(SMTP_Status status);

void setup()
{
  Serial.begin(115200);

  WiFi.begin("------", "------");
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  // Set the session config
  config.server.host_name = "smtp.gmail.com"; // for gmail
  config.server.port = 465; // for TLS with STARTTLS or 25 (Plain/TLS with STARTTLS) or 465 (SSL)
  config.login.email = "-----@gmail.com"; // set to empty for no SMTP Authentication
  config.login.password = "------"; // set to empty for no SMTP Authentication
  
  // For client identity, assign invalid string can cause server rejection
  // config.login.user_domain = "client domain or public ip";  

  /*
   Set the NTP config time
   For times east of the Prime Meridian use 0-12
   For times west of the Prime Meridian add 12 to the offset.
   Ex. American/Denver GMT would be -6. 6 + 12 = 18
   See https://en.wikipedia.org/wiki/Time_zone for a list of the GMT/UTC timezone offsets
   */
  config.time.ntp_server = "pool.ntp.org,time.nist.gov";
  config.time.gmt_offset = -18000;
  config.time.day_light_offset = 0;

  // Declare the SMTP_Message class variable to handle to message being transport
  SMTP_Message message;

  // Set the message headers
  message.sender.name = "------";
  message.sender.email = "------@gmail.com";
  message.subject = "Test sending Email ESP32";
  message.addRecipient("--------", "-----@gmail.com");
 // message.addRecipient("name2", "email2");

  //message.addCc("email3");
  //message.addBcc("email4");

  // Set the message content
  message.text.content = "This is simple plain text message";

  //Base64 data of image
  const char *greenImg = "iVBORw0KGgoAAAANSUhEUgAAAGQAAABkCAYAAABw4pVUAAAAoUlEQVR42u"
                         "3RAQ0AMAgAoJviyWxtAtNYwzmoQGT/eqwRQoQgRAhChCBECEKECBGCECEI"
                         "EYIQIQgRghCECEGIEIQIQYgQhCBECEKEIEQIQoQgBCFCECIEIUIQIgQhCB"
                         "GCECEIEYIQIQhBiBCECEGIEIQIQQhChCBECEKEIEQIQhAiBCFCECIEIUIQ"
                         "ghAhCBGCECEIEYIQIUKEIEQIQoQg5LoBBaDPbQYiMoMAAAAASUVORK5CYII=";

  // Declare the attachment data
  SMTP_Attachment att;

  // Set the attatchment info
  att.descr.filename = "green.png";
  att.descr.mime = "image/png";
  att.blob.data = (uint8_t *)greenImg;
  att.blob.size = strlen(greenImg);
  // Set the transfer encoding to base64
  att.descr.transfer_encoding = Content_Transfer_Encoding::enc_base64;
  // We set the content encoding to match the above greenImage data
  att.descr.content_encoding = Content_Transfer_Encoding::enc_base64;

  // Add attachment to the message
  message.addAttachment(att);

  // Set debug option
  smtp.debug(1);

  // Set the callback function to get the sending results
  smtp.callback(smtpCallback);

  // Connect to the server
  smtp.connect(&config);

  // Start sending Email and close the session
  if (!MailClient.sendMail(&smtp, &message))
    Serial.println("Error sending Email, " + smtp.errorReason());

}
void smtpCallback(SMTP_Status status)
{
  Serial.println(status.info());
  if (status.success())
  {
    // See example for how to get the sending result
  }
}
```

**Additional information and things you've tried:**
Have tried searching for others who have reported same problem and found solution, but no luck with my search.
Previously (about a year ago) successfully compiled and executed ESP_Mail_Client in another sketch (same board), so it is mysterious as to why I cannot compile the above sketch which is just the example for sending an email with attached file....
Thanks for any help

<!-- LEAVE THIS LINE AS-IS AND DON'T DELETE IT, OTHERWISE THE ISSUE WILL BE CLOSED AUTOMATICALLY. -->

mobizt, 2024-12-05:
This issue was introduced in v3.1.0-RC1 afterward by the ESP32 Arduino Core team.

They add the two functions in the [Client.h](https://github.com/espressif/arduino-esp32/blob/3.1.0-RC1/cores/esp32/Client.h) class that are incompatible of Arduino Core API's [Client.h](https://github.com/arduino/ArduinoCore-API/blob/master/api/Client.h) class.

```cpp
virtual int connect(IPAddress ip, uint16_t port, int32_t timeout) = 0;
virtual int connect(const char *host, uint16_t port, int32_t timeout) = 0;
```

https://github.com/espressif/arduino-esp32/commit/e1673e4c0ca89153e26d31cd783f0f8260c6d0fb#diff-5cf2184ea97bbef3467d24148540c862a33ca029b6604915dd7f6404d02c2d55R29
https://github.com/espressif/arduino-esp32/commit/e1673e4c0ca89153e26d31cd783f0f8260c6d0fb#diff-5cf2184ea97bbef3467d24148540c862a33ca029b6604915dd7f6404d02c2d55R31

This means the derived classes from Client class in any libraries and user code will not compile.

This is not the library issue but Core issue.

However, the fix will come, and I will update.

closed 2024-12-05 by mobizt

mobizt, 2024-12-05:
You can update to [v3.4.23](https://github.com/mobizt/ESP-Mail-Client/releases/tag/v3.4.23) which the issue was fixed.

HPChard, 2024-12-05:
Thank you for fixing this so quickly, very much appreciated!