mobizt/ESP-Mail-Client #370: esp32 3.1.1 compilation error

crasheg, 2025-01-08:
**Build tool used:**
- [ ] Arudino IDE 2.3.4, Windows 10 Pro
- [ ] ESP32 Core 3.1.1
- [ ] 

**Board used (ESP32/ESP8266/Arudino):**
- [ ] ESP32 Dev Board

**Other Libraries That are used:**

- Using library NTPClient at version 3.2.1 
- Using library WiFi at version 3.1.1
- Using library Networking at version 3.1.1
- Using library ESPmDNS at version 3.1.1 
- Using library WebServer at version 3.1.1
- Using library FS at version 3.1.1
- Using library EEPROM at version 3.1.1
- Using library GyverPortal at version 3.6.6 
- Using library DNSServer at version 3.1.1
- Using library ESP32 Async UDP at version 3.1.1
- Using library Update at version 3.1.1
- Using library GyverStepper at version 2.7
- Using library ESPping at version 1.0.5

**Description of problem:**

While compiling:
`In file included from ***/libraries/ESP_Mail_Client/src/client/SSLClient/client/BSSL_TCP_Client.h:58,
                 from ***/libraries/ESP_Mail_Client/src/client/SSLClient/ESP_SSLClient.h:39,
                 from ***/libraries/ESP_Mail_Client/src/client/ESP_Mail_TCPClient.h:58,
                 from ***\libraries\ESP_Mail_Client\src/ESP_Mail_Client.h:97,
                 from ***.ino:12:
***/libraries/ESP_Mail_Client/src/client/SSLClient/client/BSSL_SSL_Client.h:120:9: error: 'int BSSL_SSL_Client::connect(IPAddress, uint16_t, int32_t)' marked 'override', but does not override
  120 |     int connect(IPAddress ip, uint16_t port, int32_t timeout) override;
      |         ^~~~~~~
***/libraries/ESP_Mail_Client/src/client/SSLClient/client/BSSL_SSL_Client.h:121:9: error: 'int BSSL_SSL_Client::connect(const char*, uint16_t, int32_t)' marked 'override', but does not override
  121 |     int connect(const char *host, uint16_t port, int32_t timeout) override;
      |         ^~~~~~~
***/libraries/ESP_Mail_Client/src/client/SSLClient/client/BSSL_TCP_Client.h:111:9: error: 'int BSSL_TCP_Client::connect(IPAddress, uint16_t, int32_t)' marked 'override', but does not override
  111 |     int connect(IPAddress ip, uint16_t port, int32_t timeout) ESP32_ARDUINO_CORE_CLIENT_CONNECT_OVERRIDE;
      |         ^~~~~~~
***/libraries/ESP_Mail_Client/src/client/SSLClient/client/BSSL_TCP_Client.h:128:9: error: 'int BSSL_TCP_Client::connect(const char*, uint16_t, int32_t)' marked 'override', but does not override
  128 |     int connect(const char *host, uint16_t port, int32_t timeout) ESP32_ARDUINO_CORE_CLIENT_CONNECT_OVERRIDE;
      |         ^~~~~~~`

mobizt, 2025-01-09:
Thanks for reporting this issue.

The update [v3.4.24](https://github.com/mobizt/ESP-Mail-Client/releases/tag/v3.4.24) is available, please update.

closed 2025-01-09 by mobizt

crasheg, 2025-01-09:
> Thanks for reporting this issue.
> 
> The update [v3.4.24](https://github.com/mobizt/ESP-Mail-Client/releases/tag/v3.4.24) is available, please update.

working. thank you