mobizt/ESP-Mail-Client #376: Incompatibility with other libraires

LenStruttmann, 2025-04-28:
Good day,  I was attempting to add the ESP-Mail-Client to an existing PlatformIO project using the heltec_wifi_kit_32_V3 board and the libraries:

`lib_deps =
            rzeldent/micro-timezonedb @ 1.0.4
            olikraus/U8g2 @ 2.36.5`

After adding:

#include <ESP_Mail_Client.h>

...there were multiple compiler errors similar to:

`.pio/libdeps/heltec_wifi_kit_32_V3/ESP Mail Client/src/./extras/MB_String.h:119:22: error: expected nested-name-specifier before 'enable_if'
 #define MB_ENABLE_IF enable_if
                      ^~~~~~~~~`

To get the ESP-Mail-Client to compile, I made the following changes to:
 .pio/libdeps/heltec_wifi_kit_32_V3/ESP Mail Client/src/./extras/MB_String.h

 Change lines 114-120 from:
`#if defined(ARDUINO_ARCH_STM32) || defined(ARDUINO_ARCH_STM32F1) || defined(ARDUINO_ARCH_STM32F4)
#define MB_IS_SAME std::is_same
#define MB_ENABLE_IF std::enable_if
#else
#define MB_IS_SAME is_same
#define MB_ENABLE_IF enable_if
#endif`

...to:
`// #if defined(ARDUINO_ARCH_STM32) || defined(ARDUINO_ARCH_STM32F1) || defined(ARDUINO_ARCH_STM32F4)
#define MB_IS_SAME std::is_same
#define MB_ENABLE_IF std::enable_if
// #else
// #define MB_IS_SAME is_same
// #define MB_ENABLE_IF enable_if
// #endif`

Below is my platformio.ini file:

`[env:heltec_wifi_kit_32_V3]
platform        = espressif32
board           = heltec_wifi_kit_32_V3

framework       = arduino
; The log levels are 0=None, 1=Error, 2=Warning, 3=Info, 4=Debug, and 5=Verbose
build_flags     = -DCORE_DEBUG_LEVEL=0
                  -Wno-unused-variable
                  -Wno-unused-but-set-variable
                  -Wno-unused-function
                  -Wno-format-extra-args
                  -DHELTEC_WIFI_KIT_32_V3
; upload_speed    = 230400
monitor_speed   = 115200
monitor_filters =
                  default   ; Remove typical terminal control codes from input
                  ; time      ; Add timestamp with milliseconds for each new line
                  ; log2file  ; Log data to a file “platformio-device-monitor-*.log” located in the current working directory
                  send_on_enter
                  esp32_exception_decoder
monitor_echo    = yes
monitor_eol     = CRLF

; These are needed to get the serial monitor to work with this board
monitor_dtr     = 0
monitor_rts     = 0

lib_deps =
            rzeldent/micro-timezonedb @ 1.0.4
            olikraus/U8g2 @ 2.36.5
            https://github.com/mobizt/ESP-Mail-Client @ 3.4.24+sha.6865d7b`

mobizt, 2025-04-29:
That is not the issue of library but compiler's missing dependencies in your system.

I can compile with your platformio.ini without error.

In your platformio.ini instead of using deprecated framework from platformio team, you should use latest esp32 framework from pioarduino.

By replacing this

`platform = espressif32` 

with

`platform = https://github.com/pioarduino/platform-espressif32/releases/download/54.03.20/platform-espressif32.zip`

closed 2025-04-29 by mobizt

mobizt, 2025-04-29:
Anyway, the new fast and lightweight [ReadyMail](https://github.com/mobizt/ESP-Mail-Client?tab=readme-ov-file#new-library-is-available) library is available.

It provides the simple API but better performance, smaller size with async supports.

LenStruttmann, 2025-04-29:
Thanks for the info!